Benjamin-Dobell/Heimdall #252: Can't initialize connection with GT-I8190

minj, 2015-01-24:
```Initialising connection...
Detecting device...
      Manufacturer: "Samsung"
           Product: "Android"
         Serial No: "<censored>"

            length: 18
      device class: 0
               S/N: 4
           VID:PID: 04E8:685D
         bcdDevice: 0400
   iMan:iProd:iSer: 2:3:4
          nb confs: 1

interface[0].altsetting[0]: num endpoints = 2
   Class.SubClass.Protocol: FF.42.01
       endpoint[0].address: 81
           max packet size: 0200
          polling interval: 00
       endpoint[1].address: 01
           max packet size: 0200
          polling interval: 00
ERROR: Failed to find correct interface configuration```

3.16.7.4-1-MANJARO x86_64 (64 bit gcc: 4.9.2)
libusb-1.0.19-1
heimdall-git 69c3aaf

azrdev, 2018-01-07:
@minj same happened to me when I was not in correct "download mode". To get there, turn off the phone, then keep the three buttons Volume down and home and Power pressed until it tells something about download mode.

minj, 2018-01-08:
I had better success with this old phone some months ago using some non-git version in manjaro repos.

The phone itself struggles with new android builds of course :)

closed 2018-01-08 by minj