Benjamin-Dobell/Heimdall #254: Building problem

boriszor, 2015-01-30:
Hi all, I'm not sure if I'm doing something wrong but every time I try to build heimdall in VS2012 Professional Update 4 I get these errors and warnings:

------ Rebuild All started: Project: heimdall, Configuration: Release Win32 ------
2>  Arguments.cpp
2>  BridgeManager.cpp
2>C:\Heimdall-master\Heimdall-master\libusb-1.0\include\libusb.h(740): warning C4200: nonstandard extension used : zero-sized array in struct/union
2>          Cannot generate copy-ctor or copy-assignment operator when UDT contains a zero-sized array
2>C:\Heimdall-master\Heimdall-master\libusb-1.0\include\libusb.h(771): warning C4200: nonstandard extension used : zero-sized array in struct/union
2>          Cannot generate copy-ctor or copy-assignment operator when UDT contains a zero-sized array
2>C:\Heimdall-master\Heimdall-master\libusb-1.0\include\libusb.h(1265): warning C4200: nonstandard extension used : zero-sized array in struct/union
2>          Cannot generate copy-ctor or copy-assignment operator when UDT contains a zero-sized array
2>  ClosePcScreenAction.cpp
2>  DetectAction.cpp
2>  DownloadPitAction.cpp
2>  FlashAction.cpp
2>  HelpAction.cpp
2>  InfoAction.cpp
2>  PrintPitAction.cpp
2>  Interface.cpp
2>  main.cpp
2>  Utility.cpp
2>  VersionAction.cpp
2>BridgeManager.obj : error LNK2001: unresolved external symbol _libusb_get_string_descriptor_ascii@16
2>BridgeManager.obj : error LNK2001: unresolved external symbol _libusb_ref_device@4
2>BridgeManager.obj : error LNK2001: unresolved external symbol _libusb_claim_interface@8
2>BridgeManager.obj : error LNK2001: unresolved external symbol _libusb_exit@4
2>BridgeManager.obj : error LNK2001: unresolved external symbol _libusb_bulk_transfer@24
2>BridgeManager.obj : error LNK2001: unresolved external symbol _libusb_init@4
2>BridgeManager.obj : error LNK2001: unresolved external symbol _libusb_unref_device@4
2>BridgeManager.obj : error LNK2001: unresolved external symbol _libusb_get_device_descriptor@8
2>BridgeManager.obj : error LNK2001: unresolved external symbol _libusb_get_config_descriptor@12
2>BridgeManager.obj : error LNK2001: unresolved external symbol _libusb_close@4
2>BridgeManager.obj : error LNK2001: unresolved external symbol _libusb_open@8
2>BridgeManager.obj : error LNK2001: unresolved external symbol _libusb_free_device_list@8
2>BridgeManager.obj : error LNK2001: unresolved external symbol _libusb_free_config_descriptor@4
2>BridgeManager.obj : error LNK2001: unresolved external symbol _libusb_set_interface_alt_setting@12
2>BridgeManager.obj : error LNK2001: unresolved external symbol _libusb_get_device_list@8
2>BridgeManager.obj : error LNK2001: unresolved external symbol _libusb_release_interface@8
2>BridgeManager.obj : error LNK2001: unresolved external symbol _libusb_set_debug@8
2>C:\Heimdall-master\Heimdall-master\Win32\Release\heimdall.exe : fatal error LNK1120: 17 unresolved externals

Thanks.

frank2019, 2015-02-03:
i think  the  link error  is  maybe that     libusb-1.0.lib   is  not  in the right place.     by  settting   i saw  the setting  right  place  is  $(SolutionDir)libusb-1.0\lib\$(Platform)\$(Configuration)\;$(SolutionDir)$(Platform)\$(Configuration)\lib\

boriszor, 2015-02-03:
Frank thanks! I have made those folders and copied libusb-1.0.lib there the same errors again.

Benjamin-Dobell, 2015-02-19:
Windows builds are now built with MinGW (MSYS2) instead of Visual Studio. Please refer to the latest README for step-by-step instructions. If you run into any problems please open a new issue.

closed 2015-02-19 by Benjamin-Dobell