Benjamin-Dobell/Heimdall #255: download-pit fails; is it possible to use PIT file from the downloaded firmware files?

cowboy, 2015-01-31:
I want to revert my Galaxy S5 GM-900A to stock so that I can root it, but am having the same problems as everyone else.

```txt
Downloading device's PIT file...
ERROR: Failed to send request to end PIT file transfer!
ERROR: Failed to download PIT file!
Ending session...
ERROR: Failed to send end session packet!
Releasing device interface...
```

Is it possible to, instead of downloading the PIT file from the phone, to use the `KLTE_USA_ATT.pit` file that's in the `CSC_ATT_G900AATT1ANCE_964333_REV00_user_low_ship_MULTI_CERT.tar.md5` archive from the firmware I want to flash?

I don't know much about this, and I'd love some insight before I do something unrecoverable to my phone. Thanks!

vectorsigma, 2016-06-04:
So, I had a similar problem with a GM-900S (Sprint S5) using 1.4.1-5 (from Fedora 23 main-line).

Anyway, if you still have this phone, it's fixed as of what I cloned from master today and built/ran.

vectorsigma, 2016-06-04:
Probably can close this issue.