Benjamin-Dobell/Heimdall #257: Frontend looks horrible on high dpi screens

peterhoeg, 2015-02-10:
Lots of text is cut off and as the window cannot be resized, it's practically unusable.
[image]

Benjamin-Dobell, 2015-02-11:
Some details about your window manager would be handy. "Frontend looks horrible on high dpi screens" is a bit of an overly broad statement as it looks as it should on OS X with a retina display.

peterhoeg, 2015-02-11:
@Benjamin-Dobell my apologies, the issue should obviously have had far more details.

This is on Arch Linux with KDE (frameworks 5.6.0 and plasma desktop 5.2.0) and qt 4.8.6 and qt 5.4.0.

Other qt4 applications are not exhibiting this problem.

For starters, if the window was resizable, I think qt would simply lay things out again on resize, which should solve the problem.

CuriousTommy, 2017-06-24:
This is also a problem on Gnome 3 too, Ii was almost impossible to flash recovery until I forced the application to maximize.

Goldmaster, 2021-12-14:
This is also an issue on windows 10. when its set to 4k and display is set at 250% which are the default values. a temporary workaround is to set to 1920x1080 and set size to 100%

Before
[image]

After
[image]

roere, 2023-11-06:
Same problem here: 
- Linux Mint 21.2
- Heimdall 1.4.2
- Screen resolution 3840 x 2160 with 200% zoom 
Issue seems to be, that the text is zoomed to 200% and the windows is not. I can change the screen resolution while working with Heimdall (still then I cannot access all buttons). Would be nice if I could just resize the Heimdall Window.

[image]