Benjamin-Dobell/Heimdall #259: ERROR: Failed to download PIT file!

gerroon, 2015-02-19:
Debian Jessie SM-T320 

heimdall print-pit
Heimdall v1.4.0

Copyright (c) 2010-2013, Benjamin Dobell, Glass Echidna
http://www.glassechidna.com.au/

This software is provided free of charge. Copying and redistribution is
encouraged.

If you appreciate this software and you would like to support future
development please consider donating:
http://www.glassechidna.com.au/donate/

Initialising connection...
Detecting device...
Claiming interface...
Attempt failed. Detaching driver...
Claiming interface again...
Setting up interface...

Initialising protocol...
Protocol initialisation successful.

Beginning session...

Some devices may take up to 2 minutes to respond.
Please be patient!

Session begun.

Downloading device's PIT file...
ERROR: Failed to send request to end PIT file transfer!
ERROR: Failed to download PIT file!
Ending session...
ERROR: Failed to send end session packet!
Releasing device interface...
Re-attaching kernel driver...

julianec, 2015-03-21:
Galaxy S5, Heimdall v1.4.0, Debian jessy

sudo heimdall flash --recovery recovery.img --no-reboot --verbose                             21:33
Heimdall v1.4.0

Copyright (c) 2010-2013, Benjamin Dobell, Glass Echidna
http://www.glassechidna.com.au/

This software is provided free of charge. Copying and redistribution is
encouraged.

If you appreciate this software and you would like to support future
development please consider donating:
http://www.glassechidna.com.au/donate/

Initialising connection...
Detecting device...
      Manufacturer: "Sasmsung"
           Product: "MSM8960"
            length: 18
      device class: 2
               S/N: 0
           VID:PID: 04E8:685D
         bcdDevice: 0100
   iMan:iProd:iSer: 1:2:0
          nb confs: 1

interface[0].altsetting[0]: num endpoints = 1
   Class.SubClass.Protocol: 02.02.01
       endpoint[0].address: 82
           max packet size: 0010
          polling interval: 09

interface[1].altsetting[0]: num endpoints = 2
   Class.SubClass.Protocol: 0A.00.00
       endpoint[0].address: 81
           max packet size: 0200
          polling interval: 00
       endpoint[1].address: 01
           max packet size: 0200
          polling interval: 00
Claiming interface...
Attempt failed. Detaching driver...
Claiming interface again...
Setting up interface...

Initialising protocol...
WARNING: Control transfer #1 failed. Result: -9
WARNING: Control transfer #2 failed. Result: -9
WARNING: Control transfer #3 failed. Result: -9
WARNING: Control transfer #4 failed. Result: -9
WARNING: Control transfer #5 failed. Result: -9
WARNING: Control transfer #6 failed. Result: -9
Protocol initialisation successful.

Beginning session...

Some devices may take up to 2 minutes to respond.
Please be patient!

Session begun.

Downloading device's PIT file...
ERROR: libusb error -7 whilst sending packet. Retrying...
ERROR: libusb error -7 whilst sending packet. Retrying...
ERROR: libusb error -7 whilst sending packet. Retrying...
ERROR: libusb error -7 whilst sending packet. Retrying...
ERROR: libusb error -7 whilst sending packet. Retrying...
ERROR: libusb error -7 whilst sending packet.
ERROR: Failed to send request to end PIT file transfer!
ERROR: Failed to download PIT file!
Ending session...
ERROR: libusb error -7 whilst sending packet. Retrying...
ERROR: libusb error -7 whilst sending packet. Retrying...
ERROR: libusb error -7 whilst sending packet. Retrying...
ERROR: libusb error -7 whilst sending packet. Retrying...
ERROR: libusb error -7 whilst sending packet. Retrying...
ERROR: libusb error -7 whilst sending packet.
ERROR: Failed to send end session packet!
Releasing device interface...
Re-attaching kernel driver...

julianec, 2015-03-21:
With a build from the source I get this:

Initialising connection...
Detecting device...
      Manufacturer: "Sasmsung"
           Product: "MSM8960"

            length: 18
      device class: 2
               S/N: 0
           VID:PID: 04E8:685D
         bcdDevice: 0100
   iMan:iProd:iSer: 1:2:0
          nb confs: 1

interface[0].altsetting[0]: num endpoints = 1
   Class.SubClass.Protocol: 02.02.01
       endpoint[0].address: 82
           max packet size: 0010
          polling interval: 09

interface[1].altsetting[0]: num endpoints = 2
   Class.SubClass.Protocol: 0A.00.00
       endpoint[0].address: 81
           max packet size: 0200
          polling interval: 00
       endpoint[1].address: 01
           max packet size: 0200
          polling interval: 00
Claiming interface...
Attempt failed. Detaching driver...
Claiming interface again...
Setting up interface...

Initialising protocol...
ERROR: Failed to receive handshake response. Result: -7
ERROR: Protocol initialisation failed!

Releasing device interface...
Re-attaching kernel driver...

craigem, 2015-04-11:
__Galaxy S5, Heimdall v1.4.0, Debian Jessie__

I also get this problem. Completely unclear on how to resolve it at this point:

```text
% heimdall print-pit --verbose
Heimdall v1.4.0
Copyright (c) 2010-2013, Benjamin Dobell, Glass Echidna
http://www.glassechidna.com.au/
This software is provided free of charge. Copying and redistribution is
encouraged.
If you appreciate this software and you would like to support future
development please consider donating:
http://www.glassechidna.com.au/donate/
Initialising connection...
Detecting device...
      Manufacturer: "Sasmsung"
           Product: "MSM8960"
            length: 18
      device class: 2
               S/N: 0
           VID:PID: 04E8:685D
         bcdDevice: 0100
   iMan:iProd:iSer: 1:2:0
          nb confs: 1
interface[0].altsetting[0]: num endpoints = 1
   Class.SubClass.Protocol: 02.02.01
       endpoint[0].address: 82
           max packet size: 0010
          polling interval: 09
interface[1].altsetting[0]: num endpoints = 2
   Class.SubClass.Protocol: 0A.00.00
       endpoint[0].address: 81
           max packet size: 0200
          polling interval: 00
       endpoint[1].address: 01
           max packet size: 0200
          polling interval: 00
Claiming interface...
Attempt failed. Detaching driver...
Claiming interface again...
Setting up interface...
Initialising protocol...
WARNING: Control transfer #1 failed. Result: -9
WARNING: Control transfer #2 failed. Result: -9
WARNING: Control transfer #3 failed. Result: -9
WARNING: Control transfer #4 failed. Result: -9
WARNING: Control transfer #5 failed. Result: -9
WARNING: Control transfer #6 failed. Result: -9
Protocol initialisation successful.
Beginning session...
Some devices may take up to 2 minutes to respond.
Please be patient!
Session begun.
Downloading device's PIT file...
ERROR: libusb error -7 whilst sending packet. Retrying...
ERROR: libusb error -7 whilst sending packet. Retrying...
ERROR: libusb error -7 whilst sending packet. Retrying...
ERROR: libusb error -7 whilst sending packet. Retrying...
ERROR: libusb error -7 whilst sending packet. Retrying...
ERROR: libusb error -7 whilst sending packet.
ERROR: Failed to send request to end PIT file transfer!
ERROR: Failed to download PIT file!
Ending session...
ERROR: libusb error -7 whilst sending packet. Retrying...
ERROR: libusb error -7 whilst sending packet. Retrying...
ERROR: libusb error -7 whilst sending packet. Retrying...
ERROR: libusb error -7 whilst sending packet. Retrying...
ERROR: libusb error -7 whilst sending packet. Retrying...
ERROR: libusb error -7 whilst sending packet.
ERROR: Failed to send end session packet!
Releasing device interface...
Re-attaching kernel driver...
```

abunaideshou, 2015-05-03:
I am also getting the same errors on Debian Jessie with my Galaxy Note 3 (sm-n900p).

viniciusccarvalho, 2015-08-11:
Same errors on v1.4.0 on Galaxy S5 on a Mac

NoFaQ, 2015-08-11:
I also reinstalled firmware and recovery through Odin, then tried to pull PIT from Heimdall, and still could not.

mcamou, 2015-08-16:
Same thing on Galaxy Note 4 (N910F) on 1.4.0. I tried a `print-pit` with `--usb-log-level debug --stdout-errors --verbose` and got the following:

```
$ sudo heimdall print-pit --usb-log-level debug --stdout-errors --verbose
Heimdall v1.4.0

Copyright (c) 2010-2013, Benjamin Dobell, Glass Echidna
http://www.glassechidna.com.au/

This software is provided free of charge. Copying and redistribution is
encouraged.

If you appreciate this software and you would like to support future
development please consider donating:
http://www.glassechidna.com.au/donate/

Initialising connection...
Detecting device...
[timestamp] [threadID] facility level [function call] <message>
--------------------------------------------------------------------------------
[ 0.000354] [00000513] libusbx: debug [libusb_get_device_list] 
[ 0.001567] [00000513] libusbx: debug [process_new_device] allocating new device for location 0x15000000
[ 0.001598] [00000513] libusbx: debug [darwin_cache_device_descriptor] device descriptor:
[ 0.001602] [00000513] libusbx: debug [darwin_cache_device_descriptor]  bDescriptorType:    0x01
[ 0.001606] [00000513] libusbx: debug [darwin_cache_device_descriptor]  bcdUSB:             0x0300
[ 0.001609] [00000513] libusbx: debug [darwin_cache_device_descriptor]  bDeviceClass:       0x09
[ 0.001612] [00000513] libusbx: debug [darwin_cache_device_descriptor]  bDeviceSubClass:    0x00
[ 0.001616] [00000513] libusbx: debug [darwin_cache_device_descriptor]  bDeviceProtocol:    0x03
[ 0.001619] [00000513] libusbx: debug [darwin_cache_device_descriptor]  bMaxPacketSize0:    0x09
[ 0.001622] [00000513] libusbx: debug [darwin_cache_device_descriptor]  idVendor:           0x05ac
[ 0.001625] [00000513] libusbx: debug [darwin_cache_device_descriptor]  idProduct:          0x8007
[ 0.001628] [00000513] libusbx: debug [darwin_cache_device_descriptor]  bcdDevice:          0x0300
[ 0.001632] [00000513] libusbx: debug [darwin_cache_device_descriptor]  iManufacturer:      0x02
[ 0.001635] [00000513] libusbx: debug [darwin_cache_device_descriptor]  iProduct:           0x01
[ 0.001638] [00000513] libusbx: debug [darwin_cache_device_descriptor]  iSerialNumber:      0x00
[ 0.001642] [00000513] libusbx: debug [darwin_cache_device_descriptor]  bNumConfigurations: 0x01
[ 0.001662] [00000513] libusbx: debug [darwin_check_configuration] active config: 1, first config: 1
[ 0.001676] [00000513] libusbx: debug [process_new_device] found device with address 128 port = 0 parent = 0x7f9eac104218 at 0x0
[ 0.002035] [00000513] libusbx: debug [process_new_device] allocating new device for location 0x14000000
[ 0.002058] [00000513] libusbx: debug [darwin_cache_device_descriptor] device descriptor:
[ 0.002062] [00000513] libusbx: debug [darwin_cache_device_descriptor]  bDescriptorType:    0x01
[ 0.002065] [00000513] libusbx: debug [darwin_cache_device_descriptor]  bcdUSB:             0x0200
[ 0.002068] [00000513] libusbx: debug [darwin_cache_device_descriptor]  bDeviceClass:       0x09
[ 0.002071] [00000513] libusbx: debug [darwin_cache_device_descriptor]  bDeviceSubClass:    0x00
[ 0.002074] [00000513] libusbx: debug [darwin_cache_device_descriptor]  bDeviceProtocol:    0x01
[ 0.002077] [00000513] libusbx: debug [darwin_cache_device_descriptor]  bMaxPacketSize0:    0x09
[ 0.002089] [00000513] libusbx: debug [darwin_cache_device_descriptor]  idVendor:           0x05ac
[ 0.002092] [00000513] libusbx: debug [darwin_cache_device_descriptor]  idProduct:          0x8007
[ 0.002095] [00000513] libusbx: debug [darwin_cache_device_descriptor]  bcdDevice:          0x0300
[ 0.002098] [00000513] libusbx: debug [darwin_cache_device_descriptor]  iManufacturer:      0x02
[ 0.002101] [00000513] libusbx: debug [darwin_cache_device_descriptor]  iProduct:           0x01
[ 0.002104] [00000513] libusbx: debug [darwin_cache_device_descriptor]  iSerialNumber:      0x00
[ 0.002108] [00000513] libusbx: debug [darwin_cache_device_descriptor]  bNumConfigurations: 0x01
[ 0.002127] [00000513] libusbx: debug [darwin_check_configuration] active config: 1, first config: 1
[ 0.002132] [00000513] libusbx: debug [process_new_device] found device with address 129 port = 0 parent = 0x7f9eac104458 at 0x0
[ 0.002490] [00000513] libusbx: debug [process_new_device] allocating new device for location 0x14800000
[ 0.002724] [00000513] libusbx: debug [darwin_cache_device_descriptor] device descriptor:
[ 0.002734] [00000513] libusbx: debug [darwin_cache_device_descriptor]  bDescriptorType:    0x01
[ 0.002742] [00000513] libusbx: debug [darwin_cache_device_descriptor]  bcdUSB:             0x0200
[ 0.002746] [00000513] libusbx: debug [darwin_cache_device_descriptor]  bDeviceClass:       0x09
[ 0.002749] [00000513] libusbx: debug [darwin_cache_device_descriptor]  bDeviceSubClass:    0x00
[ 0.002752] [00000513] libusbx: debug [darwin_cache_device_descriptor]  bDeviceProtocol:    0x00
[ 0.002755] [00000513] libusbx: debug [darwin_cache_device_descriptor]  bMaxPacketSize0:    0x08
[ 0.002758] [00000513] libusbx: debug [darwin_cache_device_descriptor]  idVendor:           0x0a5c
[ 0.002761] [00000513] libusbx: debug [darwin_cache_device_descriptor]  idProduct:          0x4500
[ 0.002764] [00000513] libusbx: debug [darwin_cache_device_descriptor]  bcdDevice:          0x0100
[ 0.002767] [00000513] libusbx: debug [darwin_cache_device_descriptor]  iManufacturer:      0x01
[ 0.002770] [00000513] libusbx: debug [darwin_cache_device_descriptor]  iProduct:           0x02
[... cut ...]
ERROR: Protocol initialisation failed!

ERROR: Protocol initialisation failed!

Releasing device interface...
[15.901258] [00000513] libusbx: debug [libusb_release_interface] interface 1

[15.901584] [00000513] libusbx: debug [libusb_close] 
[15.901711] [00000513] libusbx: debug [usbi_remove_pollfd] remove fd 6
[15.901748] [00000513] libusbx: debug [libusb_unref_device] destroy device 20.15
[15.901761] [00000513] libusbx: debug [libusb_exit] 
[15.901765] [00000513] libusbx: debug [libusb_exit] destroying default context
```

Benjamin-Dobell, 2015-10-08:
Closing as I believe this was resolved Feb 21st.

closed 2015-10-08 by Benjamin-Dobell