Benjamin-Dobell/Heimdall #261: OS_LINUX option does not applyed in Lubuntu 14.04.2 LTS

Chilling-in-my-home, 2015-02-20:
I used the cord in the github.

When I try to flash something the program said

Initialising connection...
Detecting device...
      Manufacturer: "Sasmsung"
           Product: "MSM7x27"

            length: 18
      device class: 2
               S/N: 0
           VID:PID: 04E8:685D
         bcdDevice: 0100
   iMan:iProd:iSer: 1:2:0
          nb confs: 1

interface[0].altsetting[0]: num endpoints = 1
   Class.SubClass.Protocol: 02.02.01
       endpoint[0].address: 82
           max packet size: 0010
          polling interval: 09

interface[1].altsetting[0]: num endpoints = 2
   Class.SubClass.Protocol: 0A.00.00
       endpoint[0].address: 81
           max packet size: 0200
          polling interval: 00
       endpoint[1].address: 01
           max packet size: 0200
          polling interval: 00
Claiming interface...
ERROR: Claiming interface failed!

I trouble shooted what's wrong for a long time, and I finally found the reason why.
Some of codes in BridgeManager.cpp was in (#ifdef OS_LINUX >>>> #endif ) format, and they should execute on linux machine. However, my machine didn't execute the codes specifically written for linux.

The temporary solution, for me was putting a varibale bool detachedDriver=true; on the top of the file, and remove all the #ifdef OS_LINUX and #endif.(I can not fix it elegantly cause I just looked through the cord.)

Do not apply this solution when your OS is not linux, cause I don't know what would happen when these cords are applied to another OS.

After applying my solution, I could flash files to my phone.
If you are using OS_LINUX and can not flash due to the same error message as I am, try it. And please check the cords out cause the (#ifdef OS_LINUX >>>> #endif ) cords "should" work for Linux without any problem.

Benjamin-Dobell, 2015-02-20:
Whoops! Fixed now.

closed 2015-02-20 by Benjamin-Dobell

domenpuncer, 2015-03-05:
Hi Benjamin,

Are you sure about that (doesn't seem to work here)?

http://www.cmake.org/Wiki/CMake_Useful_Variables doesn't describe a predefined LINUX.

Benjamin-Dobell, 2015-03-05:
... and that's what I get for quickly pushing "fixes" without actually testing them.

commit 8cb7f3dd (2015-03-05): Actually fix #261 (CMake Linux detection)