Benjamin-Dobell/Heimdall #262: Protocol initialisation fails when connected via usb hub

munimkazia, 2015-03-11:
Basically, I want to flash a bunch of devices connected to a machine running fedora 20. I have compiled libusb from libusb.info and the latest master branch of heimdall. 
It seems to be working perfectly when I connect the device to the machine directly, but fails when connected via a hub. Rebooting the machine, reconnecting the device to the hub doesn't seem to work.  
This is kinda important because I won't have physical access to the devices, they will be connected to a remote server via a USB hub. 

The following output is for a print-pit command for Samsung Galaxy s4 (GT I9505). 

I have went through the code and I know my way around c++ (though I am new to libusb), so any helpful ideas/pointers on fixing this issue will help a lot. 

```
sudo ./heimdall print-pit --verbose --usb-log-level debug
Heimdall v1.4.1

Copyright (c) 2010-2014 Benjamin Dobell, Glass Echidna
http://www.glassechidna.com.au/

This software is provided free of charge. Copying and redistribution is
encouraged.

If you appreciate this software and you would like to support future
development please consider donating:
http://www.glassechidna.com.au/donate/

Initialising connection...
Detecting device...
[timestamp] [threadID] facility level [function call] <message>
--------------------------------------------------------------------------------
[ 0.007341] [00002df0] libusbx: debug [libusb_get_device_list]
[ 0.007405] [00002df0] libusbx: debug [discovered_devs_append] need to increase capacity
[ 0.007422] [00002df0] libusbx: debug [discovered_devs_append] need to increase capacity
[ 0.007441] [00002df0] libusbx: debug [libusb_get_device_descriptor]
[ 0.007454] [00002df0] libusbx: debug [libusb_get_device_descriptor]
[ 0.007466] [00002df0] libusbx: debug [libusb_get_device_descriptor]
[ 0.007480] [00002df0] libusbx: debug [libusb_get_device_descriptor]
[ 0.007492] [00002df0] libusbx: debug [libusb_get_device_descriptor]
[ 0.007524] [00002df0] libusbx: debug [libusb_get_device_descriptor]
[ 0.007538] [00002df0] libusbx: debug [libusb_get_device_descriptor]
[ 0.007558] [00002df0] libusbx: debug [libusb_get_device_descriptor]
[ 0.007575] [00002df0] libusbx: debug [libusb_get_device_descriptor]
[ 0.007619] [00002df0] libusbx: debug [libusb_get_device_descriptor]
[ 0.007641] [00002df0] libusbx: debug [libusb_get_device_descriptor]
[ 0.007653] [00002df0] libusbx: debug [libusb_get_device_descriptor]
[ 0.007665] [00002df0] libusbx: debug [libusb_get_device_descriptor]
[ 0.007678] [00002df0] libusbx: debug [libusb_get_device_descriptor]
[ 0.007694] [00002df0] libusbx: debug [libusb_open] open 3.6
[ 0.007742] [00002df0] libusbx: debug [usbi_add_pollfd] add fd 11 events 4
[ 0.007781] [00002df0] libusbx: debug [libusb_get_device_descriptor]
[ 0.007805] [00002df0] libusbx: debug [add_to_flying_list] arm timerfd for timeout in 1000ms (first in line)
[ 0.007854] [00002df0] libusbx: debug [libusb_handle_events_timeout_completed] doing our own event handling
[ 0.007874] [00002df0] libusbx: debug [handle_events] poll() 4 fds with timeout in 60000ms
[ 0.007974] [00002df0] libusbx: debug [handle_events] poll() returned 1
[ 0.007999] [00002df0] libusbx: debug [reap_for_handle] urb type=2 status=0 transferred=4
[ 0.008007] [00002df0] libusbx: debug [handle_control_completion] handling completion status 0
[ 0.008015] [00002df0] libusbx: debug [disarm_timerfd]
[ 0.008022] [00002df0] libusbx: debug [usbi_handle_transfer_completion] transfer 0x11cf0b8 has callback 0x7f09e6118030
[ 0.008029] [00002df0] libusbx: debug [sync_transfer_cb] actual_length=4
[ 0.008050] [00002df0] libusbx: debug [add_to_flying_list] arm timerfd for timeout in 1000ms (first in line)
[ 0.008069] [00002df0] libusbx: debug [libusb_handle_events_timeout_completed] doing our own event handling
[ 0.008076] [00002df0] libusbx: debug [handle_events] poll() 4 fds with timeout in 60000ms
[ 0.008179] [00002df0] libusbx: debug [handle_events] poll() returned 1
[ 0.008200] [00002df0] libusbx: debug [reap_for_handle] urb type=2 status=0 transferred=18
[ 0.008211] [00002df0] libusbx: debug [handle_control_completion] handling completion status 0
[ 0.008222] [00002df0] libusbx: debug [disarm_timerfd]
[ 0.008234] [00002df0] libusbx: debug [usbi_handle_transfer_completion] transfer 0x11cf0b8 has callback 0x7f09e6118030
[ 0.008240] [00002df0] libusbx: debug [sync_transfer_cb] actual_length=18
      Manufacturer: "Sasmsung"
[ 0.008257] [00002df0] libusbx: debug [add_to_flying_list] arm timerfd for timeout in 1000ms (first in line)
[ 0.008270] [00002df0] libusbx: debug [libusb_handle_events_timeout_completed] doing our own event handling
[ 0.008276] [00002df0] libusbx: debug [handle_events] poll() 4 fds with timeout in 60000ms
[ 0.008364] [00002df0] libusbx: debug [handle_events] poll() returned 1
[ 0.008383] [00002df0] libusbx: debug [reap_for_handle] urb type=2 status=0 transferred=4
[... 61 lines cut ...]
[ 1.009738] [00002df0] libusbx: debug [libusb_cancel_transfer]
[ 1.009865] [00002df0] libusbx: debug [disarm_timerfd]
[ 1.009927] [00002df0] libusbx: debug [libusb_handle_events_timeout_completed] doing our own event handling
[ 1.009951] [00002df0] libusbx: debug [handle_events] poll() 4 fds with timeout in 60000ms
[ 1.009978] [00002df0] libusbx: debug [handle_events] poll() returned 1
[ 1.010008] [00002df0] libusbx: debug [reap_for_handle] urb type=3 status=-2 transferred=0
[ 1.010028] [00002df0] libusbx: debug [handle_bulk_completion] handling completion status -2 of bulk urb 1/1
[ 1.010050] [00002df0] libusbx: debug [handle_bulk_completion] abnormal reap: urb status -2
[ 1.010061] [00002df0] libusbx: debug [handle_bulk_completion] abnormal reap: last URB handled, reporting
[ 1.010072] [00002df0] libusbx: debug [usbi_handle_transfer_cancellation] detected timeout cancellation
[ 1.010082] [00002df0] libusbx: debug [disarm_timerfd]
[ 1.010093] [00002df0] libusbx: debug [usbi_handle_transfer_completion] transfer 0x11cf0b8 has callback 0x7f09e6118030
[ 1.010103] [00002df0] libusbx: debug [sync_transfer_cb] actual_length=0
ERROR: Failed to receive handshake response. Result: -7
ERROR: Protocol initialisation failed!

Releasing device interface...
[ 1.010210] [00002df0] libusbx: debug [libusb_release_interface] interface 1

[ 1.010261] [00002df0] libusbx: debug [libusb_close]
[ 1.010288] [00002df0] libusbx: debug [usbi_remove_pollfd] remove fd 11
[ 1.010316] [00002df0] libusbx: debug [libusb_exit]
[ 1.010328] [00002df0] libusbx: debug [libusb_exit] destroying default context

```

Benjamin-Dobell, 2015-03-11:
I'm pretty sure this is lower level than Heimdall itself. Possibly a bug in libusb-1.0, kernel drivers or very likely a faulty hub. Nonetheless, rather than speculating I'd love to get to the bottom of this also.

It's a lot easier to take a look at the USB packet captures than try read debug output; so my recommendation would be that you take a USB capture from the external hub and again from your computers inbuilt hub so that we can compare them.

https://wiki.wireshark.org/CaptureSetup/USB

munimkazia, 2015-03-11:
@Benjamin-Dobell Thanks for the quick reply. Of the three possibilities, I'll rule out faulty hub as the devices do work fine on adb otherwise. I didn't think about doubting libusb because it seemed like an old mature library, but you could be right. Anyway, I'll try a USB capture and get back. Thanks again!

munimkazia, 2015-03-11:
@Benjamin-Dobell Interesting update FYI. I went down to the server room with my macbook air and tried a few experiments. It seems that the same hub with the same devices if connected to my macbook. I have the master branch which was compiled last week in my macbook, but I didn't compile libusb locally. 

This is pretty much consistent behaviour too (never works on hub when connected to linux server, always works when connected directly to server, always works when connected to mac, even via hub).  

I am going to try an older version of libusb matching with the one that's on my mac, and if that doesn't work, it is probably a special linux-only issue here.  

In any case, this is definitely not an issue with Heimdall code now.

munimkazia, 2015-03-11:
@Benjamin-Dobell Doing a yum update on the server worked! I am guessing that the new kernel had fixed some USB issues, but it could have been any other package too. I have to update another server to mine to recheck.    
FYI, this is Fedora 20. New kernel package is 3.18.7-100.fc20.x86_64 and the old one was 3.11.10-301.fc20.x86_64

closed 2015-03-11 by munimkazia