Benjamin-Dobell/Heimdall #263: Information on the build system/distro used

ntzrmtthihu777, 2015-03-15:
Greetings. I see you've now moved over to cmake from autotools (regretably, in my opinion) and as such use libusbConfig.cmake or libusb-config.cmake to find the libusb files. As my distro (archlinux) currently packages libusb, these files do not exist, so I was wondering what distro (if any) you are using that provides these files.

Benjamin-Dobell, 2015-03-16:
You haven't described the error you're getting so I'm not sure exactly how to help. However, the cmake file to find libusb is bundled in Heimdall'sc repo - it is not part of cmake itself.

https://github.com/Benjamin-Dobell/Heimdall/blob/master/cmake/Findlibusb.cmake

ntzrmtthihu777, 2015-03-17:
```
 cmake -DCMAKE_BUILD_TYPE=Release ..
-- The C compiler identification is GNU 4.9.2
-- The CXX compiler identification is GNU 4.9.2
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
CMake Error at /usr/share/cmake-3.2/Modules/FindPackageHandleStandardArgs.cmake:138 (message):
  Could NOT find libusb (missing: LIBUSB_LIBRARY)
Call Stack (most recent call first):
  /usr/share/cmake-3.2/Modules/FindPackageHandleStandardArgs.cmake:374 (_FPHSA_FAILURE_MESSAGE)
  cmake/Findlibusb.cmake:30 (find_package_handle_standard_args)
  heimdall/CMakeLists.txt:6 (find_package)

-- Configuring incomplete, errors occurred!
See also "heimdall/build/CMakeFiles/CMakeOutput.log".

````

is the result of following the instructions in `heimdall/Linux/README`

ntzrmtthihu777, 2015-03-17:
Update; apparently `heimdall/heimdall/CMakeLists.txt:6` forces the usage of a static libusb; my distro does not ship that, commenting out that line allowed it all to build fine. Personally I have no aversion to static linking if you're building a portable version of a given application, but it would be nice to have a config option one could set to chose one or the other.

Marking as closed, but you should consider adding a choice.

closed 2015-03-17 by ntzrmtthihu777

Benjamin-Dobell, 2015-03-17:
@ntzrmtthihu777 Thanks for the details.

I'll update so that static libusb-1.0 is only required on Windows and Mac - there's no real need to link statically on Linux.

ntzrmtthihu777, 2015-03-20:
Thanks a lot for that could you make a comment here when that change takes effect because I need make sure that the Archlinux PKGBUILD is fixed so I can remove my little commenting out bit

Benjamin-Dobell, 2015-03-21:
Whoops, thanks for the reminder. All done now:

https://github.com/Benjamin-Dobell/Heimdall/commit/d0526a3b74a003dfc6f805682693be9173ffcd88