Benjamin-Dobell/Heimdall #264: Claiming interface failed on Heimdall 1.4.0

darrelltan, 2015-03-24:
Tried to download the pit from the device (Samsung GT-I8552) but it showed me this message. I've reinstalled Heimdall multiple times and it's still not working. Help please!

darrelltan, 2015-03-24:
By the way, I'm using an MBP on mavericks (OSX 10.9.5)

ashleyconnor, 2015-07-21:
On Mac OS X you may see this message: "ERROR: Claiming interface failed!". 
    If so, you need to disable the default USB drivers with the following commands:
    sudo kextunload -b com.apple.driver.AppleUSBCDC;
    sudo kextunload -b com.apple.driver.AppleUSBCDCACMControl;
    sudo kextunload -b com.apple.driver.AppleUSBCDCACMData;
    If you have installed Samsung Kies, you may further need to 
    unload those USB drivers with the following commands:
    sudo kextunload -b com.devguru.driver.SamsungComposite;
    sudo kextunload -b com.devguru.driver.SamsungACMData
    sudo kextunload -b com.devguru.driver.SamsungACMControl

http://wiki.cyanogenmod.org/w/Install_CM_for_jflte