Benjamin-Dobell/Heimdall #267: N7100 (Note II) listed as "Gadget Serial" insted of "Samsung Composite Device"

ionree, 2015-03-30:
This has nothing to do with Heimdall itself, but I think it would help a lot of people with their confusion when using Heimdall for the N7100 if you added this to the instructions/readme. When in download mode, windows lists the device as "Gadget Serial" instead of "Samsung Composite USB Device".
I dont know if this happens on all Note II devices though.
Also, there's a small typo on line 41 of the Windows README :laughing:

djevertguzman, 2015-03-31:
it does not happen on the sprint note 2 or 4

frank2019, 2015-04-01:
on  I9300  it  list as  "MSM8960"

Benjamin-Dobell, 2015-10-08:
Yeah, unfortunately the devices say all sorts of garbage.

I'm closing this as there's not a "one size fits all" solution that I'm aware of.

closed 2015-10-08 by Benjamin-Dobell