Benjamin-Dobell/Heimdall #268: heimdall fails print / downloading pit on Samsung mega 5.8

beniz, 2015-04-01:
Hi there, thanks for the good work, here is the mighty error log, let me know how I can help debug / dev:
```
sudo heimdall print-pit --verbose --no-reboot

Heimdall v1.4.0

Copyright (c) 2010-2013, Benjamin Dobell, Glass Echidna
http://www.glassechidna.com.au/

This software is provided free of charge. Copying and redistribution is
encouraged.

If you appreciate this software and you would like to support future
development please consider donating:
http://www.glassechidna.com.au/donate/

Initialising connection...
Detecting device...
      Manufacturer: "Sasmsung"
           Product: "MSM8960"

            length: 18
      device class: 2
               S/N: 0
           VID:PID: 04E8:685D
         bcdDevice: 0100
   iMan:iProd:iSer: 1:2:0
          nb confs: 1

interface[0].altsetting[0]: num endpoints = 1
   Class.SubClass.Protocol: 02.02.01
       endpoint[0].address: 82
           max packet size: 0010
          polling interval: 09

interface[1].altsetting[0]: num endpoints = 2
   Class.SubClass.Protocol: 0A.00.00
       endpoint[0].address: 81
           max packet size: 0200
          polling interval: 00
       endpoint[1].address: 01
           max packet size: 0200
          polling interval: 00
Claiming interface...
Attempt failed. Detaching driver...
Claiming interface again...
Setting up interface...

Initialising protocol...
WARNING: Control transfer #1 failed. Result: -9
WARNING: Control transfer #2 failed. Result: -9
WARNING: Control transfer #3 failed. Result: -9
WARNING: Control transfer #4 failed. Result: -9
WARNING: Control transfer #5 failed. Result: -9
WARNING: Control transfer #6 failed. Result: -9
ERROR: Failed to receive handshake response. Retrying...
Protocol initialisation successful.

Beginning session...

Some devices may take up to 2 minutes to respond.
Please be patient!

Session begun.

Downloading device's PIT file...
ERROR: libusb error -7 whilst sending packet. Retrying...
ERROR: libusb error -7 whilst sending packet. Retrying...
ERROR: libusb error -7 whilst sending packet. Retrying...
ERROR: libusb error -7 whilst sending packet. Retrying...
ERROR: libusb error -7 whilst sending packet. Retrying...
ERROR: libusb error -7 whilst sending packet.
ERROR: Failed to send request to end PIT file transfer!
ERROR: Failed to download PIT file!
Ending session...
ERROR: libusb error -7 whilst sending packet. Retrying...
ERROR: libusb error -7 whilst sending packet. Retrying...
ERROR: libusb error -7 whilst sending packet. Retrying...
ERROR: libusb error -7 whilst sending packet. Retrying...
ERROR: libusb error -7 whilst sending packet. Retrying...
ERROR: libusb error -7 whilst sending packet.
ERROR: Failed to send end session packet!
Releasing device interface...
Re-attaching kernel driver...
```

beniz, 2015-04-01:
FTR, I've tried the heimdall packaged with Ubuntu 14.10, and the current head of this repo as well, to no avail. I've also tried https://github.com/sshimko/Heimdall as refered to in #203 and the error is shorter:
```
sudo ./heimdall print-pit --verbose --no-reboot
Heimdall v1.4.1

Copyright (c) 2010-2014 Benjamin Dobell, Glass Echidna
http://www.glassechidna.com.au/

This software is provided free of charge. Copying and redistribution is
encouraged.

If you appreciate this software and you would like to support future
development please consider donating:
http://www.glassechidna.com.au/donate/

Initialising connection...
Detecting device...
      Manufacturer: "Sasmsung"
           Product: "MSM8960"

            length: 18
      device class: 2
               S/N: 0
           VID:PID: 04E8:685D
         bcdDevice: 0100
   iMan:iProd:iSer: 1:2:0
          nb confs: 1

interface[0].altsetting[0]: num endpoints = 1
   Class.SubClass.Protocol: 02.02.01
       endpoint[0].address: 82
           max packet size: 0010
          polling interval: 09

interface[1].altsetting[0]: num endpoints = 2
   Class.SubClass.Protocol: 0A.00.00
       endpoint[0].address: 81
           max packet size: 0200
          polling interval: 00
       endpoint[1].address: 01
           max packet size: 0200
          polling interval: 00
Claiming interface...
Setting up interface...

Initialising protocol...
ERROR: Failed to receive handshake response. Result: -7
ERROR: Protocol initialisation failed!

Releasing device interface...
```

Andi-Moser, 2015-04-09:
I'm too trying to flash my SGS5.

You have to re-enter de download mode after every flash or print-pit attempt, otherwise you will get the handshake response failure. So maybe you just have to try again.

I will try sshimko's version and see if that works for me.

Andi-Moser, 2015-04-09:
I tried building iamtag's version (https://github.com/iamtag/Heimdall) and everything worked just fine.

see this pull: https://github.com/Benjamin-Dobell/Heimdall/pull/240

Benjamin-Dobell, 2015-04-09:
@Andi-Moser Did you try the latest on master (i.e. built with CMake etc.)?

Because I believe this should be fixed without that pull request.

Andi-Moser, 2015-04-09:
@Benjamin-Dobell I tried it with your master and your v1.4.1 commit. Both didn't really work (I just tried it with the 'normal' make command - I am by far no expert in this area).

If I have time tomorrow I can send you the output of my attempt to download my pit file.

zavorra, 2015-04-22:
@Andi-Moser do you mean you can flash your Galaxy S5?

I have tried both benjamin and iamtag versions but constantly get "ERROR: Protocol initialization failed!" when trying to print-pit, so I have never tried to actually flash something.