Benjamin-Dobell/Heimdall #269: Building with qt4 OSX - workaround for no frontend

Liontooth, 2015-04-03:
Is there a way to use qt4 instead of qt5?

I use macports' qt4-mac. Macports currently does not support parallel installation of qt4 and qt5, and lots of applications still depend on qt4.

As a workaround, I removed heimdall-frontend from CMakeLists.txt (note dependencies):

```
cmake_minimum_required(VERSION 2.8.4)

set(CMAKE_MODULE_PATH
    ${CMAKE_SOURCE_DIR}/cmake
    ${CMAKE_MODULE_PATH})

project(Heimdall)

set(CMAKE_RUNTIME_OUTPUT_DIRECTORY ${CMAKE_BINARY_DIR}/bin)

add_subdirectory(libpit)
add_subdirectory(heimdall)

add_dependencies(heimdall heimdall)
```
and successfully compiled heimdall.

BTW this version still has the same problem as the official release with OSX 10.9.5 and T989:

libusbx: error [darwin_claim_interface] USBInterfaceOpen: another process has device opened for exclusive access

If I unload these:

```
sudo kextunload -b com.devguru.driver.SamsungComposite
sudo kextunload -b com.devguru.driver.SamsungACMData
sudo kextunload -b com.devguru.driver.SamsungACMControl
```

I get a different error (issue #137) :

```
Setting up interface...
libusbx: error [get_endpoints] error getting pipe information for pipe 1: unknown error
libusbx: error [darwin_set_interface_altsetting] could not build endpoint table
ERROR: Setting up interface failed!

Cheers,
David

Benjamin-Dobell, 2015-04-03:
> Is there a way to use qt4 instead of qt5?

No, all future builds will be qt5 only, particularly the upcoming Heimdall 2.0 release (once I finally get a chance to finish it).

> I use macports' qt4-mac. Macports currently does not support parallel installation of qt4 and qt5, and lots of applications still depend on qt4.

I strongly recommend you use [Homebrew](http://brew.sh/) instead of Macports. Simultaneous installation of Qt 4 and Qt 5 is supported using Homebrew in a non-disruptive manner.

> I get a different error (issue #137) :

I'm not seeing this on my Yosemite machine (or in my Lion VM). Perhaps Homebrew is providing a newer version of libusb-1.0 than Macports..

Liontooth, 2015-04-05:
OK, thanks. Homebrew is indeed using libusb 1.0.19, the latest, while macports has 1.0.18. However, after modifying the macports port file and installing 1.0.19, I still get the same issue #137 error.

Heimdall finds the macports version of libusb:

```
$ cmake -G "Unix Makefiles" -DCMAKE_BUILD_TYPE=Release ..
-- The C compiler identification is AppleClang 5.1.0.5030040
-- The CXX compiler identification is AppleClang 5.1.0.5030040
-- Check for working C compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/cc
-- Check for working C compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/c++
-- Check for working CXX compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found libusb: /opt/local/lib/libusb-1.0.a
-- Checking if large (64-bit) file support is available...
-- Checking if large (64-bit) file support is available - yes
-- Configuring done
-- Generating done
-- Build files have been written to: /Users/liontooth/software/git/heimdall/Heimdall-master/build
```

Yet from dtruss it looks to me like heimdall is using the built-in /usr/lib/libusb-1.0.dylib rather than the macports version:

```
Heimdall-master/build/bin$ sudo dtruss ./heimdall print-pit --no-reboot
Heimdall v1.4.1

Copyright (c) 2010-2014 Benjamin Dobell, Glass Echidna
http://www.glassechidna.com.au/

This software is provided free of charge. Copying and redistribution is
encouraged.

If you appreciate this software and you would like to support future
development please consider donating:
http://www.glassechidna.com.au/donate/

SYSCALL(args) 		 = return
open("/usr/lib/dtrace/libdtrace_dyld.dylib\0", 0x0, 0x0)		 = 3 0
pread(0x3, "\312\376\272\276\0", 0x1000, 0x0)		 = 4096 0
pread(0x3, "\317\372\355\376\a\0", 0x1000, 0x1000)		 = 4096 0
fcntl(0x3, 0x3D, 0x7FFF52FB3700)		 = 0 0
mmap(0x10CC6B000, 0x2000, 0x5, 0x12, 0x3, 0x1000)		 = 0x10CC6B000 0
mmap(0x10CC6D000, 0x1000, 0x3, 0x12, 0x3, 0x3000)		 = 0x10CC6D000 0
mmap(0x10CC6E000, 0x2050, 0x1, 0x12, 0x3, 0x4000)		 = 0x10CC6E000 0
close(0x3)		 = 0 0
stat64("/usr/lib/libusb-1.0.dylib\0", 0x7FFF52FB5188, 0x7FFF52FB6020)		 = 0 0
open("/usr/lib/libusb-1.0.dylib\0", 0x0, 0x0)		 = 3 0
pread(0x3, "\317\372\355\376\a\0", 0x1000, 0x0)		 = 4096 0
mmap(0x10CC78000, 0xF000, 0x5, 0x12, 0x3, 0x0)		 = 0x10CC78000 0
mmap(0x10CC87000, 0x1000, 0x3, 0x12, 0x3, 0xF000)		 = 0x10CC87000 0
mmap(0x10CC88000, 0x33C8, 0x1, 0x12, 0x3, 0x10000)		 = 0x10CC88000 0
fcntl(0x3, 0x2C, 0x7FFF52FB33C8)		 = 0 0
close(0x3)		 = 0 0
stat64("/usr/lib/libc++.1.dylib\0", 0x7FFF52FB5188, 0x7FFF52FB6020)		 = 0 0
stat64("/usr/lib/libSystem.B.dylib\0", 0x7FFF52FB5188, 0x7FFF52FB6020)		 = 0 0
stat64("/usr/lib/libobjc.A.dylib\0", 0x7FFF52FB5058, 0x7FFF52FB5EF0)		 = 0 0
stat64("/System/Library/Frameworks/IOKit.framework/Versions/A/IOKit\0", 0x7FFF52FB5058, 0x7FFF52FB5EF0)		 = 0 0
stat64("/System/Library/Frameworks/CoreFoundation.framework/Versions/A/CoreFoundation\0", 0x7FFF52FB5058, 0x7FFF52FB5EF0)		 = 0 0
stat64("/usr/lib/libauto.dylib\0", 0x7FFF52FB4F28, 0x7FFF52FB5DC0)		 = 0 0
stat64("/usr/lib/libc++abi.dylib\0", 0x7FFF52FB4F28, 0x7FFF52FB5DC0)		 = 0 0
stat64("/usr/lib/libDiagnosticMessagesClient.dylib\0", 0x7FFF52FB4E08, 0x7FFF52FB5CA0)		 = 0 0
stat64("/usr/lib/system/libcache.dylib\0", 0x7FFF52FB4878, 0x7FFF52FB5710)		 = 0 0
[... 345 lines cut ...]
[ 1.291136] [00000507] libusb: debug [get_endpoints] interface: 1 pipe 1: dir: 1 number: 1
[ 1.291149] [00000507] libusb: debug [get_endpoints] interface: 1 pipe 2: dir: 0 number: 1
[ 1.291196] [00000507] libusb: debug [darwin_claim_interface] interface opened
Setting up interface...
[ 1.291222] [00000507] libusb: debug [libusb_set_interface_alt_setting] interface 1 altsetting 0
[ 1.327747] [00000507] libusb: debug [darwin_reset_device] darwin/reset_device: checking if device descriptor changed
[ 1.328015] [00000507] libusb: debug [darwin_reset_device] darwin/reset_device: checking if configuration descriptor 0 changed
[ 1.328240] [00000507] libusb: debug [darwin_reset_device] darwin/reset_device: device reset complete
[ 1.328258] [00000507] libusb: debug [get_endpoints] building table of endpoints.
[ 1.328282] [00000507] libusb: error [get_endpoints] error getting pipe information for pipe 1: unknown error (0xe0004061)
[ 1.328547] [00000507] libusb: error [darwin_set_interface_altsetting] could not build endpoint table
ERROR: Setting up interface failed!
Releasing device interface...
[ 1.328584] [00000507] libusb: debug [libusb_release_interface] interface 1

[ 1.328600] [00000507] libusb: debug [libusb_close] 
[ 1.328668] [00000507] libusb: debug [usbi_remove_pollfd] remove fd 8
[ 1.328703] [00000507] libusb: debug [libusb_exit] 
[ 1.328712] [00000507] libusb: debug [libusb_exit] destroying default context
```

Candidate fix: https://github.com/libimobiledevice/libirecovery/pull/14/files?diff=split ([history](http://webcache.googleusercontent.com/search?q=cache:xK_1HZin9yAJ:https://github.com/libimobiledevice/libirecovery/pull/14/files%3Fdiff%3Dsplit+&cd=1&hl=en&ct=clnk)).

Cheers,
David

Benjamin-Dobell, 2015-04-05:
This certainly looks like a bug in libusb-1.0 (or perhaps even lower level), and isn't an issue with Heimdall. Nonetheless, the bug clearly does affect Heimdall, so I'd still like to try debug the issue. However, in order to do so I need to be able to recreate the problem.

#### What are the specs of your Mac?
* Is it a Macbook? If so, which model?
* What version of OS X are you running?
* Are you using the USB port in your Mac or an external USB hub?

#### What Android device(s) are you using?
* What model is the phone? Please be specific.

Liontooth, 2015-04-05:
That would be great, thanks! This is a MacBook Pro 13-inch, Late 2011 2.8 GHz Intel Core i7 Intel HD Graphics 3000 512 MB  running OS X 10.9.5 (13F34). I'm using the USB ports in the Mac. The Android phone is a Samsung Galaxy II S T989 (t-mobile). It started out as a stock release, 4.1.2; I've since flashed it using Odin 2.6 in Parallels to run twrp 2.5 and various ROMs, currently Techniqcality. They all run into the same problem with Heimdall.

If we can formulate what exactly is going wrong we could also report it to libusb; it seems to be affecting a good number of people.

wattsteve, 2016-12-02:
Was there ever any resolution to this issue? I'm running into the same thing.