Benjamin-Dobell/Heimdall #270: The application was unable to start correctly

blakehawkins, 2015-04-08:
Hello,

I have the redistributables - `Microsoft Visual C++ 2012 Redistributable (x64) - 11.0.61030`

When I run `heimdall help` I get the error: `The application was unable to start correctly (0xc0000013). Click OK to close the application.`

Kind regards,

Benjamin-Dobell, 2015-04-08:
What version of Windows are you running?

Also, you need the x86 redistributables, not x64, although I believe they're usually bundled together.

blakehawkins, 2015-04-08:
> What version of Windows are you running?

Windows 7 x64

> Also, you need the x86 redistributables, not x64, although I believe they're usually bundled together.

They aren't bundled together, the download link offers which ones you want. With x86 redistributables the issue is fixed, however I have a new issue now:

    C:\Users\Blake\Desktop\Heimdall Suite>heimdall print-pit --no-reboot
    Heimdall v1.4.0

    Copyright (c) 2010-2013, Benjamin Dobell, Glass Echidna
    http://www.glassechidna.com.au/

    This software is provided free of charge. Copying and redistribution is
    encouraged.

    If you appreciate this software and you would like to support future
    development please consider donating:
    http://www.glassechidna.com.au/donate/

    Initialising connection...
    Detecting device...
    ERROR: Failed to access device. libusb error: -12

Benjamin-Dobell, 2015-04-09:
@blakehawkins You're right, the README definitely could be clearer. The redistributables won't be required for the next major release of Heimdall, so that should simplify the process. I'll also see what I can do about automatically installing the drivers.

The last tip I have for you would be to take a look at the "Performing a Custom Flash with Heimdall Frontend" section of the README. If you have firmware that is not a "Heimdall Firmware Package", then that's the procedure you'll need to follow.

closed 2015-04-09 by Benjamin-Dobell

blakehawkins, 2015-04-09:
Excellent, thanks a lot, and thanks for maintaining this great tool