Benjamin-Dobell/Heimdall #271: /usr/bin/heimdall: No such file or directory

francesco1119, 2015-04-09:
Hello, I have this problem on Debian Wheezy, when I do:

heimdall flash --recovery recovery-clockwork-6.0.4.6-i9300.img 

The output just says:

-su: /usr/bin/heimdall: No such file or directory

The problem is because before I had installled the .deb for Ubuntu. So I did purged hemdall and hemdall-front and installed the hemdall and hemdall-front for debian7 but the problem still persist.
Wha to do?

Benjamin-Dobell, 2015-04-09:
@francesco1119 Hmm, that typically means that you haven't installed the Heimdall (command line) package. Also, specifying "--recovery" like that is old syntax, "--RECOVERY" will probably work, but it depends on your PIT file.

I can't imagine why the packages wouldn't work, but if you really can't get them to install then I'd suggest you try compile from source. The Git repo has fixes for newer devices that haven't been included in a binary release yet, so that might be the best option regardless.

francesco1119, 2015-04-09:
Thank you. Following those instructions: http://wiki.cyanogenmod.org/w/Install_and_compile_Heimdall#Downloading_and_Installing I cannot find the files:

$ ./autogen.sh
$ ./configure

???

Benjamin-Dobell, 2015-04-09:
That is out of date, please follow the official README:
https://github.com/Benjamin-Dobell/Heimdall/tree/master/Linux

francesco1119, 2015-04-09:
at the link you gave me there are no info about how to compile heimdall. I still cannot use it because of the problem: /usr/bin/heimdall: No such file or directory

Thank you

Benjamin-Dobell, 2015-04-09:
Did you look at the README?

"Appendix B - Installing Heimdall Suite from Source"

InfinitelyManic, 2015-04-09:
You need to look at the /Heimdall-master/Linux/README
Go to the end and follow instructions. You made need to install cmake or install from source.  I built heimdall witj cmake v 2.8.12.2 with success.

Also... you should do a print-pit before trying recovery in order to see what your phone is expecting

mkdir build 
cd build
cmake - DCMAKE_BUILD_TYPE=Release .. 
make

francesco1119, 2015-04-09:
Yeah, thank you. Unfortunately I dont' have "qtbase5-dev" installed on my Debian Wheezy. In order to install it, apparently, I do habe to download this: http://www.qt.io/download-open-source/#section-2 and it's quite hard as whaen you click download it opens a page that should autoinstall and it's not working.

Any idea about how to install "qtbase5-dev". Debian Packages from Debian site are not working

Benjamin-Dobell, 2015-04-09:
@francesco1119 Try https://packages.debian.org/sid/qtbase5-dev

Benjamin-Dobell, 2015-04-09:
@francesco1119 What is the error you're getting installing qtbase5-dev?

InfinitelyManic, 2015-04-09:
Did you try? 
$ sudo apt-cache search qtbase-dev
You should see:
qtbase5-dev... 
qtbase5-dev-tools... 
... 
sudo apt-get install qtbase5-dev

francesco1119, 2015-04-09:
# gdebi qt5-default_5.3.2+dfsg-4+b1_amd64.deb Reading package lists... Done
Building dependency tree        
Reading state information... Done
Building data structures... Done 
Building data structures... Done 
This package is uninstallable
Dependency is not satisfiable: qtbase5-dev

and

# apt-get install qtbase5-dev
Reading package lists... Done
Building dependency tree       
Reading state information... Done
E: Unable to locate package qtbase5-dev

Benjamin-Dobell, 2015-04-09:
@francesco1119 
```sudo apt-get update```
then proceed.

If that doesn't work then your system must be pretty messed up, which would probably explain why Heimdall's packages weren't able to install in the first place.

francesco1119, 2015-04-09:
Following this guide too: https://github.com/mcallegari/qlcplus/wiki/Linux-build-Qt5 ... after:

apt-get install g++ make git build-essential qtbase5-dev qtscript5-dev qtmultimedia5-dev fakeroot debhelper devscripts pkg-config
Reading package lists... Done
Building dependency tree       
Reading state information... Done
E: Unable to locate package qtbase5-dev
E: Unable to locate package qtscript5-dev
E: Unable to locate package qtmultimedia5-dev

Something is terribly wrong on my Debian Wheezy

InfinitelyManic, 2015-04-09:
Can you install this? 
sudo apt-get install qt5-default qt5-make

InfinitelyManic, 2015-04-09:
Also I had to do the following on my sgh-t699
heimdall --RECOVERY recovery.img --no-reboot...

Benjamin-Dobell, 2015-04-09:
@francesco1119 As this seems to be a problem with your system and not specifically Heimdall I'm going to close this issue. Feel free to open a another issue (or comment on an existing one) if you run into a bug etc.

closed 2015-04-09 by Benjamin-Dobell

mcdiamondz, 2015-07-14:
Goto http://debgen.simplylinux.ch/ select a mirror and all that applies (make sure you select Stable(wheezy))
Generate source.list -> copy the generated repository source.list and paste it in /etc/apt/sources.list
run sudo apt-get update

now you can install "qtbase5-dev"