Benjamin-Dobell/Heimdall #272: Protocol initialization failed

simernes, 2015-04-16:
I compiled from source from current master but I get this error on an I8190 (golden):

Initialising connection...
Detecting device...
      Manufacturer: "SAMSUNG"
           Product: "SAMSUNG USB DRIVER"
         Serial No: "	"

            length: 18
      device class: 2
               S/N: 3
           VID:PID: 04E8:685D
         bcdDevice: 0100
   iMan:iProd:iSer: 1:2:3
          nb confs: 1

interface[0].altsetting[0]: num endpoints = 1
   Class.SubClass.Protocol: 02.02.01
       endpoint[0].address: 83
           max packet size: 0010
          polling interval: 09

interface[1].altsetting[0]: num endpoints = 2
   Class.SubClass.Protocol: 0A.00.00
       endpoint[0].address: 81
           max packet size: 0200
          polling interval: 00
       endpoint[1].address: 02
           max packet size: 0200
          polling interval: 00
Claiming interface...
Attempt failed. Detaching driver...
Claiming interface again...
Setting up interface...

Initialising protocol...
ERROR: Failed to receive handshake response. Result: -7
ERROR: Protocol initialisation failed!

Releasing device interface...
Re-attaching kernel driver...

It seems to be a common error no? But it's hard to find a real solution. Been struggling with this for a while. Debugging is enabled, adb works, already have a custom recovery installed and CM.

Yajo, 2015-05-08:
The same happens to me, using heimdall-1.4.1-2.fc21.x86_64 with a GT-S6500:

```
heimdall print-pit --verbose --no-reboot --resume
Heimdall v1.4.1

Copyright (c) 2010-2014 Benjamin Dobell, Glass Echidna
http://www.glassechidna.com.au/

This software is provided free of charge. Copying and redistribution is
encouraged.

If you appreciate this software and you would like to support future
development please consider donating:
http://www.glassechidna.com.au/donate/

Initialising connection...
Detecting device...
      Manufacturer: "Sasmsung"
           Product: "MSM7x27"

            length: 18
      device class: 2
               S/N: 0
           VID:PID: 04E8:685D
         bcdDevice: 0100
   iMan:iProd:iSer: 1:2:0
          nb confs: 1

interface[0].altsetting[0]: num endpoints = 1
   Class.SubClass.Protocol: 02.02.01
       endpoint[0].address: 82
           max packet size: 0010
          polling interval: 09

interface[1].altsetting[0]: num endpoints = 2
   Class.SubClass.Protocol: 0A.00.00
       endpoint[0].address: 81
           max packet size: 0200
          polling interval: 00
       endpoint[1].address: 01
           max packet size: 0200
          polling interval: 00
Claiming interface...
Setting up interface...

Beginning session...
ERROR: libusb error -7 whilst receiving packet. Retrying...
ERROR: libusb error -7 whilst receiving packet. Retrying...
ERROR: libusb error -7 whilst receiving packet. Retrying...
ERROR: libusb error -7 whilst receiving packet. Retrying...
ERROR: libusb error -7 whilst receiving packet.
Releasing device interface...
```

gozes, 2015-07-01:
I am having the same issue with a Note  4(trltetmo)