Benjamin-Dobell/Heimdall #276: Heimdall fails to download  pit file

gozes, 2015-07-02:
I am trying to use the version 1.4.1 from the Arch aur and when I run sudo heimdall flash --RECOVERY twrp-2.8.7.0-trltetmo.img --no-reboot I get the following error

Initialising connection...
Detecting device...
Claiming interface...
Attempt failed. Detaching driver...
Claiming interface again...
Setting up interface...

Initialising protocol...
Protocol initialisation successful.

Beginning session...

Some devices may take up to 2 minutes to respond.
Please be patient!

Session begun.

Downloading device's PIT file...
ERROR: Failed to send request to end PIT file transfer!
ERROR: Failed to download PIT file!
Ending session...
ERROR: Failed to send end session packet!
Releasing device interface...
Re-attaching kernel driver...
I am trying to flash a recovery on to a Note 4(trltetmo) on the stock 4.4.4 and I am on Arch Linux

duese, 2015-07-02:
Try [heimdall-git](https://aur.archlinux.org/packages/heimdall-git/). I had the same problem here, with the git package it worked. BTW: My phone is the klte.

rubix1138, 2015-07-27:
I am getting the exact same error trying to flash a S5 (klte) on Windows 8.1.  I'm not downloading Visual Studio just to get this to work.

Benjamin-Dobell, 2015-07-27:
@rubix1138 Apparently you're not bothered reading the README either. You don't need Visual Studio, you need MinGW-W64 and to follow the instructions.

Anyway, if you can't even be bothered putting in the effort to download Visual Studio (or whatever) why is it reasonable to you to come here and expect me to put in the effort to provide support to you?

rubix1138, 2015-07-27:
Wow, nice attitude.  I just wanted to update my repair tool, and since the cyanogenmod wiki suggested using this tool, I tried it.  I googled the error message and landed here.  Since the only fix is compiling code, I decided to pass.  No, I didn't read your README because this thread confirmed it's probably a bug.  It took me all of 5 minutes to download Odin, get connected, and update my repair partition.

djevertguzman, 2015-07-27:
It's not even that hard to compile code as everything is done for you .

Benjamin-Dobell, 2015-10-08:
@gozes, or anyone else with a Note 4, have you tried building master to check whether or not this is still present?

spikebike, 2015-10-10:
Still doesn't work on note 4.  I git clone'd tonight and I still can't print-pit:

```
$ git clone https://github.com/Benjamin-Dobell/Heimdall
$ cd Heimdall/heimdall
$ mkdir build
# add set(CMAKE_MODULE_PATH ${CMAKE_MODULE_PATH} "/home/me/src/Heimdall/cmake")
# to CMakeLists.txt
$ cd build
$ cmake -DCMAKE_BUILD_TYPE=Release ..
-- Found libusb: /usr/lib/x86_64-linux-gnu/libusb-1.0.so  
-- Checking if large (64-bit) file support is available...
-- Checking if large (64-bit) file support is available - yes
-- Configuring done
-- Generating done
-- Build files have been written to: /home/bill/src/Heimdall/heimdall/build
$ make
Scanning dependencies of target heimdall
[  7%] Building CXX object CMakeFiles/heimdall.dir/source/Arguments.cpp.o
[ 15%] Building CXX object CMakeFiles/heimdall.dir/source/BridgeManager.cpp.o
[ 23%] Building CXX object CMakeFiles/heimdall.dir/source/ClosePcScreenAction.cpp.o
[ 30%] Building CXX object CMakeFiles/heimdall.dir/source/DetectAction.cpp.o
[ 38%] Building CXX object CMakeFiles/heimdall.dir/source/DownloadPitAction.cpp.o
[ 46%] Building CXX object CMakeFiles/heimdall.dir/source/FlashAction.cpp.o
[ 53%] Building CXX object CMakeFiles/heimdall.dir/source/HelpAction.cpp.o
[ 61%] Building CXX object CMakeFiles/heimdall.dir/source/InfoAction.cpp.o
[ 69%] Building CXX object CMakeFiles/heimdall.dir/source/Interface.cpp.o
[ 76%] Building CXX object CMakeFiles/heimdall.dir/source/main.cpp.o
[ 84%] Building CXX object CMakeFiles/heimdall.dir/source/PrintPitAction.cpp.o
[ 92%] Building CXX object CMakeFiles/heimdall.dir/source/Utility.cpp.o
[100%] Building CXX object CMakeFiles/heimdall.dir/source/VersionAction.cpp.o
Linking CXX executable heimdall
100%] Built target heimdall

The basics work:
# ./heimdall detect
Device detected

# ./heimdall print-pit
Heimdall v1.4.1

Copyright (c) 2010-2014 Benjamin Dobell, Glass Echidna
http://www.glassechidna.com.au/

This software is provided free of charge. Copying and redistribution is
encouraged.

If you appreciate this software and you would like to support future
development please consider donating:
http://www.glassechidna.com.au/donate/

Initialising connection...
Detecting device...
Claiming interface...
Attempt failed. Detaching driver...
Claiming interface again...
Setting up interface...

Initialising protocol...
ERROR: Protocol initialisation failed!

Releasing device interface...
Re-attaching kernel driver...
```

Benjamin-Dobell, 2015-10-10:
@spikebike What is your Linux kernel version?

Also, are there any updates available for your system? In particular kernel updates and/or libusb-1.0?

    ERROR: Protocol initialisation failed!

Typically, this means that something much lower level than Heimdall failed. Could you please create a gist with the output of ```heimdall print-pit --usb-log-level debug```?

spikebike, 2015-10-10:
I'm completely up to date:
```
# lsb_release -d
Description:	Ubuntu 14.04.3 LTS
# apt-get upgrade
<snip>
# apt-get upgrade
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Calculating upgrade... Done
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
~/src/Heimdall/heimdall/build# ./heimdall print-pit --usb-log-level debug
Heimdall v1.4.1

Copyright (c) 2010-2014 Benjamin Dobell, Glass Echidna
http://www.glassechidna.com.au/

This software is provided free of charge. Copying and redistribution is
encouraged.

If you appreciate this software and you would like to support future
development please consider donating:
http://www.glassechidna.com.au/donate/

Initialising connection...
Detecting device...
[timestamp] [threadID] facility level [function call] <message>
--------------------------------------------------------------------------------
[ 0.001128] [000054d0] libusbx: debug [libusb_get_device_list] 
[ 0.001141] [000054d0] libusbx: debug [discovered_devs_append] need to increase capacity
[ 0.001145] [000054d0] libusbx: debug [libusb_get_device_descriptor] 
[ 0.001146] [000054d0] libusbx: debug [libusb_get_device_descriptor] 
[ 0.001148] [000054d0] libusbx: debug [libusb_get_device_descriptor] 
[ 0.001149] [000054d0] libusbx: debug [libusb_get_device_descriptor] 
[ 0.001150] [000054d0] libusbx: debug [libusb_get_device_descriptor] 
[ 0.001152] [000054d0] libusbx: debug [libusb_get_device_descriptor] 
[ 0.001153] [000054d0] libusbx: debug [libusb_get_device_descriptor] 
[ 0.001156] [000054d0] libusbx: debug [libusb_open] open 3.7
[ 0.001167] [000054d0] libusbx: debug [usbi_add_pollfd] add fd 11 events 4
[ 0.001173] [000054d0] libusbx: debug [libusb_get_device_descriptor] 
[ 0.001175] [000054d0] libusbx: debug [libusb_get_config_descriptor] index 0
Claiming interface...
[ 0.001182] [000054d0] libusbx: debug [libusb_claim_interface] interface 1
Setting up interface...
[ 0.001192] [000054d0] libusbx: debug [libusb_set_interface_alt_setting] interface 1 altsetting 0

Initialising protocol...
[ 0.001630] [000054d0] libusbx: debug [add_to_flying_list] arm timerfd for timeout in 1000ms (first in line)
[ 0.001652] [000054d0] libusbx: debug [submit_bulk_transfer] need 1 urbs for new transfer with length 4
[ 0.001681] [000054d0] libusbx: debug [libusb_handle_events_timeout_completed] doing our own event handling
[ 0.001689] [000054d0] libusbx: debug [handle_events] poll() 4 fds with timeout in 60000ms
[ 0.001700] [000054d0] libusbx: debug [handle_events] poll() returned 1
[ 0.001708] [000054d0] libusbx: debug [reap_for_handle] urb type=3 status=0 transferred=4
[ 0.001714] [000054d0] libusbx: debug [handle_bulk_completion] handling completion status 0 of bulk urb 1/1
[ 0.001721] [000054d0] libusbx: debug [handle_bulk_completion] last URB in transfer --> complete!
[ 0.001727] [000054d0] libusbx: debug [disarm_timerfd] 
[ 0.001733] [000054d0] libusbx: debug [usbi_handle_transfer_completion] transfer 0x1d69fa8 has callback 0x7febab91c7f0
[ 0.001740] [000054d0] libusbx: debug [sync_transfer_cb] actual_length=4
[ 0.001755] [000054d0] libusbx: debug [add_to_flying_list] arm timerfd for timeout in 1000ms (first in line)
[ 0.001762] [000054d0] libusbx: debug [submit_bulk_transfer] need 1 urbs for new transfer with length 7
[ 0.001772] [000054d0] libusbx: debug [libusb_handle_events_timeout_completed] doing our own event handling
[ 0.001778] [000054d0] libusbx: debug [handle_events] poll() 4 fds with timeout in 60000ms
[ 1.001791] [000054d0] libusbx: debug [handle_events] poll() returned 1
[ 1.001803] [000054d0] libusbx: debug [handle_events] timerfd triggered
[ 1.001806] [000054d0] libusbx: debug [libusb_cancel_transfer] 
[ 1.001838] [000054d0] libusbx: debug [disarm_timerfd] 
[ 1.001846] [000054d0] libusbx: debug [libusb_handle_events_timeout_completed] doing our own event handling
[ 1.001849] [000054d0] libusbx: debug [handle_events] poll() 4 fds with timeout in 60000ms
[ 1.001852] [000054d0] libusbx: debug [handle_events] poll() returned 1
[ 1.001855] [000054d0] libusbx: debug [reap_for_handle] urb type=3 status=-2 transferred=0
[ 1.001857] [000054d0] libusbx: debug [handle_bulk_completion] handling completion status -2 of bulk urb 1/1
[ 1.001859] [000054d0] libusbx: debug [handle_bulk_completion] abnormal reap: urb status -2
[ 1.001860] [000054d0] libusbx: debug [handle_bulk_completion] abnormal reap: last URB handled, reporting
[ 1.001861] [000054d0] libusbx: debug [usbi_handle_transfer_cancellation] detected timeout cancellation
[ 1.001863] [000054d0] libusbx: debug [disarm_timerfd] 
[ 1.001865] [000054d0] libusbx: debug [usbi_handle_transfer_completion] transfer 0x1d69fa8 has callback 0x7febab91c7f0
[ 1.001866] [000054d0] libusbx: debug [sync_transfer_cb] actual_length=0
ERROR: Protocol initialisation failed!

Releasing device interface...
[ 1.001884] [000054d0] libusbx: debug [libusb_release_interface] interface 1

[ 1.001895] [000054d0] libusbx: debug [libusb_close] 
[ 1.001900] [000054d0] libusbx: debug [usbi_remove_pollfd] remove fd 11
[ 1.001907] [000054d0] libusbx: debug [libusb_exit] 
[ 1.001909] [000054d0] libusbx: debug [libusb_exit] destroying default context
root@left:/home/bill/src/Heimdall/heimdall/build# 
```
Is there a known distro with a liveCD that works?  I'd be happy to try something like the latest fedora liveCD to see if the kernel/different libUSB fixes the problem.

Unfortunately building from today's git/master didn't help.

fasihahmed, 2015-12-18:
Same problem on my Note4 SM-N910F. Wanted to get rid of TouchWiz.

kynan, 2015-12-22:
This might be the same issue as #228. Try my fix with the udev rule.

agibson-fl, 2015-12-30:
> This might be the same issue as #228. Try my fix with the udev rule.

Creating a udev rule for the samsung devices didn't work for me.  However, Odin worked fine for me.

```
sudo heimdall print-pit --verbose --no-reboot
Heimdall v1.4.1

Copyright (c) 2010-2014 Benjamin Dobell, Glass Echidna
http://www.glassechidna.com.au/

This software is provided free of charge. Copying and redistribution is
encouraged.

If you appreciate this software and you would like to support future
development please consider donating:
http://www.glassechidna.com.au/donate/

Initialising connection...
Detecting device...
      Manufacturer: "Sasmsung"
           Product: "MSM8960"

            length: 18
      device class: 2
               S/N: 0
           VID:PID: 04E8:685D
         bcdDevice: 0100
   iMan:iProd:iSer: 1:2:0
          nb confs: 1

interface[0].altsetting[0]: num endpoints = 1
   Class.SubClass.Protocol: 02.02.01
       endpoint[0].address: 82
           max packet size: 0010
          polling interval: 09

interface[1].altsetting[0]: num endpoints = 2
   Class.SubClass.Protocol: 0A.00.00
       endpoint[0].address: 81
           max packet size: 0200
          polling interval: 00
       endpoint[1].address: 01
           max packet size: 0200
          polling interval: 00
Claiming interface...
Attempt failed. Detaching driver...
Claiming interface again...
Setting up interface...

Initialising protocol...
Protocol initialisation successful.

Beginning session...
WARNING: Empty bulk transfer after sending packet failed. Continuing anyway...

Some devices may take up to 2 minutes to respond.
Please be patient!

WARNING: Empty bulk transfer after sending packet failed. Continuing anyway...
Session begun.

Downloading device's PIT file...
WARNING: Empty bulk transfer after sending packet failed. Continuing anyway...
WARNING: Empty bulk transfer after sending packet failed. Continuing anyway...
WARNING: Empty bulk transfer after sending packet failed. Continuing anyway...
WARNING: Empty bulk transfer after sending packet failed. Continuing anyway...
WARNING: Empty bulk transfer after sending packet failed. Continuing anyway...
WARNING: Empty bulk transfer after sending packet failed. Continuing anyway...
WARNING: Empty bulk transfer after sending packet failed. Continuing anyway...
WARNING: Empty bulk transfer after sending packet failed. Continuing anyway...
WARNING: Empty bulk transfer after sending packet failed. Continuing anyway...
WARNING: Empty bulk transfer after sending packet failed. Continuing anyway...
WARNING: Empty bulk transfer after sending packet failed. Continuing anyway...
WARNING: Empty bulk transfer after sending packet failed. Continuing anyway...
WARNING: Empty bulk transfer after sending packet failed. Continuing anyway...
WARNING: Empty bulk transfer after sending packet failed. Continuing anyway...
WARNING: Empty bulk transfer after sending packet failed. Continuing anyway...
WARNING: Empty bulk transfer after sending packet failed. Continuing anyway...
WARNING: Empty bulk transfer after sending packet failed. Continuing anyway...
WARNING: Empty bulk transfer after sending packet failed. Continuing anyway...
ERROR: libusb error -7 whilst sending bulk transfer. Retrying...
ERROR: libusb error -7 whilst sending bulk transfer. Retrying...
ERROR: libusb error -7 whilst sending bulk transfer. Retrying...
ERROR: libusb error -7 whilst sending bulk transfer. Retrying...
ERROR: libusb error -7 whilst sending bulk transfer. Retrying...
ERROR: libusb error -7 whilst sending bulk transfer.
ERROR: Failed to send request to end PIT file transfer!
ERROR: Failed to download PIT file!
Ending session...
ERROR: libusb error -7 whilst sending bulk transfer. Retrying...
ERROR: libusb error -7 whilst sending bulk transfer. Retrying...
ERROR: libusb error -7 whilst sending bulk transfer. Retrying...
ERROR: libusb error -7 whilst sending bulk transfer. Retrying...
ERROR: libusb error -7 whilst sending bulk transfer. Retrying...
ERROR: libusb error -7 whilst sending bulk transfer.
ERROR: Failed to send end session packet!
Releasing device interface...
Re-attaching kernel driver...
```

viktorradnai, 2017-01-12:
Hi all, I had this problem with my Note 4 as well, but on Debian Linux. I think this is the same issue as #348. Building Heimdall from latest Git source has fixed it.

onlyjob, 2017-12-14:
1.4.2 fixed the problem for _Samsung Galaxy Tab S2 SM-T713_.

worldofpiggy, 2017-12-30:
same problem on a win7 running in virtualbox

kirk86, 2018-03-04:
I have the same exact issue I'm using heimdall verion 1.4.1 and my error message is about handshake error:

```
Initialising connection...
Detecting device...
Claiming interface...
Setting up interface...

Initialising protocol...
ERROR: Failed to send handshake!ERROR: Protocol initialisation failed!
```

The phone is note4 and I'm on OSx. Any ideas how to resolve this?