Benjamin-Dobell/Heimdall #280: is Galaxy s6 edge supported on Linux version of Heimdall ?

benoitne, 2015-07-20:
many thanks

belette

cathalgarvey, 2015-08-25:
Hi, I'm also keen to know if the Galaxy S6 is supported? When I attempt to download-pit I get the following error:

    Detecting device...
    
              length: 18
        device class: 2
                 S/N: 0
             VID:PID: 04E8:685D
           bcdDevice: 021B
      iMan:iProd:iSer: 1:2:0
            nb confs: 1
    
    interface[0].altsetting[0]: num endpoints = 1
      Class.SubClass.Protocol: 02.02.01
         endpoint[0].address: 83
             max packet size: 0010
            polling interval: 09
    
    interface[1].altsetting[0]: num endpoints = 2
     Class.SubClass.Protocol: 0A.00.00
         endpoint[0].address: 81
             max packet size: 0200
            polling interval: 00
         endpoint[1].address: 02
             max packet size: 0200
            polling interval: 00 
    Claiming interface...
    Setting up interface...
    libusb: error [op_set_interface] setintf failed error -1 errno 71
    ERROR: Setting up interface failed!
    Releasing device interface...

..I'm on Linux so Odin isn't an easy alternative, sadly.

benoitne, 2015-09-03:
I confirm to you that S6 Edge is supported as I used it to download my custom ROM / modem... recently

cathalgarvey, 2015-09-03:
I couldn't get it to work on S6 (not edge), I loaded the bootloader 
image onto the device and used DD to write it to the correct device... O_o

It worked, though!

Benjamin-Dobell, 2015-10-08:
Hmm, conflicting reports regarding whether or not the device is supported.

Error 71 is a low level hardware error. Maybe permission related?

I'm going to close this issue for now. Until we can confirm it's actually an issue supporting Galaxy S6 Edge and not something specific to @cathalgarvey's system.

@cathalgarvey if you get a chance and could try another PC that would be fantastic.

closed 2015-10-08 by Benjamin-Dobell

jerdog, 2015-11-11:
@cathalgarvey you mention that you loaded the bootloader onto the device and used DD to write it... what partition did you write to for the bootloader on the S6? Can you confirm that it worked (ro.bootloader changed)?

cathalgarvey, 2015-11-15:
I cannot recall which device it was, but using the `mount` command may be informative.

It did work, because the bootloader changed. :)