Benjamin-Dobell/Heimdall #282: GT-I9195I: ERROR: RECOVERY upload failed!

lindi2, 2015-07-30:
Summary: Trying to flash recovery.img with heimdall from git on Debian unstable to GT-I9195I (not GT-I9195!) fails with

ERROR: Failed to send file part packet!
ERROR: RECOVERY upload failed!

Details of working setup (Windows):

I can flash the firmware using Kies 3.2.15072_2 on 64-bit Windows 7 if I select "Firmware Upgrade and Initialisation" from the menu. I could not get "Simple USB Logger" to work in my test system at all. However, http://desowin.org/usbpcap/ worked and allowed me to create traffic captures that can be opened in Wireshark. I also made a copy of the firmware files from the temporary directory used by Kies.

Firmware files:

http://lindi.iki.fi/lindi/I9195IXXU1AOB1/AP_I9195IXXU1AOB1_CL4126898_QB3965535_REV00_user_low_ship_MULTI_CERT.tar.md5
http://lindi.iki.fi/lindi/I9195IXXU1AOB1/BL_I9195IXXU1AOB1_CL4126898_QB3965535_REV00_user_low_ship_MULTI_CERT.tar.md5
http://lindi.iki.fi/lindi/I9195IXXU1AOB1/CP_I9195IXXU1AOB1_REV01_CL1887753_QB3587642.tar.md5
http://lindi.iki.fi/lindi/I9195IXXU1AOB1/CSC_NEE_I9195INEE1AOC1_CL4559314_QB4481082_REV00_user_low_ship_MULTI_CERT.tar.md5

Traffic capture:

http://lindi.iki.fi/lindi/I9195IXXU1AOB1/KTU84P.I9195IXXU1AOB1_upgrade_Kies3.2.15072_2_Windows7_x64.pcap

Details of non-working setup (Debian):

I built heimdall from git (commit d0526a3b74a003dfc6f805682693be9173ffcd88) on a Debian unstable amd64 system. I extracted recovery.img from AP_I9195IXXU1AOB1_CL4126898_QB3965535_REV00_user_low_ship_MULTI_CERT.tar.md5 used by Kies and tried the following command:

```
sudo bin/heimdall flash --verbose --no-reboot --RECOVERY recovery.img
```

However, it failed with

```
ERROR: Failed to send file part packet!
ERROR: RECOVERY upload failed!

Ending session...
```
Full output is attached to this issue and also at http://lindi.iki.fi/lindi/I9195IXXU1AOB1/recovery_flash_heimdall_d0526a3b_Debian_x64.txt

USB traffic capture from Linux is available at http://lindi.iki.fi/lindi/I9195IXXU1AOB1/recovery_flash_heimdall_d0526a3b_Debian_x64.pcap

Output of print-pit is available at http://lindi.iki.fi/lindi/I9195IXXU1AOB1/pit.txt

Is there any other information that I should provide?

Benjamin-Dobell, 2015-07-30:
Thank you for providing all those details! I'll definitely take a look this weekend.

lindi2, 2015-07-31:
Just in case it helps I also used Odin 3.10.0.7 from http://forum.xda-developers.com/galaxy-s-advance/help/version-odin3-v3-10-t2859776 to flash the file AP_I9195IXXU1AOB1_CL4126898_QB3965535_REV00_user_low_ship_MULTI_CERT.tar.md5.

USB traffic capture should be bit shorter than with Kies since the device is already in download mode when the it is plugged in: http://lindi.iki.fi/lindi/I9195IXXU1AOB1/AP_flash_odin3.10_Windows7_x64.pcap

lindi2, 2015-08-05:
I just tried to compare the traffic captures and realized that bInterfaceClass is unknown in the Linux capture since the capture did not include the device descriptors. I made a second capture where I carefully started tcpdump before inserting the device to ensure that the descriptors are included, this time the captures look more similar:

http://lindi.iki.fi/lindi/I9195IXXU1AOB1/recovery_flash_heimdall_d0526a3b_Debian_x64.try2.pcap

lindi2, 2015-08-05:
The funniest thing: heimdall works if I run it under valgrind, I guess this must be some timing issue?

Here's trace where heimdall seems to work under Linux:

http://lindi.iki.fi/lindi/I9195IXXU1AOB1/recovery_flash_valgrind_heimdall_d0526a3b_Debian_x64.pcap
http://lindi.iki.fi/lindi/I9195IXXU1AOB1/recovery_flash_valgrind_heimdall_d0526a3b_Debian_x64.txt

lindi2, 2015-08-05:
heimdall seems to work also if I try to run it under strace or ltrace. Also following the advice from http://sourceforge.net/p/libusb/mailman/message/28547530/ I noticed that heimdall works also if use the USB2 port of my laptop instead of the USB3 port.

fabiopaiva, 2015-08-11:
following this issue

thessy, 2015-09-04:
Any news on this issue? Tried all the tips from lindi2 (strace, usb2), nothing worked for me. Running heimdall from ubuntu 14.04 (heimdall-flash 1.4.0-2 amd64)

thylakoid, 2016-01-22:
FYI, I had the same error message [ERROR: Failed to send file part packet!] and resolved the issue by using a different USB cable.

stiix13, 2016-03-13:
Same here,fixed by changing USB cable. Thanks thylakoid.

martyn81, 2016-05-18:
tar header contained an invalid file size what does this mean

lindi2, 2016-05-19:
Martyn81, please open a separate bug report. This bug report is about the "RECOVERY upload failed!" case.

zorteran, 2016-09-10:
Fixed by changing USB cable. Thanks

ngrash, 2017-08-22:
Using a different USB port did the trick for me.

radiocane, 2018-02-22:
With a Samsung SHV-E370K and Ubuntu 16.04 (heimdall 1.4.1) on a Lenovo W530, I've tried "strace heimdall flash --RECOVERY file.img --no-reboot" and it fails.

radiocane, 2018-02-22:
Same with heimdall v1.4.2. I've already tried all USB ports, both USB-2 and USB-3.

gogeccc, 2018-05-05:
usb cable... Can't believe :D

adrian5, 2023-03-20:
Sorry to bump this very old thread. I was at it myself just now with an old phone and scouring through related GitHub issues. For others stumbling upon this I wanted to mention that connecting [via USB2](https://github.com/Benjamin-Dobell/Heimdall/issues/282#issuecomment-128070245) (instead of USB3) got me past the `ERROR: Failed to send file part packet!` as well.