Benjamin-Dobell/Heimdall #288: Flashing Samsung Galaxy Tab Pro 10.1 Wifi (SM-T520) fails with

detly, 2015-08-23:
Device: Samsung Galaxy Tab Pro 10.1 Wifi aka. SM-T520 aka. `picassowifi`
Recovery: `clockworkmodrecovery.6051.picassowifi.img`
Heimdall host OS: Ubuntu 14.04.1 LTS (trusty)
Heimdall version: git `d0526a3`

When I try to flash a recovery to my T520, I get the following error:

```
bin • sudo ./heimdall flash --RECOVERY /home/jason/Desktop/Cyanogen/SM-T520/clockworkmodrecovery.6051.picassowifi.img --no-reboot
Heimdall v1.4.1

Copyright (c) 2010-2014 Benjamin Dobell, Glass Echidna
http://www.glassechidna.com.au/

This software is provided free of charge. Copying and redistribution is
encouraged.

If you appreciate this software and you would like to support future
development please consider donating:
http://www.glassechidna.com.au/donate/

Initialising connection...
Detecting device...
Claiming interface...
Setting up interface...

Initialising protocol...
ERROR: Protocol initialisation failed!

Releasing device interface...
```

The device doesn't indicate that any downloading has taken place, and when I reboot into recovery it's just the stock recovery.

I've successfully flashed a different device before successfully using the same USB port and cable (although an older version of Heimdall, 1.4.0).

Please let me know if there is any debugging info I can provide that might help.

detly, 2015-08-27:
Same result on Windows 7 using 1.4.0.

sub77, 2015-08-29:
finally i flashed my SM-T530 Galaxy Tab 4 with linux heimdall, iam so happy - no more need for a virtualbox with windows on it just for using odin if download mode is needed.

linux distribution is http://linuxbbq.org -  NOX: Lacipecsenye (amd64)

this is what i did:

    sudo apt-get install libudev1 libudev-dev                   \
                         libusb-1.0-0 libusb-1.0-0-dev          \
                         build-essential pkg-config zlib1g-dev  \
                         libusb-dev libqt4-dev qt4-qmake        \
                         autoconf libtool libusb-1.0-0-dev      \
                         qt5-default

    git clone git://github.com/Benjamin-Dobell/Heimdall.git

    cd Heimdall
    mkdir build
    cd build
    cmake -DCMAKE_BUILD_TYPE=Release ..
    make

sub77@grill:~/Heimdall$     mkdir build
sub77@grill:~/Heimdall$     cd build
sub77@grill:~/Heimdall/build$     cmake -DCMAKE_BUILD_TYPE=Release ..
-- The C compiler identification is GNU 5.2.1
-- The CXX compiler identification is GNU 5.2.1
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found libusb: /usr/lib/x86_64-linux-gnu/libusb-1.0.so  
-- Checking if large (64-bit) file support is available...
-- Checking if large (64-bit) file support is available - yes
-- Found ZLIB: /usr/lib/x86_64-linux-gnu/libz.so (found version "1.2.8") 
-- Configuring done
-- Generating done
-- Build files have been written to: /home/sub77/Heimdall/build

sub77@grill:~/Heimdall/build/bin$ ./heimdall flash --RECOVERY ~/recovery.img 
Heimdall v1.4.1

Copyright (c) 2010-2014 Benjamin Dobell, Glass Echidna
http://www.glassechidna.com.au/

This software is provided free of charge. Copying and redistribution is
encouraged.

If you appreciate this software and you would like to support future
development please consider donating:
http://www.glassechidna.com.au/donate/

Initialising connection...
Detecting device...
Claiming interface...
Attempt failed. Detaching driver...
Claiming interface again...
Setting up interface...

Initialising protocol...
Protocol initialisation successful.

Beginning session...

Some devices may take up to 2 minutes to respond.
Please be patient!

Session begun.

Downloading device's PIT file...
PIT file download successful.

Uploading RECOVERY
100%
RECOVERY upload successful

Ending session...
Rebooting device...
Releasing device interface...
Re-attaching kernel driver...

rbu, 2015-09-01:
I had the same issue. After reading [http://forum.xda-developers.com/galaxy-tab-pro-12-10-8/help/flash-to-sm-t520-tab-pro-10-1-wifi-t3188168](a forum thread on this problem) I tried with a different USB port on the same machine. Two out of three ports did not work.

cpfeiffer, 2015-09-15:
I tried 1.4.0 on different USB ports to no avail. Compiled current 1.4.1 and it worked at once.

Benjamin-Dobell, 2015-10-08:
Closing as reported as fixed on master.

closed 2015-10-08 by Benjamin-Dobell

calebunleashed, 2016-03-01:
This happened for me on v1.4.1. There's a few ways to fix it:

1. Ensure you have the latest & greatest version of Heimdall
2. Install all this:
`sudo apt-get install build-essential cmake zlib1g-dev qt5-default libusb-1.0-0-dev libgl1-mesa-glx libgl1-mesa-dev`
3. Reboot your computer
4. Try connecting on a different USB port - 3 of my USB ports didn't work, and one did!

danballance, 2016-03-07:
Trying a different USB port worked for me, after installing from source on Arch from here:

https://aur.archlinux.org/packages/heimdall-git/