Benjamin-Dobell/Heimdall #290: Won't compile at OS X 10.11

tiagobugarin, 2015-09-23:
I am testing OS X 10.11 El Capitan GM and while trying to compile Heimdall I got this output for `make`:

```
$ make
Scanning dependencies of target pit
[  6%] Building CXX object libpit/CMakeFiles/pit.dir/source/libpit.cpp.o
[ 12%] Linking CXX static library libpit.a
[ 12%] Built target pit
Scanning dependencies of target heimdall
[ 18%] Building CXX object heimdall/CMakeFiles/heimdall.dir/source/Arguments.cpp.o
[ 25%] Building CXX object heimdall/CMakeFiles/heimdall.dir/source/BridgeManager.cpp.o
[ 31%] Building CXX object heimdall/CMakeFiles/heimdall.dir/source/ClosePcScreenAction.cpp.o
[ 37%] Building CXX object heimdall/CMakeFiles/heimdall.dir/source/DetectAction.cpp.o
[ 43%] Building CXX object heimdall/CMakeFiles/heimdall.dir/source/DownloadPitAction.cpp.o
[ 50%] Building CXX object heimdall/CMakeFiles/heimdall.dir/source/FlashAction.cpp.o
[ 56%] Building CXX object heimdall/CMakeFiles/heimdall.dir/source/HelpAction.cpp.o
[ 62%] Building CXX object heimdall/CMakeFiles/heimdall.dir/source/InfoAction.cpp.o
[ 68%] Building CXX object heimdall/CMakeFiles/heimdall.dir/source/Interface.cpp.o
[ 75%] Building CXX object heimdall/CMakeFiles/heimdall.dir/source/main.cpp.o
[ 81%] Building CXX object heimdall/CMakeFiles/heimdall.dir/source/PrintPitAction.cpp.o
[ 87%] Building CXX object heimdall/CMakeFiles/heimdall.dir/source/Utility.cpp.o
[ 93%] Building CXX object heimdall/CMakeFiles/heimdall.dir/source/VersionAction.cpp.o
[100%] Linking CXX executable ../bin/heimdall
ld: library not found for -lusb-1.0
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [bin/heimdall] Error 1
make[1]: *** [heimdall/CMakeFiles/heimdall.dir/all] Error 2
make: *** [all] Error 2
```

Also, i made a little modification at the original to remove the frontend subdirectory as I just need to compile the CLI

Benjamin-Dobell, 2015-10-08:
I could be wrong, but this doesn't looks like an El Capitan issue. It seems as though you haven't followed the README, specifically libusb was not installed from Homebrew.

tiagobugarin, 2015-10-08:
libusb was installed but as I am not anymore with OS X 10.11 GM I will close this issue and try to compile again soon.

closed 2015-10-08 by tiagobugarin

vgeorge, 2016-05-25:
I have libusb installed as described at the instructions and I got the same error on OSX 10.11.4.

[image]

Terry-Weymouth, 2017-09-22:
Mac OsX 10.11.6.
Code from github (commit 9bcc42da350bd2e1766980bcb77d806a82d56a1d) - today
Did: 
  brew install libusb qt5 cmake - all worked (strange comments for qt5, but seems to be ok)
  mkdir build
  cd build
  cmake -G "Unix Makefiles" -DCMAKE_BUILD_TYPE=Release -
DQt5Widgets_DIR=/usr/local/opt/qt5/lib/cmake/Qt5Widgets ..
All without apparent error...
but make fails with "library not found for -lPRIVATE"
at this ld -demangle -dynamic -arch x86_64 -macosx_version_min 10.11.0 -o ../bin/heimdall ...
  (see complete statement at end)

Seems that the problem (by process of elimination) is related to 
-lPRIVATE /usr/local/lib/libusb-1.0.a which is a symbolic link to 
/usr/local/Cellar/libusb/1.0.21/lib/libusb-1.0.a

I thought the link might be a problem, so pulled the command from the trace and rewrote it with the direct page and reissued the ld command, and got the same error: specifically - 
$ cd heimdall
$ ld -v -demangle -dynamic -arch x86_64 -macosx_version_min 10.11.0 -o ../bin/heimdall -search_paths_first -headerpad_max_install_names CMakeFiles/heimdall.dir/source/Arguments.cpp.o CMakeFiles/heimdall.dir/source/BridgeManager.cpp.o CMakeFiles/heimdall.dir/source/ClosePcScreenAction.cpp.o CMakeFiles/heimdall.dir/source/DetectAction.cpp.o CMakeFiles/heimdall.dir/source/DownloadPitAction.cpp.o CMakeFiles/heimdall.dir/source/FlashAction.cpp.o CMakeFiles/heimdall.dir/source/HelpAction.cpp.o CMakeFiles/heimdall.dir/source/InfoAction.cpp.o CMakeFiles/heimdall.dir/source/Interface.cpp.o CMakeFiles/heimdall.dir/source/main.cpp.o CMakeFiles/heimdall.dir/source/PrintPitAction.cpp.o CMakeFiles/heimdall.dir/source/Utility.cpp.o CMakeFiles/heimdall.dir/source/VersionAction.cpp.o -lPRIVATE ../libpit/libpit.a -lPRIVATE /usr/local/Cellar/libusb/1.0.21/lib/libusb-1.0.a -lc++ -lSystem /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/../lib/clang/8.0.0/lib/darwin/libclang_rt.osx.a
--- output ---
@(#)PROGRAM:ld  PROJECT:ld64-274.2
configured to support archs: armv6 armv7 armv7s arm64 i386 x86_64 x86_64h armv6m armv7k armv7m armv7em (tvOS)
Library search paths:
	/usr/lib
	/usr/local/lib
Framework search paths:
	/Library/Frameworks/
	/System/Library/Frameworks/
ld: library not found for -lPRIVATE

but 
$ ls -la ../libpit/libpit.a
-rw-r--r--  1 weymouth  staff  8704 Sep 21 20:23 ../libpit/libpit.a
and
$ ls -la /usr/local/Cellar/libusb/1.0.21/lib/libusb-1.0.a
-r--r--r--  1 weymouth  admin  108160 Oct 25  2016 /usr/local/Cellar/libusb/1.0.21/lib/libusb-1.0.a

Suggestions?