Benjamin-Dobell/Heimdall #291: Installation fails on OS X El Capitan

rizkysyazuli, 2015-09-26:
First i got this warning message when i tried to install it.

[image]

But i continue anyway, which got me this:

[image]

Let me know if you need more info.

dgerosa, 2015-10-02:
Same thing here

yagomont, 2015-10-02:
Can't you see, that it is incompatible?

thy1985, 2015-10-02:
Thing is that heimdall uses an improper location to place its Library and binary during installation:
/usr/bin
and
/usr/lib

When upgrading from a previous version to El Capitan the files are quarantined:
```
/Library/SystemMigration/History/Migration-9BB03294-CD3B-4CF5-9613-5A3B69DBCE14/QuarantineRoot//usr/bin/heimdall
/Library/SystemMigration/History/Migration-9BB03294-CD3B-4CF5-9613-5A3B69DBCE14/QuarantineRoot//usr/lib/libusb-1.0.dylib
```

As they in fact should not lurk around there. A more proper location is /usr/local/ (or older /opt/local) prefix. This is what Apple also recommends.

I didn't check it so far but my guts feeling is that the installer is probably trying to write in those directories which is not possible anymore (straight forward) even with elevated privileges.

That means  you probably need to wait for the developer to make the installer point to a more suitable location or you copy the files from quarantine to somewhere else (for cli only use) [NOTE: The UUID in the path is for sure a different on your systems ;)]

Benjamin-Dobell, 2015-10-02:
> As they in fact should not lurk around there. A more proper location is /usr/local/ (or older /opt/local) prefix. This is what Apple also recommends.

It might be what Apple recommends *now*, but it's certainly not how Unix filesystems have ever worked in the past. ```/usr/bin``` and ```/usr/lib``` are with out a doubt the correct place for this kind of software. If Apple are stating otherwise, they're blatantly wrong. Do you have a reference?

```/usr/local``` is for *locally* built software i.e. software you've compiled yourself. So when you make Heimdall from source code it should go in ```/usr/local```. ```/bin``` and ```/lib``` on the other hand are system directories, these are reserved for use by the OS maintainer, in this case Apple. ```/usr/*``` on the other hand is for user installed, but third-party vendor built/maintained software.

The reason Heimdall doesn't work on El Capitan is because Apple require the binaries, kexts and installer to be signed. I (via my company Glass Echidna) have the necessary signing certificates to sign all of the above (including the kext). It's just that the last officially packaged releases of Heimdall were literally made years ago, well before El Capitan was around,

When I finally get a chance I would love to make a new release - there's just a lot of poorly supported devices at the moment, so I don't feel comfortable doing so. The reason the devices are poorly supported is because I don't have access to the vast majority of Samsung's more recent devices (they release far too many variants each year).

Anyway, when I get a chance, hopefully soon, I'll release an installer for kext only - because that's something most people won't be able to build and run on their on own system.

thy1985, 2015-10-02:
Hi Benjamin

Yay, sorry. I did not mean to offend you/anyone. Sorry if this sounded a bit hash!! Your work on heimdall is really much appreciated!
And it works fine on the El C when you cp it to a different location (of whatever name and purpose).
A signed installer that can put the stuff in place is surely a great thing.

Still, Apple's OS is no longer a system that works how "unix has ever worked before"...
I never said that i agree to the path Apple took...
You are right with the semantics for locally built software. :) Nevertheless the location can be used at least for a workaround period.
Cheers,

flybd5, 2015-10-06:
Ouch. Until this is fixed I'm stuck with a Note 10.1 that needs reflashing. :(

Royedc4, 2015-10-08:
thy1985 What files do i have to cp? I didn't have any previous versions... Thanks

Benjamin-Dobell, 2015-10-08:
I'm not running El Capitan, so this is 100% untested, but please give this a shot:

https://bitbucket.org/benjamin_dobell/heimdall/downloads/Heimdall-1.4.1-Unofficial-Signed.dmg

It's a signed copy of Heimdall CLI (and the corresponding kext), version "whatever was sitting in my source directory" :stuck_out_tongue:

Royedc4, 2015-10-08:
Successfully installed on El Capitan ;)

Benjamin-Dobell, 2015-10-08:
Great to hear.

Benjamin-Dobell, 2015-10-08:
Until I get around to making an official release, if you want a version other than "whatever was sitting in my source directory", please install the package above (as you'll need the signed kext installed by a signed installed). Then you can go and build any other version of Heimdall from source.

closed 2015-10-08 by Benjamin-Dobell

Roman2K, 2015-10-09:
Sorry, I tried to install `Heimdall-1.4.1-Unofficial-Signed.dmg` in El Capitan and it failed:

[image]
And the only relevant error in system.log:

```
Oct  9 16:52:30 Romans-iMac sandboxd[128] ([874]): mkdir(874) System Policy: deny file-write-create /usr/local
```

I tried to create `/usr/local` by hand but it won't either:

```shell
~$ sudo mkdir /usr/local
Password:
mkdir: /usr/local: Operation not permitted
```

Which to me is a good thing but I need Heimdall :confused:

Benjamin-Dobell, 2015-10-09:
Does the "Allow apps downloaded from:" setting still exist?

[image]

If so, make sure you've enabled "and identified developers".

Everything I've read and seen (WWDC talks) explicitly state that ```/usr/local``` is still the correct location for third-party (non app store) software. ```/usr/local``` is specifically an exception in ```/System/Library/Sandbox/rootless.conf``` which says that SIP is disabled. The gory details can be found on stack overflow:

http://apple.stackexchange.com/questions/193368/what-is-the-rootless-feature-in-el-capitan-really

Roman2K, 2015-10-09:
Thanks for the details about `/usr/local`. And the setting still exists:
[image]

Benjamin-Dobell, 2015-10-09:
Sp as far as I can tell Apple are publicly like:

> No way! We're not preventing users from installing third-party software. You can still write to "/usr/local"

_..in private_:

> ... if it existed. Muahahaha!

As far as I can tell, fresh installs of El Capitan have an exception to allow software to be installed and run in ```/usr/local```, it's just that the directory doesn't exist... and a user can't even create it! Not even root.

The only way to create it is by pulling messing around with the system in recovery mode e.g.

http://digitizor.com/2015/10/01/fix-homebrew-permissions-osx-el-capitan/

sunny-mittal, 2015-10-19:
I'm running El Capitan and have a /usr/local directory...I'm not sure is the issue people are having but your unofficial installer worked fine for me. Thanks!

traveller1, 2015-10-22:
Thanks, 1.4.1 worked for me. mba 10.11.1

coleHafner, 2015-11-01:
The unofficial Heimdall-1.4.1-Unofficial-Signed.dmg worked for me! Thank you! 

https://bitbucket.org/benjamin_dobell/heimdall/downloads/Heimdall-1.4.1-Unofficial-Signed.dmg

sleepingscorpion, 2015-12-02:
The 'Heimdall-1.4.1-Unofficial-Signed.dmg' worked fine for me too on El C. Cheers!

kevinhooke, 2015-12-06:
Heimdall-1.4.1-Unofficial-Signed.dmg worked for me too on OS X 10.11.1

thalesmello, 2015-12-07:
+1: maybe put this on the home page of the project? Was a little bit hard to find this thread.

gslender, 2015-12-10:
+1: yep, needed this unofficial signed release too for me on OS X 10.11.1

kalenjohnson, 2015-12-11:
Just found and successfully installed 1.4.1 :+1: 

Don't know why someone wouldn't have `/usr/local`, seems strange.

matgessel, 2015-12-13:
@Benjamin-Dobell Perhaps this issue should stay open until 1.4.1 is posted on glassechidna. I suspect that people are creating duplicate issues because this issue is filtered out of the default view.

lboix, 2015-12-21:
Thank you very much for providing to us the unofficial Heimdall-1.4.1-Unofficial-Signed.dmg 

https://bitbucket.org/benjamin_dobell/heimdall/downloads/Heimdall-1.4.1-Unofficial-Signed.dmg

Same situation here, and all worked fine with this package!

simsketch, 2015-12-29:
thanks for the link Iboix!

GoncaloPT, 2016-01-05:
Thank you  for providing 1.4.1. 
You have my positive feedback runing el capitan

kivikakk, 2016-01-08:
+1 for working on 10.11. I don't think SIP is responsible for the failure here.

hoanghua, 2016-01-31:
+1 works for me with last El Capitan (10.11.3). Thanks

hiddentao, 2016-02-01:
+1 unofficial installer works for me too (El Capitan 10.11.2). Thanks!

nikhilninawe, 2016-02-01:
It installed for successfully for me too. Thanks!!

mkraft, 2016-02-01:
@Benjamin-Dobell Thanks very much for your build.

Gustto, 2016-02-12:
Can anyone explain me how I install 1.4.1 and use it with the frontend/suite?
I'm having trouble connecting to my Samsung S4 using Heimdall on OSX (10.10.5) and hope 1.4.1 will help me. 
Thanks all!

rahulvenus, 2016-02-12:
Thank you for your build ....worked for me too...

grizmt, 2016-02-17:
Installation says successful, however the front-end is not in my applications folder. Running 10.11.3, any insight?

Benjamin-Dobell, 2016-02-17:
@grizmt Sorry, it's just the command line tool in that package.

grizmt, 2016-02-17:
@Benjamin-Dobell Oh right, I didn't realise. Thanks!

QueenAdrienne, 2016-02-24:
Thanks @Benjamin-Dobell appreciate the information and the working file. It's clear that Apple is moving toward restricting their environment to the point that unless Apple can monetize it, it's not happening. The Apple/iphone environment is being insulated.I sent my iphone 6 back. Can't download ring tones, Can't install non itunes apps,etc. I think my next laptop will be either Android, Unix, or god forbid Windows.

FrustratedTechNoob, 2016-02-26:
Thanks for the solution - it installed for me fine (1st time ever i have had an issue, googled it and the first result solved it). Now to the next step of getting my s3 running cyanogen...

jeridgas, 2016-03-05:
Can anyone help? It says the install was successful yet I can't locate 1.4.1 anywhere......any thoughts on this?

jeridgas, 2016-03-05:
@Benjamin-Dobell where do I get the proper frontend package for this version?

ronparchita, 2016-03-22:
I have Mavericks and I have the same problem
At first I downloaded from http://mac.softpedia.com/get/Utilities/Heimdall.shtml v1.4.1
2 clicks, reboot, nothing in Applications at heimdall.
So I see here in this tread another link. I downloaded it, 2 clicks, reboot.
It is the same. Nothing, nada, rien in the applications folder !
Why, I don't understand what happens.
Please, can you help me ?
Please I'd like to know if it's ok with this download only or if I have to download others folders.

Many thanks

R

jonn26, 2016-03-29:
If you want a front-end for the heimdall cli try JOdin3.

https://jodin3.casual-dev.com/

ronparchita, 2016-04-05:
Thank you jonn26

Please, can you tell mi what I have to do with that :+1: 
[image]

Thank you

thelangosta, 2016-04-10:
The Jodin3 site has been down since yesterday. I'm not sure if Adam Outler decided to stop the server since he doesn't seem to support Jodin any more. It's getting harder to find the right tools to restore a bricked samsung. Fastboot would make things much easier.

StevenXL, 2016-05-01:
Installed https://bitbucket.org/benjamin_dobell/heimdall/downloads/Heimdall-1.4.1-Unofficial-Signed.dmg on OS X El Capitan (10.11.4). Worked! Thank you.

KingHebes, 2016-05-13:
This does not work on my version of 10.11.4. That download file is much smaller than the official version from http://glassechidna.com.au/heimdall/ I installed the file that @Benjamin-Dobell put up top in his post and then tried the official version and it still did not install.

mrgame64, 2016-06-07:
The Unsigned installer works on OSX 10.11 here too! Thanks again :)

weareshipco, 2016-06-20:
Installed perfectly on OSX 10.11 here as well. Thank you!

fawyna, 2016-06-20:
Installed perfectly on OSX 10.11 here as well. Thank you!

gcheron, 2016-06-30:
Hi,

" Failed to confirm end of file transfer sequence! ERROR: RECOVERY upload failed! "

I managed to install the unofficial package under El Capitan (thank you very much!) but when it is used with JOdin3 I'm getting an issue when I try to root my Note 4. Heimdall managed to automatically download the PIT with JOdin but fails when using the flash command. I run JOdin3 from a terminal and here is the output:

[...]
[DEBUG]Run Heimdall from DeviceCommunicationProtocol:/usr/local/bin/heimdall flash --PIT /var/folders/zz/zyxvpxvq6csfxvn_n0000000000000/T/CASUALroot-2016-06-30-10.37.13/part.pit --CACHE /var/folders/zz/zyxvpxvq6csfxvn_n0000000000000/T/CASUALroot-2016-06-30-10.37.13/cache.img.ext4 --RECOVERY /var/folders/zz/zyxvpxvq6csfxvn_n0000000000000/T/CASUALroot-2016-06-30-10.37.13/recovery.img
[DEBUG]###executing real-time command: /usr/local/bin/heimdall###
[ERROR]ANALYSIS DETECTED ERROR:Failed to confirm end of file transfer sequence!
[DEBUG]State Change Detected, The new state is: 0
[DEBUG]@stateDisconnected
[DEBUG]Device Removed
[DEBUG]Run Heimdall from DeviceCommunicationProtocol:/usr/local/bin/heimdall detect
[VERBOSE]Heimdall Device detected!
[VERBOSE]Starting: mode:Heimdall
Connected:true

[DEBUG]Run Heimdall from DeviceCommunicationProtocol:/usr/local/bin/heimdall detect
[VERBOSE]Heimdall Device detected!
[DEBUG]State Change Detected, The new state is: 1
[DEBUG]Device Connected
[DEBUG]@stateConnected

Any idea?

BirchardEricAvery, 2016-07-08:
That was awesome, I've never gotten this to work--until now.  

Running OSX 10.11.5 and all I did was blindly copy/ paste to every location mentioned--it worked the very first time.  So the two quarantined files are correct and run perfectly.  

Should on choose to, you can always delete the extra copies, but it works so not going to fix something that isn't broke.  However, if some else reads this try one place at time so as to find the most correct spot.  It's only two files and a few locations which leaves on a few combinations.

ttimz1313, 2016-07-16:
Cool

luizfelipetx, 2016-07-27:
Works for me, 

@ttimz1313  update the link in website. From there is not working.
http://glassechidna.com.au/heimdall/#downloads (1.4.0 no work in el c, 1.4.1 unsg works fine.)

cederom, 2016-09-29:
> It might be what Apple recommends now, but it's certainly not how Unix filesystems have ever worked in the past. /usr/bin and /usr/lib are with out a doubt the correct place for this kind of [...]

Apple OSX/macOS is a BSD Unix OS. You can easily see exactly the same approach in FreeBSD of separation /bin and /usr/bin as the OS BASE from /usr/local USERLAND. This is because base may be booted from another place than userland. Also noone wants to modify base. :-)

matiasnj, 2016-10-08:
Successfully installed also on mac OS Sierra

tutejszy, 2016-11-22:
Confirm successful install 1.4.1 on MacOS 10.12.2 Sierra

papei, 2016-11-26:
I installed the unofficial Heimdall-1.4.1-Unofficial-Signed.dmg with success. However after installing it I went to update my Homebrew and got the following warning.

Warning: Unbrewed dylibs were found in /usr/local/lib.
If you didn't put them there on purpose they could cause problems when building Homebrew formulae, and may need to be deleted.

Unexpected dylibs:
/usr/local/lib/libusb-1.0.dylib

Should I delete libusb-1.0.dylib or leave it there of the time being and see what happens? If I delete, will it have an effect on Heimdall?

Running OSX 10.11.6

psrrao, 2016-12-08:
I'm running macOS 10.12.1, installed Heimdall-1.4.1-Unofficial-Signed.dmg successfully.

Thanks..!!

jeff303, 2017-01-21:
For the folks saying the Heimdall-1.4.1-Unofficial-Signed.dmg installer worked, where do you find the Heimdall Frontend after installing?  I've tried Finder and the locate tool from the command line and am unable to locate any files after it completes.  As someone else noted, it's around 120kb whereas the 1.4.0 dmg file (which doesn't work on Sierra) is 7.3MB, so something is clearly off there.

[25 more replies]