Benjamin-Dobell/Heimdall #292: failed to send request to end pit file transfer

pradeepreddychimmula, 2015-09-26:
i am trying to download my device pit file

log is here

Heimdall v1.4.0

Copyright (c) 2010-2013, Benjamin Dobell, Glass Echidna
http://www.glassechidna.com.au/

This software is provided free of charge. Copying and redistribution is
encouraged.

If you appreciate this software and you would like to support future
development please consider donating:
http://www.glassechidna.com.au/donate/

Initialising connection...
Detecting device...
Claiming interface...
Setting up interface...

Initialising protocol...
Protocol initialisation successful.

Beginning session...

Some devices may take up to 2 minutes to respond.
Please be patient!

Session begun.

Downloading device's PIT file...
ERROR: Failed to send request to end PIT file transfer!
ERROR: Failed to download PIT file!
Ending session...
ERROR: Failed to send end session packet!
Releasing device interface...

spikebike, 2015-10-01:
I've the same problem, I was trying to install a recovery image and it failed at the downloading PIT step.

I'm using ubuntu-14.04 LTS (fully patched) with my T-mobile note 4 (the original, not the newer T3 model).  I get exactly the same error:
'''
Some devices may take up to 2 minutes to respond.
Please be patient!

Session begun.

Downloading device's PIT file...
ERROR: Failed to send request to end PIT file transfer!
ERROR: Failed to download PIT file!
Ending session...
ERROR: Failed to send end session packet!
Releasing device interface...
Re-attaching kernel driver...
'''

I tried compiling heimdall 1.4.1 from git, it was a bit tricky, but it fails in a very similar way:
'''
Initialising connection...
Detecting device...
Claiming interface...
Setting up interface...

Initialising protocol...
ERROR: Protocol initialisation failed!

Both are with my device in recovery mode saying:
ODIN MODE (HIGH SPEED)
PRODUCT NAME: SM-910T
...
Downloading...
Do not turn off target.
'''

yagomont, 2015-10-01:
Try heimdall 1.2,

yagomont, 2015-10-01:
It saved my galaxy s4 lte

spikebike, 2015-10-01:
From where?  I found 1.4.0 here: https://bitbucket.org/benjamin_dobell/heimdall/downloads

v1.1.1, v1.3.0, v1.3.1, v1.3.2, v1.4.0, and v1.4.1 here:
https://github.com/Benjamin-Dobell/Heimdall/releases

Where can I get 1.2?

yagomont, 2015-10-01:
I dont know what happened to 1.2. But i remember it worked.Now I run Odin in a VM to use it, havent used heimdall in a while. Flashes recovery, ROM and it's Done.

yagomont, 2015-10-01:
Might be your kernel, too, heimdall seems not to like liquorix kernel 4.x

Benjamin-Dobell, 2015-10-01:
@yagomont The kernel having an impact seems unlikely. Heimdall communicates with the secondary bootloader. Depending on how you flashed your kernel perhaps you replaced the secondary bootloader at the same time.

@spikebike @yagomont Old releases (including 1.2) can be found here:
https://github.com/Benjamin-Dobell/Heimdall/downloads

Quite a lot has changed since then, I'd be a bit surprised if it worked with anything beyond an S3.

yagomont, 2015-10-01:
I am talking about the linux kernel in my computer.

Benjamin-Dobell, 2015-10-01:
> I am talking about the linux kernel in my computer.
Whoops, sorry! That most certainly could lead to wildly different USB behaviour, libusb-1.0 seems to be a lot more finicky than I'd like.

yagomont, 2015-10-01:
Hey, you can always download the pit file for your phone in xda developers.  Ser if that works.

Benjamin-Dobell, 2015-10-01:
@yagomont That will only work if you are repartitioning... and if Heimdall is having difficulty communicating with your device then I strongly advise against doing that.

spikebike, 2015-10-01:
@Benjamin-Dobell I installed heimdall_1.2.0_amd64.deb, but when I run it:
Heimdall v1.0.2b, Copyright (c) 2010-2011, Benjamin Dobell, Glass Echidna

spikebike, 2015-10-01:
I have a completely stock Note 4 and a collection of ubuntu-14.04 LTS desktops and laptops running a linux 3.13.0-30-lowlatency low latency kernel (the default).  I tried all the versions I could find of heimdall and built a few from source as well.  Any idea on what to try to get pit downloads to work?

Benjamin-Dobell, 2015-10-08:
I'm closing this issue as I'm pretty sure the problem is already resolved if you compile from master.

I know compiling is not ideal, I've opened #295

closed 2015-10-08 by Benjamin-Dobell

seegwen, 2015-10-13:
Hi, thanks much for this software,
Mac OSX (10.8.5) here.
Today I've compiled from master and tried downloading the SM-T533's pit file, but I stille get an error:
```log
Initialising connection...
Detecting device...
Claiming interface...
Setting up interface...

Initialising protocol...
ERROR: Protocol initialisation failed!

Releasing device interface...
```

Any help would be appreciated

cjac, 2016-05-13:
I can repro with code from master.

Germano0, 2016-08-27:
In my experience, moving from 1.4.1 to master 1.4.2 solved the problem

l0b0, 2017-01-09:
Reproducible on a Samsung A3 on commit b6fe7f8535355eb8025f4872efef6cd7d3993db6 (current master).

lebarde, 2017-11-19:
@Germano0 was right: it is solved in 1.4.2.

Especially for Debian users, see **[this comment](https://github.com/Benjamin-Dobell/Heimdall/issues/421#issuecomment-330546807)** to compile Heimdall, because 1.4.2 is not yet in the repositories.

minakitty, 2017-11-27:
@lebarde @Benjamin-Dobell , I feel like that comment should be integrated into the Linux readme, because I was stymied for hours trying to get it to compile on an Ubuntu based system because of that last command.  Maybe later if need be I can throw together a PR for it

c-800, 2018-01-21:
I had the same issue with a Galaxy S5 (SM-G900F) and Windows 7.
Compiling Heimdall 1.4.2 solved the problem.
But it was a disaster to get the make-process run successfully.

It seems, that the recent version of MSYS64 has changed in some behaviours (well, that's just my assumption, because I didn't used msys64 before and there are differences to the readme, which was described as pretty good in some comments I read):
- There was no `make.exe` in `msys64/mingw64/bin`
- The `mingw32-make.exe` in `msys64/mingw64/bin` was not recognized resp. used
- The `cmake` command line option `-DQt5Widgets_DIR` had no effect
- The `cmake` command line option `-G "MSYS Makefiles"` had to be changed to `-G "MinGW Makefiles"`

I had no experience with make files before. Also this is only a memory log (memorandum/minutes from memory) of which actions resulted in progress. So there may be simpler solutions. 
 
My solution was:
(1) Install MSYS2 to default directory: `C:\msys64`
On a Question something like: "2 versions of package xyz: use abc or git" I chose git-version (did not try the other version, so it may work too).

(2) Update package data base (like in the readme):
`Pacman -Syu`

(3) Install packages (like in the readme):
`Pacman -S mingw-w64-x86_64 mingw-w64-x86_64-clang mingw-w64-x86_64-cmake mingw-w64-x86_64-libusb mingw-w64-x86_64-qt5-static make`

(4) Install additional packages:
`pacman -S mingw-w64-x86_64-libwebp mingw-w64-x86_64-jasper mingw-w64-x86_64-dbus`
`pacman -S mingw-w64-x86_64-gcc`
Notes:
- I think the package `mingw-w64-x86_64-gcc` was already installed
- jasper was a missing dependency in the linking process. I don't know, if the others here are really needed
- There may be other additional packages which I forgot. If this is the case, it should be no problem to solve this with some research (Well, I Just don't want to do delete and repeat everything ;-)

(5) Go to directory `[msys64-install-dir]/mingw64/bin/`
If there is no `make.exe`:
Copy and rename `mingw32-make.exe` to `make.exe`

(6) Go the directory were the Heimdall sources are located (I used: `c:\cbuilds\heimdall`)
Open `CMakeLists.txt `and insert:
`set (CMAKE_PREFIX_PATH "C:\\msys64\\mingw64\\qt5-static\\lib\\cmake\\Qt5Widgets")`
Notes:
- The path to the msys64 directory may be adjusted.
- I don't know if it is relevant at which position this line is inserted. I inserted it right after `cmake_minimum_required(...)`

(7) Open Terminal in the heimdall source directory:
```
mkdir build
cd build
cmake -G "MinGW Makefiles" -DCMAKE_BUILD_TYPE=Release ..
make
```

(8) The executable files are located in `[heimdall-source-dir]/build/bin`