Benjamin-Dobell/Heimdall #295: Release 1.4.2

Benjamin-Dobell, 2015-10-08:
There are lots of fixes on master that aren't available in the current official releases.

I had planned to reach 2.0 next (which has a completely overhauled Heimdall Frontend), but admittedly that's a long while off. So an interim bug fix release should be made.

spikebike, 2015-10-10:
I have a note 4 t-mobile (completely stock), and ubuntu 14.04.

I ran:
'''
$ git clone https://github.com/Benjamin-Dobell/Heimdall
$ cd Heimdall/heimdall
$ mkdir build
# add set(CMAKE_MODULE_PATH ${CMAKE_MODULE_PATH} "/home/me/src/Heimdall/cmake")
# to CMakeLists.txt
$ cd build
$ cmake -DCMAKE_BUILD_TYPE=Release ..
-- Found libusb: /usr/lib/x86_64-linux-gnu/libusb-1.0.so  
-- Checking if large (64-bit) file support is available...
-- Checking if large (64-bit) file support is available - yes
-- Configuring done
-- Generating done
-- Build files have been written to: /home/bill/src/Heimdall/heimdall/build
$ make
Scanning dependencies of target heimdall
[  7%] Building CXX object CMakeFiles/heimdall.dir/source/Arguments.cpp.o
[ 15%] Building CXX object CMakeFiles/heimdall.dir/source/BridgeManager.cpp.o
[ 23%] Building CXX object CMakeFiles/heimdall.dir/source/ClosePcScreenAction.cpp.o
[ 30%] Building CXX object CMakeFiles/heimdall.dir/source/DetectAction.cpp.o
[ 38%] Building CXX object CMakeFiles/heimdall.dir/source/DownloadPitAction.cpp.o
[ 46%] Building CXX object CMakeFiles/heimdall.dir/source/FlashAction.cpp.o
[ 53%] Building CXX object CMakeFiles/heimdall.dir/source/HelpAction.cpp.o
[ 61%] Building CXX object CMakeFiles/heimdall.dir/source/InfoAction.cpp.o
[ 69%] Building CXX object CMakeFiles/heimdall.dir/source/Interface.cpp.o
[ 76%] Building CXX object CMakeFiles/heimdall.dir/source/main.cpp.o
[ 84%] Building CXX object CMakeFiles/heimdall.dir/source/PrintPitAction.cpp.o
[ 92%] Building CXX object CMakeFiles/heimdall.dir/source/Utility.cpp.o
[100%] Building CXX object CMakeFiles/heimdall.dir/source/VersionAction.cpp.o
Linking CXX executable heimdall
100%] Built target heimdall

The basics work:
# ./heimdall detect
Device detected

# ./heimdall print-pit
Heimdall v1.4.1

Copyright (c) 2010-2014 Benjamin Dobell, Glass Echidna
http://www.glassechidna.com.au/

This software is provided free of charge. Copying and redistribution is
encouraged.

If you appreciate this software and you would like to support future
development please consider donating:
http://www.glassechidna.com.au/donate/

Initialising connection...
Detecting device...
Claiming interface...
Attempt failed. Detaching driver...
Claiming interface again...
Setting up interface...

Initialising protocol...
ERROR: Protocol initialisation failed!

Releasing device interface...
Re-attaching kernel driver...
'''

Benjamin-Dobell, 2015-10-10:
Err, why did you post that here? Please look to see if the same issue exists elsewhere, if not, create a new issue.

spikebike, 2015-10-10:
Sorry, you closed that ticket and said you thought it was closed in master and pointed here.

I'll open a new ticket.

seitzbg, 2015-11-02:
Can you release a windows 1.4.2 ASAP as I have the bug closed in #292 and do not have a build that works?

TIA,

Pauldb8, 2016-04-09:
Hey, i'm still having the #292 issue.. It would really help me if you could give the latest binary as I can't compile it on my computer because I don't have the appropriate tools. Thank you very much :).

jonassmedegaard, 2016-07-07:
any news on your release plans?

abs-zero-ian, 2017-09-23:
For anyone wanting to use this that has arrived here following the failure in the Linux repo versions, the copy here on GIT works - I'm no expert in GIT or compiling code, but the instructions are clear enough right at the bottom of the Linux install instructions readme and it worked first go on my Samsung Galaxy S5 (klte), which is now running LineageOS.

Many thanks to @Benjamin-Dobell and everyone else who made this work. :-)

c-800, 2018-01-21:
I had the same issue with a Galaxy S5 (SM-G900F) and Windows 7.
Compiling Heimdall 1.4.2 solved the problem.
But it was a disaster to get the make-process run successfully.

It seems, that the recent version of MSYS64 has changed in some behaviours (well, that's just my assumption, because I didn't used msys64 before and there are differences to the readme, which was described as pretty good in some comments I read):
- There was no `make.exe` in `msys64/mingw64/bin`
- The `mingw32-make.exe` in `msys64/mingw64/bin` was not recognized resp. used
- The `cmake` command line option `-DQt5Widgets_DIR` had no effect
- The `cmake` command line option `-G "MSYS Makefiles"` had to be changed to `-G "MinGW Makefiles"`

I had no experience with make files before. Also this is only a memory log (memorandum/minutes from memory) of which actions resulted in progress. So there may be simpler solutions. 
 
My solution was:
(1) Install MSYS2 to default directory: `C:\msys64`
On a Question something like: "2 versions of package xyz: use abc or git" I chose git-version (did not try the other version, so it may work too).

(2) Update package data base (like in the readme):
`Pacman -Syu`

(3) Install packages (like in the readme):
`Pacman -S mingw-w64-x86_64 mingw-w64-x86_64-clang mingw-w64-x86_64-cmake mingw-w64-x86_64-libusb mingw-w64-x86_64-qt5-static make`

(4) Install additional packages:
`pacman -S mingw-w64-x86_64-libwebp mingw-w64-x86_64-jasper mingw-w64-x86_64-dbus`
`pacman -S mingw-w64-x86_64-gcc`
Notes:
- I think the package `mingw-w64-x86_64-gcc` was already installed
- jasper was a missing dependency in the linking process. I don't know, if the others here are really needed
- There may be other additional packages which I forgot. If this is the case, it should be no problem to solve this with some research (Well, I Just don't want to do delete and repeat everything ;-)

(5) Go to directory `[msys64-install-dir]/mingw64/bin/`
If there is no `make.exe`:
Copy and rename `mingw32-make.exe` to `make.exe`

(6) Go the directory were the Heimdall sources are located (I used: `c:\cbuilds\heimdall`)
Open `CMakeLists.txt `and insert:
`set (CMAKE_PREFIX_PATH "C:\\msys64\\mingw64\\qt5-static\\lib\\cmake\\Qt5Widgets")`
Notes:
- The path to the msys64 directory may be adjusted.
- I don't know if it is relevant at which position this line is inserted. I inserted it right after `cmake_minimum_required(...)`

(7) Open Terminal in the heimdall source directory:
```
mkdir build
cd build
cmake -G "MinGW Makefiles" -DCMAKE_BUILD_TYPE=Release ..
make
```

(8) The executable files are located in `[heimdall-source-dir]/build/bin`

tothphu, 2018-02-11:
I've built it for Win 10 64bit (with help from @c-800). Here you find the binary:
https://github.com/tothphu/heimdall_build

I join in thanking @Benjamin-Dobell for the excellent work.