Benjamin-Dobell/Heimdall #297: SM-T533 "Protocol initialisation failed"

seegwen, 2015-10-14:
Hi, thanks much for this software,
Mac OSX (10.8.5) here.
Yesterday I've compiled heimdall and heimdall-frontend from master and tried downloading the SM-T533's pit file, but I still get an error (heimdall-frontend output right below).

````log
Initialising connection...
Detecting device...
Claiming interface...
Setting up interface...

Initialising protocol...
ERROR: Protocol initialisation failed!

Releasing device interface...
````
Any help would be appreciated.
And if you need more infos, I'll be glad to provide.
For now I've used Odin in a VM to do the task at hand.

Benjamin-Dobell, 2015-10-14:
Please provide ```--verbose --usb-log-level debug``` output in a gist.

seegwen, 2015-10-14:
Here you are:
[sm-t533 heimdall error](https://gist.github.com/seegwen/74346108cca03888ca04#file-sm-t533-heimdall-error)

I should add that the error I got for debugging is not the same as the one I initially reported.

Benjamin-Dobell, 2015-10-14:
That's a different error than you got last time. I suggest you try a different USB port and/or cable.

seegwen, 2015-10-17:
i tried with every usb port I had available, which amounts to 2, and the cable is a stock, brand new one from samsung. I also tried with a camera usb cable, and got the same error.