Benjamin-Dobell/Heimdall #303: failure on g900f: "ERROR: Protocol initialisation failed!"

morgwai, 2015-10-23:
current master _always_ fails on g900f with "ERROR: Protocol initialisation failed!" message. SaburoJiro's fork _always_ works otoh. in both cases the phone is connected to the same usb port via the same cable. terminal session log of both cases attached below:

```
s5 $ sudo ./heimdall-1.4.1 flash --RECOVERY twrp-2.8.7.0-klte.img --no-reboot
Heimdall v1.4.1

Copyright (c) 2010-2014 Benjamin Dobell, Glass Echidna
http://www.glassechidna.com.au/

This software is provided free of charge. Copying and redistribution is
encouraged.

If you appreciate this software and you would like to support future
development please consider donating:
http://www.glassechidna.com.au/donate/

Initialising connection...
Detecting device...
Claiming interface...
Attempt failed. Detaching driver...
Claiming interface again...
Setting up interface...

Initialising protocol...
ERROR: Protocol initialisation failed!

Releasing device interface...
Re-attaching kernel driver...

s5 $ sudo ./heimdall-s flash --RECOVERY twrp-2.8.7.0-klte.img --no-reboot
Heimdall v1.4.0

Copyright (c) 2010-2013, Benjamin Dobell, Glass Echidna
http://www.glassechidna.com.au/

This software is provided free of charge. Copying and redistribution is
encouraged.

If you appreciate this software and you would like to support future
development please consider donating:
http://www.glassechidna.com.au/donate/

Initialising connection...
Detecting device...
Claiming interface...
Setting up interface...

Initialising protocol...
Protocol initialisation successful.

Beginning session...

Some devices may take up to 2 minutes to respond.
Please be patient!

Session begun.

Downloading device's PIT file...
PIT file data received, read 'empty' packet...
PIT file data receive completed, send kRequestEndTransfer...
PIT file download successful.

Uploading RECOVERY
100%
RECOVERY upload successful

Ending session...
Releasing device interface...

s5 $ 
```

ideaman924, 2015-12-02:
Happens to me as well. Used my Mac. Now I'm going to try on Windows and see what happens..

smurfix, 2016-03-27:
Why is this issue closed? It's still a problem. Happens to me too. No solution other than WIndows which I don't even own?

Benjamin-Dobell, 2016-03-27:
Doesn't the big green "Open" indicate this issue is open, not closed? :wink:

betseg, 2016-06-05:
@SaburoJiro's fork doesn't work for me.
`Heimdall` v1.4.1

Copyright (c) 2010-2014 Benjamin Dobell, Glass Echidna
http://www.glassechidna.com.au/

This software is provided free of charge. Copying and redistribution is
encouraged.

If you appreciate this software and you would like to support future
development please consider donating:
http://www.glassechidna.com.au/donate/

Initialising connection...
Detecting device...
      Manufacturer: "Sasmsung"
           Product: "MSM8960"

            length: 18
      device class: 2
               S/N: 0
           VID:PID: 04E8:685D
         bcdDevice: 0100
   iMan:iProd:iSer: 1:2:0
          nb confs: 1

interface[0].altsetting[0]: num endpoints = 1
   Class.SubClass.Protocol: 02.02.01
       endpoint[0].address: 82
           max packet size: 0010
          polling interval: 09

interface[1].altsetting[0]: num endpoints = 2
   Class.SubClass.Protocol: 0A.00.00
       endpoint[0].address: 81
           max packet size: 0200
          polling interval: 00
       endpoint[1].address: 01
           max packet size: 0200
          polling interval: 00
Claiming interface...
Setting up interface...

Initialising protocol...
ERROR: Failed to receive handshake response. Result: -7
ERROR: Protocol initialisation failed!

Releasing device interface...`

JochenHoch2, 2016-08-08:
Hi @all,

for what it's worth: A self-compiled heimdall from current master (as of 28.7.2016 - sorry, no commit ID...) had absolutely no problems with my SM-G900F (S5 international version/klte). I did have a bad USB port/cable combination (USB device resetting after every open()), but changing cables did the trick.

What I did:

* Compiled heimdall from master on Kubuntu 16.04 by following Linux/README.
* Updated stock ROM to 6.0.1 on phone
* Entered Download Mode
* Downloaded PIT without rebooting, worked correctly
* Printed PIT without rebooting, worked correctly
* Flashed TWRP 3.0.1, worked correctly
* Booted into Recovery

Hope this helps anyone!