Benjamin-Dobell/Heimdall #307: Heimdall + note4/SM-N910T broken with ubuntu 15.10 and fedora 23.

spikebike, 2015-11-06:
I've been trying to get heimdall (from various packages and compiled from source) working on my ubuntu-14.04 box.

There was somediscussion here that maybe it was an older libusb.  So I decided to try the 2 newest linux's I could find and ended up with ubuntu-15.10 (released in Oct) and fedora 23 (released in the last week or so).

I'll include my attempts below.  Any suggestions on which heimdall and which linux I should try to get it to work with my note4/SM-N910T?

So my attempt on fedora 23:
[liveuser@left ~]$ cat /etc/redhat-release
Fedora release 23 (Twenty Three)
# dnf install heimdall
...
  Installing  : heimdall-1.4.1-5.fc23.x86_64                                1/1
  Verifying   : heimdall-1.4.1-5.fc23.x86_64                                1/1

Installed:
  heimdall.x86_64 1.4.1-5.fc23

root@left liveuser]# heimdall print-pit
Heimdall v1.4.1

Copyright (c) 2010-2014 Benjamin Dobell, Glass Echidna
http://www.glassechidna.com.au/

This software is provided free of charge. Copying and redistribution is
encouraged.

If you appreciate this software and you would like to support future
development please consider donating:
http://www.glassechidna.com.au/donate/
Initialising connection...
Detecting device...
Claiming interface...
Attempt failed. Detaching driver...
Claiming interface again...
Setting up interface...

Initialising protocol...
ERROR: Protocol initialisation failed!

Releasing device interface...
Re-attaching kernel driver...

root@left liveuser]#

My attempt on ubuntu-15.10:

root@ubuntu:/home/ubuntu# lsb_release -d
Description:   Ubuntu 15.10
root@ubuntu:/home/ubuntu# apt-get install heimdall-flash
...
Preparing to unpack .../heimdall-flash_1.4.0-2_amd64.deb ...
Unpacking heimdall-flash (1.4.0-2) ...
Setting up heimdall-flash (1.4.0-2) ...
root@ubuntu:/home/ubuntu# heimdall print-pit
Heimdall v1.4.0

Copyright (c) 2010-2013, Benjamin Dobell, Glass Echidna
http://www.glassechidna.com.au/

This software is provided free of charge. Copying and redistribution is
encouraged.

If you appreciate this software and you would like to support future
development please consider donating:
http://www.glassechidna.com.au/donate/

Initialising connection...
Detecting device...
Claiming interface...
Attempt failed. Detaching driver...
Claiming interface again...
Setting up interface...

Initialising protocol...
Protocol initialisation successful.

Beginning session...

Some devices may take up to 2 minutes to respond.
Please be patient!

Session begun.

Downloading device's PIT file...
ERROR: Failed to send request to end PIT file transfer!
ERROR: Failed to download PIT file!
Ending session...
ERROR: Failed to send end session packet!
Releasing device interface...
Re-attaching kernel driver...

root@ubuntu:/home/ubuntu# heimdall detect
Device detected
root@ubuntu:/home/ubuntu#

kuri65536, 2015-11-14:
Me too, Fedora 23 x86_64, heimdall-1.4.1-5.fc23.x86_64.rpm.

fat-tire, 2015-11-17:
FWIW it's working fine for me (odn't get the error above) with Ubuntu 15.10 x86_64.  Built from source-- something like

~~~
git clone git@github.com:SaburoJiro/Heimdall.git
cd Heimdall/libpit
./configure
make
cd ../heimdall
./configure
make
sudo ./heimdall print-pit
~~~

Worked fine.

spikebike, 2015-11-19:
I can confirm that SaburoJiro/Heimdall.git allows print-pit to work on my ubuntu-14.04 machine.

My previous attempts with ubuntu's provided heimdall and building source from Benjamin-Dobell/Heimdall failed.

kynan, 2015-12-22:
This might be the same issue as #228. Try my fix with the udev rule.

fat-tire, 2017-03-15:
Quick note-- SaburuJiro's repo appears to be gone, so I pushed the branch I was using to my repository:

https://github.com/fat-tire/Heimdall

Hope it works for y'all.