Benjamin-Dobell/Heimdall #308: OSX compile error CLOCK_REALTIME

BlueRidgeDog, 2015-11-06:
error: use of undeclared identifier 'CLOCK_REALTIME'.

Any help getting over this.  I have an S5 and need to compile the current version due to the PIT error.

Benjamin-Dobell, 2015-11-06:
Errr, could you please try be more specific?

I assume you're trying to build Heimdall, on __some__ operating system... but Heimdall definitely doesn't make use of ```CLOCK_REALTIME```, at least not directly.

BlueRidgeDog, 2015-11-06:
Sorry for the limited info.  When compiling on OSX 10.10.5 I get two errors with the above description.

I will past them after I fire my laptop up that has the development tools.

BlueRidgeDog, 2015-11-06:
$ make
/Applications/Xcode.app/Contents/Developer/usr/bin/make  all-am
depbase=`echo source/BridgeManager.o | sed 's|[^/]*$|.deps/&|;s|\.o$||'`;\
	/usr/bin/clang++ -DHAVE_CONFIG_H -I.  -I/opt/local/include/libusb-1.0  -I../libpit/Source   -g -O2 -MT source/BridgeManager.o -MD -MP -MF $depbase.Tpo -c -o source/BridgeManager.o source/BridgeManager.cpp &&\
	mv -f $depbase.Tpo $depbase.Po
source/BridgeManager.cpp:545:33: warning: '&&' within '||'
      [-Wlogical-op-parentheses]
        if (!(0 <= rc || pipe_error_ok && rc == LIBUSB_ERROR_PIPE))
                      ~~ ~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~
source/BridgeManager.cpp:545:33: note: place parentheses around the '&&'
      expression to silence this warning
        if (!(0 <= rc || pipe_error_ok && rc == LIBUSB_ERROR_PIPE))
                                       ^
                         (                                       )
source/BridgeManager.cpp:1769:16: error: use of undeclared identifier
      'CLOCK_REALTIME'
        clock_gettime(CLOCK_REALTIME, &ts_0);
                      ^
source/BridgeManager.cpp:1776:18: error: use of undeclared identifier
      'CLOCK_REALTIME'
                        clock_gettime(CLOCK_REALTIME, &ts);
                                      ^
1 warning and 2 errors generated.
make[1]: *** [source/BridgeManager.o] Error 1
make: *** [all] Error 2

Benjamin-Dobell, 2015-11-06:
That output doesn't make any sense to me at all. Which commit are you trying to build?

Could you please start the build over and paste all the commands you've entered and their output in a gist?

BlueRidgeDog, 2015-11-06:
First:

git clone git://github.com/marshray/Heimdall.git

(omitting the install of libpit as it compiled and installed ok)

Changing into the downloaded directory then:

export CC=/usr/bin/clang CXX=/usr/bin/clang++

FighterKite:heimdall cjervin$ ./configure && make
checking for pkg-config... /usr/local/bin/pkg-config
checking pkg-config is at least version 0.9.0... yes
checking for DEPS... yes
checking for udevadm... no
checking build system type... x86_64-apple-darwin14.5.0
checking host system type... x86_64-apple-darwin14.5.0
checking target system type... x86_64-apple-darwin14.5.0
checking for a BSD-compatible install... /usr/bin/install -c
checking whether build environment is sane... yes
checking for a thread-safe mkdir -p... ./install-sh -c -d
checking for gawk... no
checking for mawk... no
checking for nawk... no
checking for awk... awk
checking whether make sets $(MAKE)... yes
checking whether the C++ compiler works... yes
checking for C++ compiler default output file name... a.out
checking for suffix of executables... 
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether we are using the GNU C++ compiler... yes
checking whether /usr/bin/clang++ accepts -g... yes
checking for style of include used by make... GNU
checking dependency style of /usr/bin/clang++... gcc3
checking for gcc... /usr/bin/clang
checking whether we are using the GNU C compiler... yes
checking whether /usr/bin/clang accepts -g... yes
checking for /usr/bin/clang option to accept ISO C89... none needed
checking dependency style of /usr/bin/clang... gcc3
checking for a sed that does not truncate output... /usr/bin/sed
checking for grep that handles long lines and -e... /usr/bin/grep
checking for egrep... /usr/bin/grep -E
checking for fgrep... /usr/bin/grep -F
checking for ld used by /usr/bin/clang... /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/ld
checking if the linker (/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/ld) is GNU ld... no
checking for BSD- or MS-compatible name lister (nm)... /usr/bin/nm
checking the name lister (/usr/bin/nm) interface... BSD nm
checking whether ln -s works... yes
checking the maximum length of command line arguments... 196608
checking whether the shell understands some XSI constructs... yes
checking whether the shell understands "+="... yes
checking for /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/ld option to reload object files... -r
checking for objdump... no
checking how to recognize dependent libraries... pass_all
checking for ar... ar
checking for strip... strip
checking for ranlib... ranlib
checking command to parse /usr/bin/nm output from /usr/bin/clang object... ok
checking for dsymutil... dsymutil
checking for nmedit... nmedit
checking for lipo... lipo
checking for otool... otool
checking for otool64... no
checking for -single_module linker flag... yes
checking for -exported_symbols_list linker flag... yes
checking how to run the C preprocessor... /usr/bin/clang -E
checking for ANSI C header files... yes
checking for sys/types.h... yes
checking for sys/stat.h... yes
[... 46 lines cut ...]
/Applications/Xcode.app/Contents/Developer/usr/bin/make  all-am
depbase=`echo source/BridgeManager.o | sed 's|[^/]*$|.deps/&|;s|\.o$||'`;\
	/usr/bin/clang++ -DHAVE_CONFIG_H -I.  -I/usr/local/Cellar/libusb/1.0.20/include/libusb-1.0 -I../libpit/Source   -g -O2 -MT source/BridgeManager.o -MD -MP -MF $depbase.Tpo -c -o source/BridgeManager.o source/BridgeManager.cpp &&\
	mv -f $depbase.Tpo $depbase.Po
source/BridgeManager.cpp:545:33: warning: '&&' within '||'
      [-Wlogical-op-parentheses]
        if (!(0 <= rc || pipe_error_ok && rc == LIBUSB_ERROR_PIPE))
                      ~~ ~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~
source/BridgeManager.cpp:545:33: note: place parentheses around the '&&'
      expression to silence this warning
        if (!(0 <= rc || pipe_error_ok && rc == LIBUSB_ERROR_PIPE))
                                       ^
                         (                                       )
source/BridgeManager.cpp:1769:16: error: use of undeclared identifier
      'CLOCK_REALTIME'
        clock_gettime(CLOCK_REALTIME, &ts_0);
                      ^
source/BridgeManager.cpp:1776:18: error: use of undeclared identifier
      'CLOCK_REALTIME'
                        clock_gettime(CLOCK_REALTIME, &ts);
                                      ^
1 warning and 2 errors generated.
make[1]: *** [source/BridgeManager.o] Error 1
make: *** [all] Error 2
FighterKite:heimdall cjervin$

Benjamin-Dobell, 2015-11-06:
https://github.com/ __marshray__ /Heimdall... That is not the correct repo. Please clone *this* repo and follow the instructions to build in the README for your platform.

closed 2015-11-06 by Benjamin-Dobell