Benjamin-Dobell/Heimdall #310: compiling error

bodaonline, 2015-11-13:
Hi
never compiled before. Can someone please help with this:

$ cmake -DCMAKE_BUILD_TYPE=Release ..

CMake Error at CMakeLists.txt:9 (find_package):
  By not providing "Findlibusb.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "libusb", but
  CMake did not find one.

  Could not find a package configuration file provided by "libusb" with any
  of the following names:

    libusbConfig.cmake
    libusb-config.cmake

  Add the installation prefix of "libusb" to CMAKE_PREFIX_PATH or set
  "libusb_DIR" to a directory containing one of the above files.  If "libusb"
  provides a separate development package or SDK, be sure it has been
  installed.

-- Configuring incomplete, errors occurred!
See also "/home/zsolt/Desktop/heimdall/build/CMakeFiles/CMakeOutput.log".

Attached my cmakelists.txt
[CMakeLists.txt](https://github.com/Benjamin-Dobell/Heimdall/files/33969/CMakeLists.txt)

I have libusb installed.

Benjamin-Dobell, 2015-11-13:
I'm confused...

`/home/zsolt/Desktop/heimdall/build/CMakeFiles/CMakeOutput.log`

Seems to indicate that you're building from the root. However, the error you're receiving is what I'd expect if you tried to build inside the "heimdall" subdirectory (which won't work).

Is this a fresh clone?

bodaonline, 2015-11-13:
Hi
thx for the quick reply. It is a fresh clone yes. So in which dir should I be in when I build?

Benjamin-Dobell, 2015-11-13:
In the root of the cloned Git repo.

So something like:

```
git clone https://github.com/Benjamin-Dobell/Heimdall.git
cd Heimdall
mkdir build
cd build
cmake -DCMAKE_BUILD_TYPE=Release ..
make
```

You do of course need to make sure you have installed libusb and the other dependencies as described in the appropriate platform-specific README (down the very bottom).

Mind you the reason I thought it looked like you were in the wrong directory was because you received the message:

> By not providing "Findlibusb.cmake" in CMAKE_MODULE_PATH this project has
asked CMake to find a package configuration file provided by "libusb", but
CMake did not find one.

Which is strange, because `Findlibusb.cmake` exists at `cmake/Findlibusb.cmake`.

rguillome, 2017-01-05:
I had the same error because I was trying a "cmake" at the Heimdall root directory before. RTFM is always the solution :) 
To fix it : 
1. Delete the Heimdall directory
2. Follow Benjamin's instructions and all is ok !

mauiaaron, 2018-03-31:
I encountered the same error, but resolved it with this build command : `cmake -DCMAKE_MODULE_PATH=$(pwd)/../cmake -DCMAKE_BUILD_TYPE=Release ..`

This was on Debian 9.