Benjamin-Dobell/Heimdall #311: Compile Error On Macbook

zbtang, 2015-11-22:
Hi,
I follow the README to compile the source, but I got the following error when i make:

```   
[ 10%] Building CXX object heimdall/CMakeFiles/heimdall.dir/source/Arguments.cpp.o
[ 13%] Building CXX object heimdall/CMakeFiles/heimdall.dir/source/BridgeManager.cpp.o
[ 16%] Building CXX object heimdall/CMakeFiles/heimdall.dir/source/ClosePcScreenAction.cpp.o
[ 20%] Building CXX object heimdall/CMakeFiles/heimdall.dir/source/DetectAction.cpp.o
[ 23%] Building CXX object heimdall/CMakeFiles/heimdall.dir/source/DownloadPitAction.cpp.o
[ 26%] Building CXX object heimdall/CMakeFiles/heimdall.dir/source/FlashAction.cpp.o
[ 30%] Building CXX object heimdall/CMakeFiles/heimdall.dir/source/HelpAction.cpp.o
[ 33%] Building CXX object heimdall/CMakeFiles/heimdall.dir/source/InfoAction.cpp.o
[ 36%] Building CXX object heimdall/CMakeFiles/heimdall.dir/source/Interface.cpp.o
[ 40%] Building CXX object heimdall/CMakeFiles/heimdall.dir/source/main.cpp.o
[ 43%] Building CXX object heimdall/CMakeFiles/heimdall.dir/source/PrintPitAction.cpp.o
[ 46%] Building CXX object heimdall/CMakeFiles/heimdall.dir/source/Utility.cpp.o
[ 50%] Building CXX object heimdall/CMakeFiles/heimdall.dir/source/VersionAction.cpp.o
[ 53%] Linking CXX executable ../bin/heimdall
ld: library not found for -lusb-1.0
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [bin/heimdall] Error 1
make[1]: *** [heimdall/CMakeFiles/heimdall.dir/all] Error 2
make: *** [all] Error 2
```  

It seems that it can't find libusb, but I am sure that I have libusb installed.

Could you help me?

n8henrie, 2015-11-23:
Can you add a few more details? 

- How did you install libusb?
- What version of OS X are you running?

zbtang, 2015-11-23:
I install libusb with Homebrew
```
 brew install libusb qt5 cmake
```
and my OS X version is 10.11.1 (15B42)

Thanks !

n8henrie, 2015-11-23:
It looks like you can't `sudo ldconfig` on Mac -- have you tried rebooting?

zbtang, 2015-11-23:
Yes, I have tried. But it still doesn't work.

There is no `ldconfig` cmd on Mac . Look [here](http://stackoverflow.com/questions/1451047/ldconfig-for-mac-os-x).

n8henrie, 2015-11-23:
Right, that's what I meant, which is why I recommended you try another way to refresh your `ld` path (rebooting), which you've done.

Do you also have `libusb-compat` installed? `brew list | grep libusb`. I happen to have both installed, not sure whether or not the `-compat` version is also installed automatically.

I'm also running OS X 10.11.1 and had no issues building from source yesterday with current `HEAD`.

zbtang, 2015-11-24:
I didn't install `libusb-compat`
```
brew list | grep libusb
libusb
```
And then I install it with `brew install libusb-compat`
```
brew list | grep libusb
libusb
libusb-compat
```
And then removed all makefiles and remake
```
cd build 
rm -rf *
cmake -G "Unix Makefiles" -DCMAKE_BUILD_TYPE=Release -DQt5Widgets_DIR=/usr/local/opt/qt5/lib/cmake/Qt5Widgets ..
make
```
And then I got the same error
```
[#8# 09:53 @build]$make
Scanning dependencies of target pit
[  3%] Building CXX object libpit/CMakeFiles/pit.dir/source/libpit.cpp.o
[  6%] Linking CXX static library libpit.a
[  6%] Built target pit
Scanning dependencies of target heimdall
[ 10%] Building CXX object heimdall/CMakeFiles/heimdall.dir/source/Arguments.cpp.o
[ 13%] Building CXX object heimdall/CMakeFiles/heimdall.dir/source/BridgeManager.cpp.o
[ 16%] Building CXX object heimdall/CMakeFiles/heimdall.dir/source/ClosePcScreenAction.cpp.o
[ 20%] Building CXX object heimdall/CMakeFiles/heimdall.dir/source/DetectAction.cpp.o
[ 23%] Building CXX object heimdall/CMakeFiles/heimdall.dir/source/DownloadPitAction.cpp.o
[ 26%] Building CXX object heimdall/CMakeFiles/heimdall.dir/source/FlashAction.cpp.o
[ 30%] Building CXX object heimdall/CMakeFiles/heimdall.dir/source/HelpAction.cpp.o
[ 33%] Building CXX object heimdall/CMakeFiles/heimdall.dir/source/InfoAction.cpp.o
[ 36%] Building CXX object heimdall/CMakeFiles/heimdall.dir/source/Interface.cpp.o
[ 40%] Building CXX object heimdall/CMakeFiles/heimdall.dir/source/main.cpp.o
[ 43%] Building CXX object heimdall/CMakeFiles/heimdall.dir/source/PrintPitAction.cpp.o
[ 46%] Building CXX object heimdall/CMakeFiles/heimdall.dir/source/Utility.cpp.o
[ 50%] Building CXX object heimdall/CMakeFiles/heimdall.dir/source/VersionAction.cpp.o
[ 53%] Linking CXX executable ../bin/heimdall
ld: library not found for -lusb-1.0
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [bin/heimdall] Error 1
make[1]: *** [heimdall/CMakeFiles/heimdall.dir/all] Error 2
make: *** [all] Error 2
```

And that is the results of `brew doctor`
```
[#10# 09:53 @build]$brew doctor
Your system is ready to brew.
```

Benjamin-Dobell, 2015-11-24:
What is the output from:

```brew ls libusb```

and

```ls -la /usr/local/lib | grep usb```

zbtang, 2015-11-24:
Here is the output:
```
[#17# 10:08 @build]$brew ls libusb
/usr/local/Cellar/libusb/1.0.20/include/libusb-1.0/libusb.h
/usr/local/Cellar/libusb/1.0.20/lib/libusb-1.0.0.dylib
/usr/local/Cellar/libusb/1.0.20/lib/pkgconfig/libusb-1.0.pc
/usr/local/Cellar/libusb/1.0.20/lib/ (2 other files)
[#18# 10:11 @build]$ls -la /usr/local/lib | grep usb
lrwxr-xr-x    1 tangzhibin  admin    52 11 24 09:52 libusb-0.1.4.dylib -> ../Cellar/libusb-compat/0.1.5/lib/libusb-0.1.4.dylib
lrwxr-xr-x    1 tangzhibin  admin    46 11 22 01:45 libusb-1.0.0.dylib -> ../Cellar/libusb/1.0.20/lib/libusb-1.0.0.dylib
lrwxr-xr-x    1 tangzhibin  admin    40 11 22 01:45 libusb-1.0.a -> ../Cellar/libusb/1.0.20/lib/libusb-1.0.a
lrwxr-xr-x    1 tangzhibin  admin    44 11 22 01:45 libusb-1.0.dylib -> ../Cellar/libusb/1.0.20/lib/libusb-1.0.dylib
lrwxr-xr-x    1 tangzhibin  admin    42 11 24 09:52 libusb.a -> ../Cellar/libusb-compat/0.1.5/lib/libusb.a
lrwxr-xr-x    1 tangzhibin  admin    46 11 24 09:52 libusb.dylib -> ../Cellar/libusb-compat/0.1.5/lib/libusb.dylib
```

Jake000, 2015-12-23:
For anyone else running into this problem, I managed to work around it by modifying the link.txt file located at build/heimdall/CMakeFiles/heimdall.dir/link.txt and adding the additional flag -L/usr/local/lib, after which compilation succeeded.

joshgav, 2016-06-05:
You can also prepend the `brew` or `make` command with `LIBRARY_PATH=/usr/local/lib`.