Benjamin-Dobell/Heimdall #314: libusb error -7

tdct12, 2015-12-02:
I've tried a fork, 2 versions, no matter what I do I am plagued by "ERROR: libusb error -7 whilst receiving bulk transfer. Retrying..." Linux mint 17.2 is my OS and this is my preferred recovery tool on osx and windows. Unfortunately I only have access to linux laptop this week, so this is pretty sad :(

SaburoJiro, 2015-12-28:
Please visit my site
https://github.com/SaburoJiro/Heimdall/blob/v1.4.1-hlte/HowToCompileForSCL22.txt

SaburoJiro, 2015-12-29:
I updated install manual. please go to 
https://github.com/SaburoJiro/Heimdall/blob/v1.4.1-hlte/README.md

SaburoJiro, 2015-12-29:
I apologize that my instruction is incorrect.
Then I removed my instruction.

Beeblebrox-BSD, 2016-01-03:
Hello. Similar error, but more details:

I can't download the PIT file using Heimdall, and flashing the device using Odin for root access fails at the "read PIT file" stage. Before posting, I have:

* Followed this step from issues/318:  "Please try to do
$ sudo apt-get install build-essential pkg-config zlib1g-dev libusb-dev libqt4-dev qt4-qmake autoconf libtool libusb-1.0-0-dev"

* From the same issue, the suggestion to blacklist the cdc_acm module made no difference.

It's very possible that I have some level of corruption in my PIT file, and this may be the reason for the error. I have a TWRP backup from when I first rooted some time back, but this does not help with PIT files AFAIK (device needs to be up & running that particuler image?) However, since there are 3 issues on the same topic (this, 313 & 318) it might be an error that came about after a software update.

I have had this error since mid-September 2015. This has been tried on Debian 4 & 6, Odin on WinXP.

<code>Heimdall v1.4.0

Copyright (c) 2010-2013, Benjamin Dobell, Glass Echidna
http://www.glassechidna.com.au/

This software is provided free of charge. Copying and redistribution is
encouraged.

If you appreciate this software and you would like to support future
development please consider donating:
http://www.glassechidna.com.au/donate/

Initialising connection...
Detecting device...
      Manufacturer: "SAMSUNG"
           Product: "Gadget Serial"

            length: 18
      device class: 2
               S/N: 0
           VID:PID: 04E8:685D
         bcdDevice: 021B
   iMan:iProd:iSer: 1:2:0
          nb confs: 1

interface[0].altsetting[0]: num endpoints = 1
   Class.SubClass.Protocol: 02.02.01
       endpoint[0].address: 83
           max packet size: 0010
          polling interval: 09

interface[1].altsetting[0]: num endpoints = 2
   Class.SubClass.Protocol: 0A.00.00
       endpoint[0].address: 81
           max packet size: 0040
          polling interval: 00
       endpoint[1].address: 02
           max packet size: 0040
          polling interval: 00
Claiming interface...
Setting up interface...

Initialising protocol...
WARNING: Control transfer #2 failed. Result: -7
WARNING: Control transfer #3 failed. Result: -7
WARNING: Control transfer #4 failed. Result: -7
WARNING: Control transfer #5 failed. Result: -7
WARNING: Control transfer #6 failed. Result: -1
ERROR: Failed to send data: "ODIN"
Releasing device interface...</code>

Beeblebrox-BSD, 2016-02-27:
I have compiled Heimdall from source on Debian. The "download-pit" command fails with
ERROR: Protocol initialisation failed!
Output of command with usb-log-level debug and verbose commands attached.
[dbg-usb.txt](https://github.com/Benjamin-Dobell/Heimdall/files/149549/dbg-usb.txt)
[dbg-ver.txt](https://github.com/Benjamin-Dobell/Heimdall/files/149548/dbg-ver.txt)

luis-pereira, 2016-07-19:
I do conffirm this findings.
Running the **master** banch (https://github.com/Benjamin-Dobell/Heimdall/commit/9784dd2e2a34b567a239f39aa0d550c519272040). See the [log](https://gist.github.com/luis-pereira/0eff6515ef55ec139b655670d5e17a65).