Benjamin-Dobell/Heimdall #315: Samsung Galaxy S6 Edge+ [SM-G928F]

wadadli, 2015-12-05:
madmin@debian-ws:~$ sudo heimdall detect --verbose --stdout-errors --usb-log-level debug
[timestamp] [threadID] facility level [function call] <message>
--------------------------------------------------------------------------------
[ 0.002751] [0000618f] libusb: debug [libusb_get_device_list] 
[ 0.002821] [0000618f] libusb: debug [discovered_devs_append] need to increase capacity
[ 0.002839] [0000618f] libusb: debug [libusb_get_device_descriptor] 
[ 0.002853] [0000618f] libusb: debug [libusb_get_device_descriptor] 
[ 0.002864] [0000618f] libusb: debug [libusb_get_device_descriptor] 
[ 0.002875] [0000618f] libusb: debug [libusb_get_device_descriptor] 
[ 0.002886] [0000618f] libusb: debug [libusb_get_device_descriptor] 
[ 0.002897] [0000618f] libusb: debug [libusb_get_device_descriptor] 
[ 0.002907] [0000618f] libusb: debug [libusb_get_device_descriptor] 
[ 0.002918] [0000618f] libusb: debug [libusb_get_device_descriptor] 
Device detected
[ 0.002952] [0000618f] libusb: debug [libusb_exit] 
[ 0.002964] [0000618f] libusb: debug [libusb_exit] destroying default context
madmin@debian-ws:~$ sudo heimdall print-pit --verbose --no-reboot --stdout-errors --usb-log-level debug
Heimdall v1.4.0

Copyright (c) 2010-2013, Benjamin Dobell, Glass Echidna
http://www.glassechidna.com.au/

This software is provided free of charge. Copying and redistribution is
encouraged.

If you appreciate this software and you would like to support future
development please consider donating:
http://www.glassechidna.com.au/donate/

Initialising connection...
Detecting device...
[timestamp] [threadID] facility level [function call] <message>
--------------------------------------------------------------------------------
[ 0.002769] [0000619a] libusb: debug [libusb_get_device_list] 
[ 0.002797] [0000619a] libusb: debug [discovered_devs_append] need to increase capacity
[ 0.002809] [0000619a] libusb: debug [libusb_get_device_descriptor] 
[ 0.002814] [0000619a] libusb: debug [libusb_get_device_descriptor] 
[ 0.002819] [0000619a] libusb: debug [libusb_get_device_descriptor] 
[ 0.002823] [0000619a] libusb: debug [libusb_get_device_descriptor] 
[ 0.002827] [0000619a] libusb: debug [libusb_get_device_descriptor] 
[ 0.002831] [0000619a] libusb: debug [libusb_get_device_descriptor] 
[ 0.002835] [0000619a] libusb: debug [libusb_get_device_descriptor] 
[ 0.002840] [0000619a] libusb: debug [libusb_get_device_descriptor] 
[ 0.002847] [0000619a] libusb: debug [libusb_open] open 1.106
[ 0.002874] [0000619a] libusb: debug [usbi_add_pollfd] add fd 11 events 4
[ 0.002892] [0000619a] libusb: debug [libusb_get_device_descriptor] 
[ 0.002902] [0000619a] libusb: debug [add_to_flying_list] arm timerfd for timeout in 1000ms (first in line)
[ 0.002932] [0000619a] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
[ 0.002939] [0000619a] libusb: debug [handle_events] poll() 4 fds with timeout in 60000ms
[ 0.002967] [0000619a] libusb: debug [handle_events] poll() returned 1
[ 0.002975] [0000619a] libusb: debug [reap_for_handle] urb type=2 status=-71 transferred=0
[ 0.002981] [0000619a] libusb: debug [handle_control_completion] handling completion status -71
[ 0.002992] [0000619a] libusb: debug [handle_control_completion] low-level bus error occurred
[ 0.002998] [0000619a] libusb: debug [disarm_timerfd] 
[ 0.003004] [0000619a] libusb: debug [usbi_handle_transfer_completion] transfer 0x23c2ea0 has callback 0x7fb1942db1d0
[ 0.003011] [0000619a] libusb: debug [sync_transfer_cb] actual_length=0
[ 0.003026] [0000619a] libusb: debug [add_to_flying_list] arm timerfd for timeout in 1000ms (first in line)
[ 0.003037] [0000619a] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
[ 0.003042] [0000619a] libusb: debug [handle_events] poll() 4 fds with timeout in 60000ms
[ 0.003098] [0000619a] libusb: debug [handle_events] poll() returned 1
[ 0.003143] [0000619a] libusb: debug [reap_for_handle] urb type=2 status=-71 transferred=0
[ 0.003157] [0000619a] libusb: debug [handle_control_completion] handling completion status -71
[ 0.003170] [0000619a] libusb: debug [handle_control_completion] low-level bus error occurred
[ 0.003183] [0000619a] libusb: debug [disarm_timerfd] 
[ 0.003197] [0000619a] libusb: debug [usbi_handle_transfer_completion] transfer 0x23c2ea0 has callback 0x7fb1942db1d0
[ 0.003210] [0000619a] libusb: debug [sync_transfer_cb] actual_length=0

            length: 18
      device class: 2
               S/N: 0
           VID:PID: 04E8:685D
         bcdDevice: 021B
   iMan:iProd:iSer: 1:2:0
          nb confs: 1
[ 0.003313] [0000619a] libusb: debug [libusb_get_config_descriptor] index 0

interface[0].altsetting[0]: num endpoints = 1
   Class.SubClass.Protocol: 02.02.01
       endpoint[0].address: 83
           max packet size: 0010
          polling interval: 09

interface[1].altsetting[0]: num endpoints = 2
   Class.SubClass.Protocol: 0A.00.00
       endpoint[0].address: 81
           max packet size: 0200
          polling interval: 00
       endpoint[1].address: 02
           max packet size: 0200
          polling interval: 00
Claiming interface...
[ 0.003495] [0000619a] libusb: debug [libusb_claim_interface] interface 1
Setting up interface...
[ 0.003547] [0000619a] libusb: debug [libusb_set_interface_alt_setting] interface 1 altsetting 0
[ 0.004080] [0000619a] libusb: error [op_set_interface] setintf failed error -1 errno 71
ERROR: Setting up interface failed!
ERROR: Setting up interface failed!
Releasing device interface...
[ 0.004126] [0000619a] libusb: debug [libusb_release_interface] interface 1

[ 0.004153] [0000619a] libusb: debug [libusb_close] 
[ 0.004166] [0000619a] libusb: debug [usbi_remove_pollfd] remove fd 11
[ 0.004180] [0000619a] libusb: debug [libusb_exit] 
[ 0.004185] [0000619a] libusb: debug [libusb_exit] destroying default context
madmin@debian-ws:~$

zeroepoch, 2016-03-12:
Anyone wondering who may find this ticket, it worked with 86b7be8329fc5e14e016968b6b1bf22a6ef821d1 (HEAD at this time) no other patches on Fedora 22 on my Galaxy S6 (T-Mobile).  It's model SM-G920T.

zeroepoch, 2016-03-12:
Just tested flashing recovery and it works!  I have to reboot the phone and enter download mode again for another command to work.  It failed to flash recovery but that wasn't heimdall's fault since it prints some error on the screen regarding some lock and using the latest ODIN it also failed.

zeroepoch, 2016-03-12:
Sorry for the comment spam, but in case someone reads this later I was able to flash TWRP.  FRP is Factory Reset Protection (http://www.samsung.com/us/support/answer/ANS00042882/997379756).  It can be disabled under the developer options by enabling OEM unlock.