Benjamin-Dobell/Heimdall #318: Setting up interface failed? libusb: errno 110

feryardiant, 2015-12-12:
I run `sudo ./bin/heimdall print-pit --verbose` from my terminal and got this:

```
Heimdall v1.4.1

Copyright (c) 2010-2014 Benjamin Dobell, Glass Echidna
http://www.glassechidna.com.au/

This software is provided free of charge. Copying and redistribution is
encouraged.

If you appreciate this software and you would like to support future
development please consider donating:
http://www.glassechidna.com.au/donate/

Initialising connection...
Detecting device...

            length: 18
      device class: 2
               S/N: 3
           VID:PID: 04E8:685D
         bcdDevice: 0001
   iMan:iProd:iSer: 2:1:3
          nb confs: 1

interface[0].altsetting[0]: num endpoints = 1
   Class.SubClass.Protocol: 02.02.01
       endpoint[0].address: 83
           max packet size: 0040
          polling interval: 05

interface[1].altsetting[0]: num endpoints = 2
   Class.SubClass.Protocol: 0A.00.00
       endpoint[0].address: 81
           max packet size: 0200
          polling interval: 20
       endpoint[1].address: 02
           max packet size: 0200
          polling interval: 20
Claiming interface...
Setting up interface...
libusb: error [op_set_interface] setintf failed error -1 errno 110
ERROR: Setting up interface failed!
ERROR: Setting up interface failed!
Releasing device interface...
```

OS: Kubuntu 15.10
Device: GT-S5312 (Neo Duos)

SaburoJiro, 2015-12-28:
Please try to do
$ sudo apt-get install build-essential pkg-config zlib1g-dev libusb-dev libqt4-dev qt4-qmake autoconf libtool libusb-1.0-0-dev

feryardiant, 2015-12-28:
Hi @SaburoJiro,

Thanks for your response, just install `pkg-config`, `libqt4-dev` and `qt4-qmake` (because almost packages you suggest are already installed). Unfortunately I can't test it right now, the phone isn't in my hand. I'll tell you when get it back..

Best :smile:

SaburoJiro, 2015-12-29:
I updated install manual. please go to 
https://github.com/SaburoJiro/Heimdall/blob/v1.4.1-hlte/README.md

feryardiant, 2015-12-29:
Nice :smile:

---
Best :)
Fery Wardiyanto

SaburoJiro, 2015-12-29:
Hi,feryardiant
Did your issue solve?

Benjamin-Dobell, 2015-12-29:
@SaburoJiro you do realise you've just gone around to various threads posting incorrect installation instructions, right? Heimdall doesn't (and hasn't for some time) use autotools anyone. I did note that you've told them to clone your fork, which is a bad idea also unless you're talking about just one or two devices. I do realise your fork fixes support for a couple of devices but it also breaks support for many devices.

Benjamin-Dobell, 2015-12-29:
@SaburoJiro you do realise you've just gone around to various threads posting incorrect installation instructions, right?

Heimdall doesn't (and hasn't for some time) use autotools anymore. I did note that you've told them to clone your fork, which is a bad idea also unless you're talking about just one or two devices.

I do realise your fork fixes support for a couple of devices but it also breaks support for many devices.

feryardiant, 2015-12-29:
@SabudoJiro like I said above, i can't test it just yet.

@Benjamin-Dobell, hi ben. Nice to aee your responses here. Actually, I was thinking that @SabudoJiro is your maintener, wasn't he?

SaburoJiro, 2015-12-29:
I apologize that my instruction is incorrect.
Then I removed my instruction.

feryardiant, 2015-12-30:
Just test it with `sudo ./bin/heimdall print-pit --verbose` and got this.

```
Heimdall v1.4.0

Copyright (c) 2010-2013, Benjamin Dobell, Glass Echidna
http://www.glassechidna.com.au/

This software is provided free of charge. Copying and redistribution is
encouraged.

If you appreciate this software and you would like to support future
development please consider donating:
http://www.glassechidna.com.au/donate/

Initialising connection...
Detecting device...
ERROR: Failed to access device. libusb error: 1736502931
```

Any other solution?

Anyway @Benjamin-Dobell, I'm not sure it is about Heimdal' bug, but did you have any clue for me to solve this problem?

Thanks

j-alonso, 2016-01-02:
Could you try the following:
- Put the device in ODIN mode (no USB cable)
- Execute the commands as root:
```sh
     echo "blacklist cdc_acm" >/etc/modprobe.d/cdc_acm-blacklist.conf
     rmmod cdc_acm # ignore ERROR in use
```
- Plug the USB cable and use Heimdall (v1.4.1 is ok).
- At end remove the file /etc/modprobe.d/cdc_acm-blacklist.conf

martinopavese, 2016-02-18:
Hi guis, I'm trying to use heimdall to flash a recovery image (heimdall flash --RECOVERY twrp-3.0.0-0-slte-1.0.img) and i get the libusb error too.
`heimdall version
v1.4.0
heimdall flash --RECOVERY twrp-3.0.0-0-slte-1.0.img --verboseHeimdall v1.4.0

Copyright (c) 2010-2013, Benjamin Dobell, Glass Echidna
http://www.glassechidna.com.au/

This software is provided free of charge. Copying and redistribution is
encouraged.

If you appreciate this software and you would like to support future
development please consider donating:
http://www.glassechidna.com.au/donate/

Initialising connection...
Detecting device...

            length: 18
      device class: 2
               S/N: 0
           VID:PID: 04E8:685D
         bcdDevice: 021B
   iMan:iProd:iSer: 1:2:0
          nb confs: 1

interface[0].altsetting[0]: num endpoints = 1
   Class.SubClass.Protocol: 02.02.01
       endpoint[0].address: 83
           max packet size: 0010
          polling interval: 09

interface[1].altsetting[0]: num endpoints = 2
   Class.SubClass.Protocol: 0A.00.00
       endpoint[0].address: 81
           max packet size: 0200
          polling interval: 00
       endpoint[1].address: 02
           max packet size: 0200
          polling interval: 00
Claiming interface...
Setting up interface...
libusb: error [op_set_interface] setintf failed error -1 errno 71
ERROR: Setting up interface failed!
Releasing device interface...`
I've tried disabling the cdc_acm module, without any success.
Am I missing something?

bammorgan, 2016-10-30:
really struggling with this. 
when I print-pit I get this
`Heimdall v1.3.1, Copyright (c) 2010-2011, Benjamin Dobell, Glass Echidna
http://www.glassechidna.com.au

This software is provided free of charge. Copying and redistribution is
encouraged.

If you appreciate this software and you would like to support future
development please consider donating:
http://www.glassechidna.com.au/donate/

Initialising connection...
Detecting device...

            length: 18
      device class: 2
               S/N: 0
           VID:PID: 04E8:685D
         bcdDevice: 021B
   iMan:iProd:iSer: 1:2:0
          nb confs: 1

interface[0].altsetting[0]: num endpoints = 1
   Class.SubClass.Protocol: 02.02.01
       endpoint[0].address: 83
           max packet size: 0010
          polling interval: 09

interface[1].altsetting[0]: num endpoints = 2
   Class.SubClass.Protocol: 0A.00.00
       endpoint[0].address: 81
           max packet size: 0200
          polling interval: 00
       endpoint[1].address: 02
           max packet size: 0200
          polling interval: 00
Claiming interface...
Setting up interface...
ERROR: Setting up interface failed!`

I am new to this so be gentle please.
I just want to root my galaxy S6 Edge

cannedbrain, 2017-11-20:
Tried it with `sudo heimdall print-pit --verbose` on Mac OS X 10.11.6 El Capitan

`Heimdall v1.4.1

Copyright (c) 2010-2014 Benjamin Dobell, Glass Echidna
http://www.glassechidna.com.au/

This software is provided free of charge. Copying and redistribution is
encouraged.

If you appreciate this software and you would like to support future
development please consider donating:
http://www.glassechidna.com.au/donate/

Initialising connection...
Detecting device...
      Manufacturer: "SAMSUNG"
           Product: "SAMSUNG USB"

            length: 18
      device class: 2
               S/N: 0
           VID:PID: 04E8:685D
         bcdDevice: 0100
   iMan:iProd:iSer: 1:2:0
          nb confs: 1

interface[0].altsetting[0]: num endpoints = 1
   Class.SubClass.Protocol: 02.02.01
       endpoint[0].address: 82
           max packet size: 0010
          polling interval: 09

interface[1].altsetting[0]: num endpoints = 2
   Class.SubClass.Protocol: 0A.00.00
       endpoint[0].address: 81
           max packet size: 0200
          polling interval: 00
       endpoint[1].address: 01
           max packet size: 0200
          polling interval: 00
Claiming interface...
Setting up interface...
libusbx: error [get_endpoints] error getting pipe information for pipe 1: unknown error
libusbx: error [darwin_set_interface_altsetting] could not build endpoint table
ERROR: Setting up interface failed!
Releasing device interface...`

antroyd, 2017-12-14:
Same issue here:

Notebook: Linux Mint 17.3 (based on Ubuntu 14.04 Trusty Tahr)
Phone: Samsung Galaxy S7 (SM-G930F) herolte
Heimdall: v1.4.0-2 (from repository)

	sudo heimdall flash --RECOVERY recovery-twrp-3.2.1-0-herolte.img --no-reboot

	Heimdall v1.4.0

	Copyright (c) 2010-2013, Benjamin Dobell, Glass Echidna
	http://www.glassechidna.com.au/

	This software is provided free of charge. Copying and redistribution is
	encouraged.

	If you appreciate this software and you would like to support future
	development please consider donating:
	http://www.glassechidna.com.au/donate/

	Initialising connection...
	Detecting device...
	Claiming interface...
	Setting up interface...
	libusbx: error [op_set_interface] setintf failed error -1 errno 71
	ERROR: Setting up interface failed!
	Releasing device interface...

Disabling the cdc_acm module had no effect.

I am not convinced that this is a Heimfall issue, since my setup has successfully flashed two Samsung Galaxy S4 Mini (GT-I9195) serranoltexx just a few days ago.

cheseremtitus24, 2023-03-12:
This issue is solved by restarting the device in download mode just before you run the heimdall command. 

NOTE: Before this worked on my SM-J20F device model I had built and installed libusb from source.

martinopavese, 2023-03-13:
Early 2016.
Wow.

k2ll3n, 2026-08-19:
@cheseremtitus24 Indeed, thanks!

I always tried to run:
```
 ./build/bin/heimdall print-pit --no-reboot
```
And next got this error
```
./build/bin/heimdall flash --RECOVERY ~/Downloads/recovery.img --no-reboot --verbose
...
Resetting device...
ERROR: Failed to receive handshake response. Result: -7
ERROR: Protocol initialisation failed!

```
Turns out, I simply had to run the flash command on its own, right after going into odin mode.