Benjamin-Dobell/Heimdall #320: gt-n5110 flashing gives recovery  does not exist err

jurgynet, 2015-12-23:
trying to root my stock galaxy note 8 (4.4.2 )
but this command:
heimdall flash --RECOVERY recovery.img --CACHE cache.img

gives:
...
Downloading device's PIT file...
PIT file download successful.
ERROR: Partition "recovery" does not exist in the specified PIT.
Ending session...

Am using this file:
#  sum CF-Auto-Root-konawifi-konawifixx-gtn5110.zip recovery.img  cache.img
32855 17956 CF-Auto-Root-konawifi-konawifixx-gtn5110.zip
28467  6668 recovery.img
41023 15605 cache.img

Is there a newer recovery image for gt-n5110 ?

ildar, 2016-03-14:
I have N5100 and may help.
Please show us the output of `heimdall print-pit`

Benjamin-Dobell, 2016-03-14:
Hmm, I missed it at the time, but I strongly suspect this was a user error. This isn't really a help forum, but is instead a replace to report issues with Heimdall. As such I'm going to close this issue (I'll reopen it if it was really a bug in Heimdall).

For help you can either email me (address on the Glass Echidna website), or seek help at XDA developers. I don't always have time to respond so I'd suggest trying XDA Developers first.

closed 2016-03-14 by Benjamin-Dobell