Benjamin-Dobell/Heimdall #335: gt-s5660: ERROR: Protocol initialisation failed

ildar, 2016-03-14:
`heimdall download-pit --output pit --no-reboot --usb-log-level debug`

> Heimdall v1.4.1
> 
> Copyright (c) 2010-2014 Benjamin Dobell, Glass Echidna
> http://www.glassechidna.com.au/
> 
> This software is provided free of charge. Copying and redistribution is
> encouraged.
> 
> If you appreciate this software and you would like to support future
> development please consider donating:
> http://www.glassechidna.com.au/donate/
> 
> Initialising connection...
> Detecting device...
> [timestamp] [threadID] facility level [function call] <message>
> --------------------------------------------------------------------------------
> [ 0.003362] [000019b4] libusb: debug [libusb_get_device_list] 
> [ 0.003387] [000019b4] libusb: debug [discovered_devs_append] need to increase capacity
> [ 0.003392] [000019b4] libusb: debug [libusb_get_device_descriptor] 
> [ 0.003395] [000019b4] libusb: debug [libusb_get_device_descriptor] 
> [ 0.003397] [000019b4] libusb: debug [libusb_get_device_descriptor] 
> [ 0.003398] [000019b4] libusb: debug [libusb_get_device_descriptor] 
> [ 0.003400] [000019b4] libusb: debug [libusb_get_device_descriptor] 
> [ 0.003402] [000019b4] libusb: debug [libusb_get_device_descriptor] 
> [ 0.003404] [000019b4] libusb: debug [libusb_get_device_descriptor] 
> [ 0.003407] [000019b4] libusb: debug [libusb_get_device_descriptor] 
> [ 0.003411] [000019b4] libusb: debug [libusb_open] open 2.120
> [ 0.003430] [000019b4] libusb: debug [usbi_add_pollfd] add fd 11 events 4
> [ 0.003436] [000019b4] libusb: debug [libusb_get_device_descriptor] 
> [ 0.003439] [000019b4] libusb: debug [libusb_get_config_descriptor] index 0
> Claiming interface...
> [ 0.003451] [000019b4] libusb: debug [libusb_claim_interface] interface 1
> Setting up interface...
> [ 0.003469] [000019b4] libusb: debug [libusb_set_interface_alt_setting] interface 1 altsetting 0
> 
> Initialising protocol...
> [ 0.005754] [000019b4] libusb: debug [libusb_alloc_transfer] transfer 0x663308
> [ 0.005758] [000019b4] libusb: debug [libusb_submit_transfer] transfer 0x663308
> [ 0.005761] [000019b4] libusb: debug [add_to_flying_list] arm timerfd for timeout in 1000ms (first in line)
> [ 0.005768] [000019b4] libusb: debug [submit_bulk_transfer] need 1 urbs for new transfer with length 4
> [ 0.005779] [000019b4] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
> [ 0.005786] [000019b4] libusb: debug [handle_events] poll fds modified, reallocating
> [ 0.005792] [000019b4] libusb: debug [handle_events] poll() 3 fds with timeout in 60000ms
> [ 0.005798] [000019b4] libusb: debug [handle_events] poll() returned 1
> [ 0.005803] [000019b4] libusb: debug [reap_for_handle] urb type=3 status=0 transferred=4
> [ 0.005807] [000019b4] libusb: debug [handle_bulk_completion] handling completion status 0 of bulk urb 1/1
> [ 0.005811] [000019b4] libusb: debug [handle_bulk_completion] last URB in transfer --> complete!
> [ 0.005814] [000019b4] libusb: debug [disarm_timerfd] 
> [ 0.005819] [000019b4] libusb: debug [usbi_handle_transfer_completion] transfer 0x663308 has callback 0x7fb18a8f8870
> [ 0.005823] [000019b4] libusb: debug [sync_transfer_cb] actual_length=4
> [ 0.005832] [000019b4] libusb: debug [libusb_free_transfer] transfer 0x663308
> [ 0.005853] [000019b4] libusb: debug [libusb_alloc_transfer] transfer 0x663308
> [ 0.005856] [000019b4] libusb: debug [libusb_submit_transfer] transfer 0x663308
> [ 0.005860] [000019b4] libusb: debug [add_to_flying_list] arm timerfd for timeout in 1000ms (first in line)
> [ 0.005865] [000019b4] libusb: debug [submit_bulk_transfer] need 1 urbs for new transfer with length 7
> [ 0.005873] [000019b4] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
> [ 0.005877] [000019b4] libusb: debug [handle_events] poll() 3 fds with timeout in 60000ms
> [ 1.005912] [000019b4] libusb: debug [handle_events] poll() returned 1
> [ 1.005929] [000019b4] libusb: debug [handle_events] timerfd triggered
> [ 1.005935] [000019b4] libusb: debug [libusb_cancel_transfer] transfer 0x663308
> [ 1.005961] [000019b4] libusb: debug [disarm_timerfd] 
> [ 1.005965] [000019b4] libusb: debug [handle_events] poll() 3 fds with timeout in 0ms
> [ 1.005969] [000019b4] libusb: debug [handle_events] poll() returned 1
> [ 1.005973] [000019b4] libusb: debug [reap_for_handle] urb type=3 status=-2 transferred=0
> [ 1.005977] [000019b4] libusb: debug [handle_bulk_completion] handling completion status -2 of bulk urb 1/1
> [ 1.005981] [000019b4] libusb: debug [handle_bulk_completion] abnormal reap: urb status -2
> [ 1.005993] [000019b4] libusb: debug [handle_bulk_completion] abnormal reap: last URB handled, reporting
> [ 1.005998] [000019b4] libusb: debug [usbi_handle_transfer_cancellation] detected timeout cancellation
> [ 1.006001] [000019b4] libusb: debug [disarm_timerfd] 
> [ 1.006006] [000019b4] libusb: debug [usbi_handle_transfer_completion] transfer 0x663308 has callback 0x7fb18a8f8870
> [ 1.006010] [000019b4] libusb: debug [sync_transfer_cb] actual_length=0
> [ 1.006016] [000019b4] libusb: debug [libusb_free_transfer] transfer 0x663308
> ERROR: Protocol initialisation failed!
> 
> Releasing device interface...
> [ 1.006089] [000019b4] libusb: debug [libusb_release_interface] interface 1
> 
> [ 1.006110] [000019b4] libusb: debug [libusb_close] 
> [ 1.006118] [000019b4] libusb: debug [usbi_remove_pollfd] remove fd 11
> [ 1.006132] [000019b4] libusb: debug [libusb_exit] 
> [ 1.006136] [000019b4] libusb: debug [libusb_exit] destroying default context
>

Benjamin-Dobell, 2016-03-14:
Which device and which OS?

ildar, 2016-03-14:
As written, it is Samsung gt-s5660, build `S5660XXKTI` [GT-S5660-user 2.3.6
GINGERBREAD XXKTI release-keys]

Benjamin-Dobell, 2016-03-14:
Sorry, I meant your computer OS. I'm going to assume Linux, but you didn't
state.

ildar, 2016-03-14:
Yeah, that's Linux x86_64. I suspected some kernel modules interfere but
that wasn't true: another sam-phone gave its pit allright.

Benjamin-Dobell, 2016-03-14:
I just google'd the GT-S5660 and released that's the Gio. I don't have access to one so can't confirm, but I think that device is like the Galaxy Y i.e. not a Galaxy S phone, and therefore uses a different version of the flashing protocol (which has almost nothing in common with the protocol implemented by Heimdall).

ildar, 2016-03-14:
Oh! Do you mean that some Galaxy phones are supposed to be unsupported? I
found #108 , but is there a metabug (or wiki page about ...) for
unsupported families of bootloaders?

Benjamin-Dobell, 2016-03-14:
Well, I wouldn't say "supposed to be unsupported", but technically there's only very few devices that are officially supported - and they're the ones I have access to (http://glassechidna.com.au/heimdall/). However, there are definitely *many* more devices than that that I do my best to support.

Heimdall was originally written for the Galaxy S (GT-I9000). Samsung have made incremental changes to the protocol with newer Galaxy S devices, and for the most part I've tried to keep up with those changes. However, some Samsung devices come from a different device family altogether and use a completely different protocol; it's actually an "older" Odin protocol, but from what I've seen it's *very* different.

If someone were to add support for that protocol to Heimdall (and do so cleanly) then I'd happily merge it in. However, I don't really have the time perform that undertaking myself unfortunately.

Benjamin-Dobell, 2016-03-14:
Re: A Wiki page. You're welcome to start one, although trying to maintain an exhaustive list probably isn't a good idea. Maybe just a page about known unsupported devices and *why* they're presently unsupported.