Benjamin-Dobell/Heimdall #336: "This package is incompatible with this version of OS X and may fail to install."

mfn, 2016-03-26:
- Downloaded from https://bitbucket.org/benjamin_dobell/heimdall/downloads/heimdall-suite-1.4.0-mac.dmg .
- Opened dmg, opened pkg, got

> This package is incompatible with this version of OS X and may fail to install.
> Installing this package may damage your system, and the installation may fail. To stop the program and the installation, click “Cancel Installation”.
> [ ] Install Anyway  [ ] Cancel Installation

Tested on OS X El Capitan 10.11.4 (15E65)

retorquere, 2016-04-18:
Same problem here.

gfrancesco, 2016-04-18:
Same here, if you continue the installation actually fail.
Tried also to install with `brew`, no difference.

leoshibata, 2016-05-09:
Same problem here.

Benjamin-Dobell, 2016-05-20:
https://github.com/Benjamin-Dobell/Heimdall/issues/291

closed 2016-05-20 by Benjamin-Dobell

SMN3110, 2017-09-26:
I have the same issues, what I need to do?
Thanks!

socatres, 2018-01-14:
I just changed the instalation drive and it works

jedisct1, 2018-01-20:
Same. I don't have multiple installation drives.

benslamajihed, 2022-08-16:
Work on mac os 12.5

lifeeric, 2023-07-06:
Tried both brew and dmg Same problem on 11.6