Benjamin-Dobell/Heimdall #339: Galaxy S III (Sprint) does not act on recovery upload (osx)

kristovatlas, 2016-04-18:
I've tried this with the 1.4.0 binary for osx, as well as building from current master (7ebee1e87728a468bda9547c83b42317acd0e676).

Sample output:
```
$ ./heimdall flash --RECOVERY ~/Downloads/twrp-2.8.7.0-d2spr.img --no-reboot --verbose
Heimdall v1.4.1

Copyright (c) 2010-2014 Benjamin Dobell, Glass Echidna
http://www.glassechidna.com.au/

This software is provided free of charge. Copying and redistribution is
encouraged.

If you appreciate this software and you would like to support future
development please consider donating:
http://www.glassechidna.com.au/donate/

Initialising connection...
Detecting device...
      Manufacturer: "Sasmsung"
           Product: "MSM8960"

            length: 18
      device class: 2
               S/N: 0
           VID:PID: 04E8:685D
         bcdDevice: 0100
   iMan:iProd:iSer: 1:2:0
          nb confs: 1

interface[0].altsetting[0]: num endpoints = 1
   Class.SubClass.Protocol: 02.02.01
       endpoint[0].address: 82
           max packet size: 0010
          polling interval: 09

interface[1].altsetting[0]: num endpoints = 2
   Class.SubClass.Protocol: 0A.00.00
       endpoint[0].address: 81
           max packet size: 0200
          polling interval: 00
       endpoint[1].address: 01
           max packet size: 0200
          polling interval: 00
Claiming interface...
Setting up interface...

Initialising protocol...
Protocol initialisation successful.

Beginning session...
WARNING: Empty bulk transfer after sending packet failed. Continuing anyway...

Some devices may take up to 2 minutes to respond.
Please be patient!

WARNING: Empty bulk transfer after sending packet failed. Continuing anyway...
Session begun.

WARNING: Empty bulk transfer after sending packet failed. Continuing anyway...
Downloading device's PIT file...
WARNING: Empty bulk transfer after sending packet failed. Continuing anyway...
WARNING: Empty bulk transfer after sending packet failed. Continuing anyway...
WARNING: Empty bulk transfer after sending packet failed. Continuing anyway...
WARNING: Empty bulk transfer after sending packet failed. Continuing anyway...
WARNING: Empty bulk transfer after sending packet failed. Continuing anyway...
WARNING: Empty bulk transfer after sending packet failed. Continuing anyway...
WARNING: Empty bulk transfer after sending packet failed. Continuing anyway...
WARNING: Empty bulk transfer after sending packet failed. Continuing anyway...
WARNING: Empty bulk transfer after sending packet failed. Continuing anyway...
WARNING: Empty bulk transfer after sending packet failed. Continuing anyway...
WARNING: Empty bulk transfer after sending packet failed. Continuing anyway...
WARNING: Empty bulk transfer after sending packet failed. Continuing anyway...
WARNING: Empty bulk transfer after sending packet failed. Continuing anyway...
WARNING: Empty bulk transfer after sending packet failed. Continuing anyway...
WARNING: Empty bulk transfer after sending packet failed. Continuing anyway...
WARNING: Empty bulk transfer after sending packet failed. Continuing anyway...
WARNING: Empty bulk transfer after sending packet failed. Continuing anyway...
WARNING: Empty bulk transfer after sending packet failed. Continuing anyway...
WARNING: Empty bulk transfer after receiving packet failed. Continuing anyway...
WARNING: Empty bulk transfer after sending packet failed. Continuing anyway...
PIT file download successful.

Uploading RECOVERY
WARNING: Empty bulk transfer after sending packet failed. Continuing anyway...
0%WARNING: Empty bulk transfer after sending packet failed. Continuing anyway...

11%

23%

35%

47%

58%

70%

82%

94%

100%
WARNING: Empty bulk transfer after sending packet failed. Continuing anyway...
RECOVERY upload successful

Ending session...
WARNING: Empty bulk transfer after sending packet failed. Continuing anyway...
Releasing device interface...
```

After this, the device remains in download mode with no changes to indicate that the recovery upload was acted upon.

kristovatlas, 2016-04-18:
Here's the results using the CyanogenMod Recovery image for my device:

```
$ ./heimdall flash --RECOVERY ~/Downloads/cm-13.0-20160418-NIGHTLY-d2spr-recovery.img --no-reboot --verbose
Heimdall v1.4.1

Copyright (c) 2010-2014 Benjamin Dobell, Glass Echidna
http://www.glassechidna.com.au/

This software is provided free of charge. Copying and redistribution is
encouraged.

If you appreciate this software and you would like to support future
development please consider donating:
http://www.glassechidna.com.au/donate/

Initialising connection...
Detecting device...
      Manufacturer: "Sasmsung"
           Product: "MSM8960"

            length: 18
      device class: 2
               S/N: 0
           VID:PID: 04E8:685D
         bcdDevice: 0100
   iMan:iProd:iSer: 1:2:0
          nb confs: 1

interface[0].altsetting[0]: num endpoints = 1
   Class.SubClass.Protocol: 02.02.01
       endpoint[0].address: 82
           max packet size: 0010
          polling interval: 09

interface[1].altsetting[0]: num endpoints = 2
   Class.SubClass.Protocol: 0A.00.00
       endpoint[0].address: 81
           max packet size: 0200
          polling interval: 00
       endpoint[1].address: 01
           max packet size: 0200
          polling interval: 00
Claiming interface...
Setting up interface...

Initialising protocol...
Protocol initialisation successful.

Beginning session...
WARNING: Empty bulk transfer after sending packet failed. Continuing anyway...

Some devices may take up to 2 minutes to respond.
Please be patient!

WARNING: Empty bulk transfer after sending packet failed. Continuing anyway...
Session begun.

WARNING: Empty bulk transfer after sending packet failed. Continuing anyway...
Downloading device's PIT file...
WARNING: Empty bulk transfer after sending packet failed. Continuing anyway...
WARNING: Empty bulk transfer after sending packet failed. Continuing anyway...
WARNING: Empty bulk transfer after sending packet failed. Continuing anyway...
WARNING: Empty bulk transfer after sending packet failed. Continuing anyway...
WARNING: Empty bulk transfer after sending packet failed. Continuing anyway...
WARNING: Empty bulk transfer after sending packet failed. Continuing anyway...
WARNING: Empty bulk transfer after sending packet failed. Continuing anyway...
WARNING: Empty bulk transfer after sending packet failed. Continuing anyway...
WARNING: Empty bulk transfer after sending packet failed. Continuing anyway...
WARNING: Empty bulk transfer after sending packet failed. Continuing anyway...
WARNING: Empty bulk transfer after sending packet failed. Continuing anyway...
WARNING: Empty bulk transfer after sending packet failed. Continuing anyway...
WARNING: Empty bulk transfer after sending packet failed. Continuing anyway...
WARNING: Empty bulk transfer after sending packet failed. Continuing anyway...
WARNING: Empty bulk transfer after sending packet failed. Continuing anyway...
WARNING: Empty bulk transfer after sending packet failed. Continuing anyway...
WARNING: Empty bulk transfer after sending packet failed. Continuing anyway...
WARNING: Empty bulk transfer after sending packet failed. Continuing anyway...
WARNING: Empty bulk transfer after receiving packet failed. Continuing anyway...
WARNING: Empty bulk transfer after sending packet failed. Continuing anyway...
PIT file download successful.

Uploading RECOVERY
WARNING: Empty bulk transfer after sending packet failed. Continuing anyway...
0%WARNING: Empty bulk transfer after sending packet failed. Continuing anyway...

13%

26%

39%

52%

65%

78%

91%

100%
WARNING: Empty bulk transfer after sending packet failed. Continuing anyway...
RECOVERY upload successful

Ending session...
WARNING: Empty bulk transfer after sending packet failed. Continuing anyway...
Releasing device interface...
```

kristovatlas, 2016-04-18:
Also tried running this using `sudo` but no changes.

Benjamin-Dobell, 2016-04-18:
`RECOVERY upload successful`

It succeeded.

> ./heimdall flash --RECOVERY ~/Downloads/cm-13.0-20160418-NIGHTLY-d2spr-recovery.img --no-reboot --verbose
Heimdall v1.4.1

...

> Also tried running this using sudo but no changes.

You told the device not to reboot when it completed flashing.

closed 2016-04-18 by Benjamin-Dobell

kristovatlas, 2016-04-18:
Maybe.

I'm following this guide: https://wiki.cyanogenmod.org/w/Install_CM_for_d2spr

`--no-reboot` must be used because the device will overwrite the custom recovery file if allowed to boot normally, according to this guide.

The guide states:

> A blue transfer bar will appear on the device showing the recovery being transferred.

and:

> It may be necessary to run the heimdall flash command as root in Linux or Mac OS X: sudo heimdall flash [...]. Typical symptoms of this are:
> a blue bar doesn't show up after the process has finished

The blue bar does not show up in my device.

Once I power off the device and attempt to enter recovery mode, I see a brief flash of blue text in the upper left corner, followed by nothing. I realize that you're not providing support for the recovery files I'm attempt to flash, but this along with the lack of expected UI behavior leads me to wonder whether the operation actually succeeded.

Thanks.

kristovatlas, 2016-04-18:
After trying to boot into recovery mode about a dozen more times, it finally succeeded. No idea why the UI didn't reflect the "blue bar," but it doesn't matter. Thanks for the great software, and sorry for the nuisance.