Benjamin-Dobell/Heimdall #340: error setting up interface failed

ankurhora, 2016-04-27:
Hi,

on opening heimdall fronthand my device gets detected in the utilities tab in download mode ( volume down + power + home button ) but when try to initiate download .pit file it fails giving the error 
"error setting up interface failed" 
Please guide what to do 
Using Samsung Galaxy Y GT-S5360

sgitkene, 2016-05-02:
`[max@SurfaceMax Downloads]$ sudo heimdall flash --RECOVERY recovery.img --no-reboot --verbose --usb-log-level debug
Heimdall v1.4.1

Copyright (c) 2010-2014 Benjamin Dobell, Glass Echidna
http://www.glassechidna.com.au/

This software is provided free of charge. Copying and redistribution is
encouraged.

If you appreciate this software and you would like to support future
development please consider donating:
http://www.glassechidna.com.au/donate/

Initialising connection...
Detecting device...
[timestamp] [threadID] facility level [function call] <message>
--------------------------------------------------------------------------------
[ 0.002249] [0000765a] libusb: debug [libusb_get_device_list] 
[ 0.002271] [0000765a] libusb: debug [libusb_get_device_descriptor] 
[ 0.002275] [0000765a] libusb: debug [libusb_get_device_descriptor] 
[ 0.002278] [0000765a] libusb: debug [libusb_get_device_descriptor] 
[ 0.002280] [0000765a] libusb: debug [libusb_get_device_descriptor] 
[ 0.002283] [0000765a] libusb: debug [libusb_get_device_descriptor] 
[ 0.002286] [0000765a] libusb: debug [libusb_get_device_descriptor] 
[ 0.002291] [0000765a] libusb: debug [libusb_open] open 1.56
[30.175363] [0000765a] libusb: error [_get_usbfs_fd] libusb couldn't open USB device /dev/bus/usb/001/056: No such device
[30.175380] [0000765a] libusb: debug [libusb_open] open 1.56 returns -1
ERROR: Failed to access device. libusb error: -1
[30.175415] [0000765a] libusb: debug [libusb_exit] 
[30.175418] [0000765a] libusb: debug [libusb_exit] destroying default context
[max@SurfaceMax Downloads]$ sudo heimdall flash --RECOVERY recovery.img --no-reboot --verbose --usb-log-level debug
Heimdall v1.4.1

Copyright (c) 2010-2014 Benjamin Dobell, Glass Echidna
http://www.glassechidna.com.au/

This software is provided free of charge. Copying and redistribution is
encouraged.

If you appreciate this software and you would like to support future
development please consider donating:
http://www.glassechidna.com.au/donate/

Initialising connection...
Detecting device...
[timestamp] [threadID] facility level [function call] <message>
--------------------------------------------------------------------------------
[ 3.421279] [00002506] libusb: debug [libusb_get_device_list] 
[ 3.421317] [00002506] libusb: debug [libusb_get_device_descriptor] 
[ 3.421319] [00002506] libusb: debug [libusb_get_device_descriptor] 
[ 3.421321] [00002506] libusb: debug [libusb_get_device_descriptor] 
[ 3.421324] [00002506] libusb: debug [libusb_get_device_descriptor] 
[ 3.421330] [00002506] libusb: debug [libusb_get_device_descriptor] 
[ 3.421334] [00002506] libusb: debug [libusb_get_device_descriptor] 
[ 3.421338] [00002506] libusb: debug [libusb_open] open 1.57
[ 3.421353] [00002506] libusb: debug [usbi_add_pollfd] add fd 10 events 4
[ 3.421356] [00002506] libusb: debug [libusb_get_device_descriptor] 
[ 3.421359] [00002506] libusb: debug [libusb_alloc_transfer] transfer 0x94e088
[ 3.421363] [00002506] libusb: debug [libusb_submit_transfer] transfer 0x94e088
[ 3.421366] [00002506] libusb: debug [add_to_flying_list] arm timerfd for timeout in 1000ms (first in line)
[ 3.421378] [00002506] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
[ 3.421381] [00002506] libusb: debug [handle_events] poll fds modified, reallocating
[ 3.421384] [00002506] libusb: debug [handle_events] poll() 3 fds with timeout in 60000ms
[ 3.933336] [00002507] libusb: debug [linux_get_device_address] getting address for device: 1-1 detached: 0
[ 3.933352] [00002507] libusb: debug [linux_get_device_address] scan 1-1
[ 3.933384] [00002507] libusb: debug [linux_get_device_address] bus=1 dev=57
[ 3.933386] [00002507] libusb: debug [udev_hotplug_event] udev hotplug event. action: add.
[ 3.933388] [00002507] libusb: debug [linux_enumerate_device] busnum 1 devaddr 57 session_id 313
[ 3.933391] [00002507] libusb: debug [linux_enumerate_device] session_id 313 already exists
[ 4.421419] [00002506] libusb: debug [handle_events] poll() returned 1
[... 67 lines cut ...]
           max packet size: 0200
          polling interval: 20
       endpoint[1].address: 02
           max packet size: 0200
          polling interval: 20
Claiming interface...
[ 6.421943] [00002506] libusb: debug [libusb_claim_interface] interface 1
Attempt failed. Detaching driver...
[ 6.421950] [00002506] libusb: debug [libusb_detach_kernel_driver] interface 1
Claiming interface again...
[ 6.422081] [00002506] libusb: debug [libusb_claim_interface] interface 1
Setting up interface...
[ 6.422097] [00002506] libusb: debug [libusb_set_interface_alt_setting] interface 1 altsetting 0
[11.420693] [00002506] libusb: error [op_set_interface] setintf failed error -1 errno 110
ERROR: Setting up interface failed!
Releasing device interface...
[11.420767] [00002506] libusb: debug [libusb_release_interface] interface 1
Re-attaching kernel driver...
[11.420801] [00002506] libusb: debug [libusb_attach_kernel_driver] interface 1

[11.420827] [00002506] libusb: debug [libusb_close] 
[11.420846] [00002506] libusb: debug [usbi_remove_pollfd] remove fd 10
[11.420865] [00002506] libusb: debug [libusb_exit] 
[11.420870] [00002506] libusb: debug [libusb_exit] destroying default context
`

joshgav, 2016-06-05:
I was able to work around this by skipping the call to `libusb_set_interface_alt_setting`, that is by commenting out https://github.com/Benjamin-Dobell/Heimdall/blob/master/heimdall/source/BridgeManager.cpp#L264 and recompiling.

That works for me cause the alt setting is 0, which is probably the default, anyway.