Benjamin-Dobell/Heimdall #341: Having trouble compiling.

Rurounin, 2016-05-04:
Hey, i'm not sure if this is the appropriate place to report issues like this but i have looked around on forums and found nothing of help to a novice like me.

I'm unable to download the pit file off my I9100 on the 1.4 build, and reading up about it it would seem i must compile it myself.

Compiling stuff is something i always avoid when i can since things tends to go wrong when i do, this being no exception, i guess the real problem is that i don't have a clue how to fix it or where to even begin looking as i'm no programmer beyond very basic stuff.

That said I've been following this guide (and some other that didn't work at all):
https://wiki.cyanogenmod.org/w/Install_and_compile_Heimdall

It seems to fail during the make phase, specifically when making heimdall-frontend.exe. (cannot find -ljasper, -ldbus-1, -lwebp etc.)

Now, i'm not sure what information is relevant or where to find it or i'd post it here, if there is a log i have yet to find it.

I'm on Windows 10 pro x64, and again, sorry for cluttering this section with a presumably non-issue.

nygmatron, 2016-05-09:
I got the very same problem here.

(https://wiki.cyanogenmod.org/w/Install_and_compile_Heimdall) worked flawlessly until the "make" part.

I never compiled anything before, so i really don't know what is missing from the tutorial / what I did wrong.

Thank you so much for your assistance

$ make
Scanning dependencies of target pit
[  3%] Building CXX object libpit/CMakeFiles/pit.dir/source/libpit.cpp.obj
[  6%] Linking CXX static library libpit.a
[  6%] Built target pit
Scanning dependencies of target heimdall
[ 10%] Building CXX object heimdall/CMakeFiles/heimdall.dir/source/Arguments.cpp.obj
[ 13%] Building CXX object heimdall/CMakeFiles/heimdall.dir/source/BridgeManager.cpp.obj
[ 16%] Building CXX object heimdall/CMakeFiles/heimdall.dir/source/ClosePcScreenAction.cpp.obj
[ 20%] Building CXX object heimdall/CMakeFiles/heimdall.dir/source/DetectAction.cpp.obj
[ 23%] Building CXX object heimdall/CMakeFiles/heimdall.dir/source/DownloadPitAction.cpp.obj
[ 26%] Building CXX object heimdall/CMakeFiles/heimdall.dir/source/FlashAction.cpp.obj
[ 30%] Building CXX object heimdall/CMakeFiles/heimdall.dir/source/HelpAction.cpp.obj
[ 33%] Building CXX object heimdall/CMakeFiles/heimdall.dir/source/InfoAction.cpp.obj
[ 36%] Building CXX object heimdall/CMakeFiles/heimdall.dir/source/Interface.cpp.obj
[ 40%] Building CXX object heimdall/CMakeFiles/heimdall.dir/source/main.cpp.obj
[ 43%] Building CXX object heimdall/CMakeFiles/heimdall.dir/source/PrintPitAction.cpp.obj
[ 46%] Building CXX object heimdall/CMakeFiles/heimdall.dir/source/Utility.cpp.obj
[ 50%] Building CXX object heimdall/CMakeFiles/heimdall.dir/source/VersionAction.cpp.obj
[ 53%] Linking CXX executable ../bin/heimdall.exe
[ 53%] Built target heimdall
Scanning dependencies of target heimdall-frontend_automoc
[ 56%] Automatic moc for target heimdall-frontend
Generating moc_aboutform.cpp
Generating moc_mainwindow.cpp
[ 56%] Built target heimdall-frontend_automoc
[ 60%] Generating qrc_mainwindow.cpp
RCC: Warning: No resources in 'C:/msys64/home/JOEL/Heimdall/heimdall-frontend/mainwindow.qrc'.
[ 63%] Generating ui_mainwindow.h
[ 66%] Generating ui_aboutform.h
Scanning dependencies of target heimdall-frontend
[ 70%] Building CXX object heimdall-frontend/CMakeFiles/heimdall-frontend.dir/source/aboutform.cpp.obj
[ 73%] Building CXX object heimdall-frontend/CMakeFiles/heimdall-frontend.dir/source/Alerts.cpp.obj
[ 76%] Building CXX object heimdall-frontend/CMakeFiles/heimdall-frontend.dir/source/FirmwareInfo.cpp.obj
[ 80%] Building CXX object heimdall-frontend/CMakeFiles/heimdall-frontend.dir/source/main.cpp.obj
[ 83%] Building CXX object heimdall-frontend/CMakeFiles/heimdall-frontend.dir/source/mainwindow.cpp.obj
[ 86%] Building CXX object heimdall-frontend/CMakeFiles/heimdall-frontend.dir/source/PackageData.cpp.obj
[ 90%] Building CXX object heimdall-frontend/CMakeFiles/heimdall-frontend.dir/source/Packaging.cpp.obj
[ 93%] Building CXX object heimdall-frontend/CMakeFiles/heimdall-frontend.dir/qrc_mainwindow.cpp.obj
[ 96%] Building CXX object heimdall-frontend/CMakeFiles/heimdall-frontend.dir/heimdall-frontend_automoc.cpp.obj
[100%] Linking CXX executable ../bin/heimdall-frontend.exe
**C:/msys64/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/5.3.0/../../../../x86_64-w64-mingw32/bin/ld.exe: cannot find -ljasper
C:/msys64/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/5.3.0/../../../../x86_64-w64-mingw32/bin/ld.exe: cannot find -ldbus-1
C:/msys64/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/5.3.0/../../../../x86_64-w64-mingw32/bin/ld.exe: cannot find -lwebp
C:/msys64/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/5.3.0/../../../../x86_64-w64-mingw32/bin/ld.exe: cannot find -ldbus-1
C:/msys64/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/5.3.0/../../../../x86_64-w64-mingw32/bin/ld.exe: cannot find -lwebp
C:/msys64/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/5.3.0/../../../../x86_64-w64-mingw32/bin/ld.exe: cannot find -lsicuin
C:/msys64/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/5.3.0/../../../../x86_64-w64-mingw32/bin/ld.exe: cannot find -lsicuuc
C:/msys64/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/5.3.0/../../../../x86_64-w64-mingw32/bin/ld.exe: cannot find -lsicudt
C:/msys64/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/5.3.0/../../../../x86_64-w64-mingw32/bin/ld.exe: cannot find -lsicuin
C:/msys64/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/5.3.0/../../../../x86_64-w64-mingw32/bin/ld.exe: cannot find -lsicuuc
C:/msys64/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/5.3.0/../../../../x86_64-w64-mingw32/bin/ld.exe: cannot find -lsicudt
collect2.exe: error: ld returned 1 exit status
heimdall-frontend/CMakeFiles/heimdall-frontend.dir/build.make:349: recipe for target 'bin/heimdall-frontend.exe' failed
make[2]: *** [bin/heimdall-frontend.exe] Error 1
CMakeFiles/Makefile2:198: recipe for target 'heimdall-frontend/CMakeFiles/heimdall-frontend.dir/all' failed
make[1]: *** [heimdall-frontend/CMakeFiles/heimdall-frontend.dir/all] Error 2
Makefile:83: recipe for target 'all' failed
make: *** [all] Error 2**

davincho, 2016-05-13:
I have got the same problem - any hints?

Rurounin, 2016-05-19:
Yeah, that's pretty much the exact same problem, the easiest solution would probably be to have someone trustworthy compile it for us that are not on Win10 pro x64, the optimal solution would of course be an official build update.

nefty1029, 2016-05-29:
I was able to add some of the missing dependencies using the below command:
$ pacman -S mingw-w64-x86_64-libwebp mingw-w64-x86_64-jasper mingw-w64-x86_64-dbus

Which resulted in this:
[  6%] Built target pit
[ 53%] Built target heimdall
[ 56%] Automatic moc for target heimdall-frontend
[ 56%] Built target heimdall-frontend_automoc
[ 60%] Linking CXX executable ../bin/heimdall-frontend.exe
C:/msys64/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/5.3.0/../../../../x86_64-w64      -mingw32/bin/ld.exe: cannot find -lsicuin
C:/msys64/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/5.3.0/../../../../x86_64-w64      -mingw32/bin/ld.exe: cannot find -lsicuuc
C:/msys64/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/5.3.0/../../../../x86_64-w64      -mingw32/bin/ld.exe: cannot find -lsicudt
C:/msys64/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/5.3.0/../../../../x86_64-w64      -mingw32/bin/ld.exe: cannot find -lsicuin
C:/msys64/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/5.3.0/../../../../x86_64-w64      -mingw32/bin/ld.exe: cannot find -lsicuuc
C:/msys64/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/5.3.0/../../../../x86_64-w64      -mingw32/bin/ld.exe: cannot find -lsicudt
collect2.exe: error: ld returned 1 exit status
make[2]: *** [heimdall-frontend/CMakeFiles/heimdall-frontend.dir/build.make:352:       bin/heimdall-frontend.exe] Error 1
make[1]: *** [CMakeFiles/Makefile2:199: heimdall-frontend/CMakeFiles/heimdall-fr      ontend.dir/all] Error 2
make: *** [Makefile:84: all] Error 2

Can anyone help?

Rurounin, 2016-05-29:
I've fixed my issue with my phone using Odin but that isn't really relevant to this issue so i'll leave it open.

schindld, 2016-12-10:
@nefty1029, your command to add the missing dependencies just needs one more dependency to get it working. 
`pacman -S mingw-w64-x86_64-gcc`
plus your command:
`pacman -S mingw-w64-x86_64-libwebp mingw-w64-x86_64-jasper mingw-w64-x86_64-dbus`
got it working for me. Heimdall.exe and heimdall-frontend.exe built 100% once I added all those extra dependencies with pacman.

nefty1029, 2016-12-11:
@schindld, Thanks, but I am currently using Odin for flashing my phone. 
Hopefully this will help help those who are currently having this issue.