Benjamin-Dobell/Heimdall #342: LibUSB errors with Samsung Dart T499

cbacott, 2016-05-12:
Hello,

I know development hasn't been extremely active lately, and the app works fantastic for my other devices (GS3, GS5, GTab4).

However, I have an old Samsung Dart T499 that I've been playing with hacking around on. Unfortunately I can't get Heimdall to talk to the dang thing. I pulled the source and built from scratch with a few debug print statements to try to figure out what is going wrong. 

There seems to be some error with libusb_set_interface_alt_setting with this device. The best I've been able to pull out of libusb is LIBUSB_ERROR_OTHER, which really doesn't help. I've dug around in the libusb docs but really I can't pin down what's causing the failure exactly. Maybe someone has some better ideas? 

Output:

```
$ bin/heimdall print-pit --verbose --usb-log-level debug
Heimdall v1.4.1

Copyright (c) 2010-2014 Benjamin Dobell, Glass Echidna
http://www.glassechidna.com.au/

This software is provided free of charge. Copying and redistribution is
encouraged.

If you appreciate this software and you would like to support future
development please consider donating:
http://www.glassechidna.com.au/donate/

Initialising connection...
Detecting device...
[timestamp] [threadID] facility level [function call] <message>
--------------------------------------------------------------------------------
[ 0.002402] [00004ddf] libusbx: debug [libusb_get_device_list] 
[ 0.002440] [00004ddf] libusbx: debug [discovered_devs_append] need to increase capacity
[ 0.002448] [00004ddf] libusbx: debug [discovered_devs_append] need to increase capacity
[ 0.002456] [00004ddf] libusbx: debug [libusb_get_device_descriptor] 
[ 0.002463] [00004ddf] libusbx: debug [libusb_open] open 3.51
[ 0.002487] [00004ddf] libusbx: debug [usbi_add_pollfd] add fd 11 events 4
[ 0.002500] [00004ddf] libusbx: debug [libusb_get_device_descriptor] 
[ 0.002508] [00004ddf] libusbx: debug [add_to_flying_list] arm timerfd for timeout in 1000ms (first in line)
[ 0.002534] [00004ddf] libusbx: debug [libusb_handle_events_timeout_completed] doing our own event handling
[ 0.002546] [00004ddf] libusbx: debug [handle_events] poll() 4 fds with timeout in 60000ms
[ 0.010444] [00004ddf] libusbx: debug [handle_events] poll() returned 1
[ 0.010477] [00004ddf] libusbx: debug [reap_for_handle] urb type=2 status=-71 transferred=0
[ 0.010484] [00004ddf] libusbx: debug [handle_control_completion] handling completion status -71
[ 0.010489] [00004ddf] libusbx: debug [handle_control_completion] low-level bus error occurred
[ 0.010492] [00004ddf] libusbx: debug [disarm_timerfd] 
[ 0.010497] [00004ddf] libusbx: debug [usbi_handle_transfer_completion] transfer 0x1e84088 has callback 0x7fbd980157f0
[ 0.010499] [00004ddf] libusbx: debug [sync_transfer_cb] actual_length=0
[ 0.010516] [00004ddf] libusbx: debug [add_to_flying_list] arm timerfd for timeout in 1000ms (first in line)
[ 0.010534] [00004ddf] libusbx: debug [libusb_handle_events_timeout_completed] doing our own event handling
[ 0.010537] [00004ddf] libusbx: debug [handle_events] poll() 4 fds with timeout in 60000ms
[ 0.018694] [00004ddf] libusbx: debug [handle_events] poll() returned 1
[ 0.018727] [00004ddf] libusbx: debug [reap_for_handle] urb type=2 status=-71 transferred=0
[ 0.018730] [00004ddf] libusbx: debug [handle_control_completion] handling completion status -71
[ 0.018732] [00004ddf] libusbx: debug [handle_control_completion] low-level bus error occurred
[ 0.018745] [00004ddf] libusbx: debug [disarm_timerfd] 
[ 0.018751] [00004ddf] libusbx: debug [usbi_handle_transfer_completion] transfer 0x1e84088 has callback 0x7fbd980157f0
[ 0.018754] [00004ddf] libusbx: debug [sync_transfer_cb] actual_length=0

            length: 18
      device class: 2
               S/N: 0
           VID:PID: 04E8:6601
         bcdDevice: 0100
   iMan:iProd:iSer: 1:2:0
          nb confs: 1
[ 0.018772] [00004ddf] libusbx: debug [libusb_get_config_descriptor] index 0

interface[0].altsetting[0]: num endpoints = 1
   Class.SubClass.Protocol: 02.02.01
       endpoint[0].address: 82
           max packet size: 0010
          polling interval: 09

interface[1].altsetting[0]: num endpoints = 2
   Class.SubClass.Protocol: 0A.00.00
       endpoint[0].address: 81
           max packet size: 0200
          polling interval: 00
       endpoint[1].address: 01
           max packet size: 0200
          polling interval: 00
Claiming interface...
[ 0.018797] [00004ddf] libusbx: debug [libusb_detach_kernel_driver] interface 1
[ 0.018803] [00004ddf] libusbx: debug [libusb_claim_interface] interface 1
Setting up interface...
[ 0.018823] [00004ddf] libusbx: debug [libusb_set_interface_alt_setting] interface 1 altsetting 0
[ 0.026929] [00004ddf] libusbx: error [op_set_interface] setintf failed error -1 errno 71
DEBUG: Setup result: -99
DEBUG: LIBUSB_ERROR_OTHER
ERROR: Setting up interface failed!
Releasing device interface...
[ 0.026970] [00004ddf] libusbx: debug [libusb_release_interface] interface 1

[ 0.026987] [00004ddf] libusbx: debug [libusb_close] 
[ 0.026997] [00004ddf] libusbx: debug [usbi_remove_pollfd] remove fd 11
[ 0.027008] [00004ddf] libusbx: debug [libusb_exit] 
[ 0.027010] [00004ddf] libusbx: debug [libusb_exit] destroying default context
```

Benjamin-Dobell, 2016-05-12:
`libusbx` was merged back into mainline `libusb-1.0` and `libusbx` is now deprecated/unmaintained. Perhaps try your luck with the latest `libusb-1.0`.

cbacott, 2016-05-12:
Interesting. I installed libusb-1.0-dev before compiling the latest, so apologies for the noob question but I'm not familiar with cmake. How can I force my local heimdall build to use libusb-1.0 rather than whatever it appears to be using?

Benjamin-Dobell, 2016-05-12:
It looks in these directories:

```
        /usr/local/lib
        /opt/local/lib
        /usr/lib
```

from top to bottom, stopping on first match. Those paths are defined in https://github.com/Benjamin-Dobell/Heimdall/blob/master/cmake/Findlibusb.cmake

Depending on what OS/version you're using you may well be on the latest libusb-1.0/libusbx available from your package manager, and I do believe that there was a lot of weird stuff going on where certain distros shipped libusbx as libusb-1.0.

I'd suggest compiling the libusb-1.0 from source (into `/usr/local/lib`) - then you know for sure you're running the latest / greatest (theoretically).

cbacott, 2016-05-12:
Yep, thank you again. After my last comment I went searching into this. I'm using Mint but my version of libusb is 1.0.17 from Ubuntu, though who knows what trickery they did with the package. I pulled down the source for 1.0.20 and I'll give that a shot by sticking it in /usr/local/lib. 

Much appreciated.

cbacott, 2016-05-12:
Okay, just to close this out I compiled the latest libusb-1.0.20 and put it in /usr/local/lib, then wiped out the  cmake cache and did make clean, ensured that heimdall built with the libusb in /usr/local/lib, then set my LD_LIBRARY_PATH and confirmed we were loading libusb from /usr/local rather than the distro version. Unfortunately same result. However, given this is a libusb issue and not a Heimdall issue I suppose I'll need to go bug them about this. Thanks again for looking into this.
```
Claiming interface...
[ 0.018955] [000064d7] libusb: debug [libusb_detach_kernel_driver] interface 1
[ 0.018960] [000064d7] libusb: debug [libusb_claim_interface] interface 1
Setting up interface...
[ 0.018975] [000064d7] libusb: debug [libusb_set_interface_alt_setting] interface 1 altsetting 0
[ 0.027074] [000064d7] libusb: error [op_set_interface] setintf failed error -1 errno 71
DEBUG: Setup result: -99
DEBUG: LIBUSB_ERROR_OTHER
ERROR: Setting up interface failed!
Releasing device interface...
[ 0.027124] [000064d7] libusb: debug [libusb_release_interface] interface 1
```

closed 2016-05-12 by cbacott