Benjamin-Dobell/Heimdall #344: Heimdall not working in El Capitan

AgneevM, 2016-06-05:
Heimfall not installing properly and not recognized by Terminal (Screenshots attached)
[image]
[image]
[image]
[image]

callumgare, 2016-06-08:
Yeah, definitly still an issue. Here's a possible work around though: https://github.com/Benjamin-Dobell/Heimdall/issues/291

Benjamin-Dobell, 2016-06-08:
Closed as there's an existing issue, and solution (linked above).

closed 2016-06-08 by Benjamin-Dobell

AgneevM, 2016-06-08:
I appreciate the prompt reply.