Benjamin-Dobell/Heimdall #347: ERROR: Failed to receive file part response!

danielebonadeo, 2016-06-19:
I was trying to flash twrp recovery on my SM-A310F and this is what i always get
`$ sudo heimdall flash --RECOVERY /Users/Daniele/Desktop/recovery.img   --no-reboot
Heimdall v1.4.1

Copyright (c) 2010-2014 Benjamin Dobell, Glass Echidna
http://www.glassechidna.com.au/

This software is provided free of charge. Copying and redistribution is
encouraged.

If you appreciate this software and you would like to support future
development please consider donating:
http://www.glassechidna.com.au/donate/

Initialising connection...
Detecting device...
Claiming interface...
Setting up interface...

Initialising protocol...
Protocol initialisation successful.

Beginning session...

Some devices may take up to 2 minutes to respond.
Please be patient!

Session begun.

Downloading device's PIT file...
PIT file download successful.

Uploading RECOVERY
0%
ERROR: Failed to receive file part response!

ERROR: Retrying...
ERROR: Expected file part index: 0 Received: 1
ERROR: RECOVERY upload failed!

Ending session...
ERROR: Failed to receive session end confirmation!
Releasing device interface...

`

michaelcadilhac, 2016-09-01:
I concur: A310F does not seem to work with Heimdall.  I ended up with the very same error.

iamantony, 2016-10-31:
I faced the same problem with my A310F (A310FXXU1AOL7) when tried to flash boot.img

iamantony, 2016-11-08:
Finally I found solution to the problem. It's more like a **hack**. Be careful. You have been **warned**!

I noticed that we get error when trying to receive response from our device that it have received first file packet ([BridgeManager.cpp, line 1115][bm-receve-error]). It seems to me that device actually do not send response for the first file packet. If we skip this first response error, then file will be successfully sent to the device.

In BridgeManager.cpp replace code from line [1117][bm-1117] till [1153][bm-1153] by this code:

```
if (filePartIndex == 0)
{
    // Hack
    success = true;
    receivedPartIndex = filePartIndex;
}
else
{
    for (int retry = 0; retry < 4; ++retry)
    {
        Interface::PrintErrorSameLine("\n");
        Interface::PrintError("Retrying...");

        // Send
        sendFilePartPacket = new SendFilePartPacket(file, fileTransferPacketSize);
        success = SendPacket(sendFilePartPacket, kDefaultTimeoutSend, sendEmptyTransferFlags);
        delete sendFilePartPacket;

        if (!success)
        {
            Interface::PrintErrorSameLine("\n");
            Interface::PrintError("Failed to send file part packet 2!\n");
            return (false);
        }

        // Response
        sendFilePartResponse = new SendFilePartResponse();
        success = ReceivePacket(sendFilePartResponse);
        unsigned int receivedPartIndex = sendFilePartResponse->GetPartIndex();

        delete sendFilePartResponse;

        if (receivedPartIndex != filePartIndex)
        {
            Interface::PrintErrorSameLine("\n");
            Interface::PrintError("Expected file part index: %d Received: %d\n", filePartIndex, receivedPartIndex);
            return (false);
        }

        if (success)
            break;
    }

    if (!success)
        return (false);
}
```

[bm-receve-error]: https://github.com/Benjamin-Dobell/Heimdall/blob/master/heimdall/source/BridgeManager.cpp#L1115
[bm-1117]: https://github.com/Benjamin-Dobell/Heimdall/blob/master/heimdall/source/BridgeManager.cpp#L1117
[bm-1153]: https://github.com/Benjamin-Dobell/Heimdall/blob/master/heimdall/source/BridgeManager.cpp#L1153

erikman, 2016-12-12:
Thanks a lot! Made it possible for me to reflash a stock image on my A310F. :)

eddydonk, 2017-06-19:
Hello, i found a solution too.
I had exactly the same problem, and even the code replacement didn't work.
To succeed flash, i've just changed permission of the file(s) i would flash as executable for heimdall, with the command " chmod +x recovery.img (or anything else)"

vvvverre, 2017-07-05:
I came across this problem as well on my Galaxy A5 (2016), SM-A510F. @iamantony fix/hack worked for me.

brauliobo, 2017-08-19:
same error here with A520F/DS and 
[A5Y17LTE_EUR_OPEN_HIDDEN100M.pit.zip](https://github.com/Benjamin-Dobell/Heimdall/files/1236591/A5Y17LTE_EUR_OPEN_HIDDEN100M.pit.zip)

vitalif, 2017-08-27:
same with A510F, the hack also worked

sleeplessCortes, 2018-01-24:
where do I find the BridgeManager.cpp file on my ubuntu 16.04 system to open in editor for change???
could not find anywhere, or how do i perform this hack?

iamantony, 2018-01-24:
@sleeplessCortes you should download source code from github, edit BridgeManager.cpp and compile Heimdall

iMonZ, 2021-07-20:
> Finally I found solution to the problem. It's more like a **hack**. Be careful. You have been **warned**!
> 
> I noticed that we get error when trying to receive response from our device that it have received first file packet ([BridgeManager.cpp, line 1115](https://github.com/Benjamin-Dobell/Heimdall/blob/master/heimdall/source/BridgeManager.cpp#L1115)). It seems to me that device actually do not send response for the first file packet. If we skip this first response error, then file will be successfully sent to the device.
> 
> In BridgeManager.cpp replace code from line [1117](https://github.com/Benjamin-Dobell/Heimdall/blob/master/heimdall/source/BridgeManager.cpp#L1117) till [1153](https://github.com/Benjamin-Dobell/Heimdall/blob/master/heimdall/source/BridgeManager.cpp#L1153) by this code:
> 
> ```
> if (filePartIndex == 0)
> {
>     // Hack
>     success = true;
>     receivedPartIndex = filePartIndex;
> }
> else
> {
>     for (int retry = 0; retry < 4; ++retry)
>     {
>         Interface::PrintErrorSameLine("\n");
>         Interface::PrintError("Retrying...");
> 
>         // Send
>         sendFilePartPacket = new SendFilePartPacket(file, fileTransferPacketSize);
>         success = SendPacket(sendFilePartPacket, kDefaultTimeoutSend, sendEmptyTransferFlags);
>         delete sendFilePartPacket;
> 
>         if (!success)
>         {
>             Interface::PrintErrorSameLine("\n");
>             Interface::PrintError("Failed to send file part packet 2!\n");
>             return (false);
>         }
> 
>         // Response
>         sendFilePartResponse = new SendFilePartResponse();
>         success = ReceivePacket(sendFilePartResponse);
>         unsigned int receivedPartIndex = sendFilePartResponse->GetPartIndex();
> 
>         delete sendFilePartResponse;
> 
>         if (receivedPartIndex != filePartIndex)
>         {
>             Interface::PrintErrorSameLine("\n");
>             Interface::PrintError("Expected file part index: %d Received: %d\n", filePartIndex, receivedPartIndex);
>             return (false);
>         }
> 
>         if (success)
>             break;
>     }
> 
>     if (!success)
>         return (false);
> }
> ```

I tried this and still the same issue :(

LittleZombie1, 2021-08-20:
Hack worked with my A510F too

smartass101, 2022-04-25:
Just wanted to report that this hack also worked on a Samsung Galaxy S5 neo (SM-G903F). Standard Heimdall build indeed gave me the `ERROR: Failed to receive file part response!` message. The phone had stock firmware with Android 5.1.1 to begin with (in case it matters). The hack enabled me to upload the latest TWRP as well as the Android 6.0.1 stock ROM and afterwards a crDroid (Android 11) build flashed (via TWRP) cleanly. Due to my laziness I actually compiled the fork with the hack applied in https://github.com/alexax66/Heimdall, thanks @alexax66!

nerdoc, 2022-06-04:
@danielebonadeo Is it possible to include that in heimdall by default?

djelouze, 2023-05-04:
Didn't try the hack, but the solution from https://github.com/Benjamin-Dobell/Heimdall/issues/347#issuecomment-309370130 worked for me, with a Samsun A510F

Adelineenileda, 2023-11-07:
I have the same error and the solution [#347](https://github.com/Benjamin-Dobell/Heimdall/issues/347#issuecomment-309370130) don't work... I don't known how try the hack... A solution ?

Shamil-Xero, 2024-12-25:
> Just wanted to report that this hack also worked on a Samsung Galaxy S5 neo (SM-G903F). Standard Heimdall build indeed gave me the `ERROR: Failed to receive file part response!` message. The phone [...]

Can you suggest how to compile the heimdall build please!!