Benjamin-Dobell/Heimdall #348: S5 ERROR: Failed to send request to end PIT file transfer!

JewsDid911, 2016-06-25:
G'day mate, trying hard to install Cyanogenmod onto my S5, however I'm being met with this rough issue. I've tried using the frontend provided to use a custom PIT file I downloaded but that didn't seem to work, and this is all the CM instructions say to do.

Please help.

`C:\heimdall-suite-1.4.0-win32\Heimdall Suite>heimdall flash --RECOVERY twrp-3.0.
2-0-kltekdi.img --no-reboot
Heimdall v1.4.0

Copyright (c) 2010-2013, Benjamin Dobell, Glass Echidna
http://www.glassechidna.com.au/

This software is provided free of charge. Copying and redistribution is
encouraged.

If you appreciate this software and you would like to support future
development please consider donating:
http://www.glassechidna.com.au/donate/

Initialising connection...
Detecting device...
Claiming interface...
Setting up interface...

Initialising protocol...
Protocol initialisation successful.

Beginning session...

Some devices may take up to 2 minutes to respond.
Please be patient!

Session begun.

Downloading device's PIT file...
ERROR: Failed to send request to end PIT file transfer!
ERROR: Failed to download PIT file!
Ending session...
ERROR: Failed to send end session packet!
Releasing device interface...`

Gargravarr2112, 2016-07-17:
Hi, having the same issue using v1.4.1 from the Ubuntu 16.04 repos:

```
gargravarr@Hercules:~/Downloads$ sudo heimdall flash --RECOVERY cm-12.1-20151007-SNAPSHOT-YOG4PAO333-klte-recovery.img --no-reboot
Heimdall v1.4.1

Copyright (c) 2010-2014 Benjamin Dobell, Glass Echidna
http://www.glassechidna.com.au/

This software is provided free of charge. Copying and redistribution is
encouraged.

If you appreciate this software and you would like to support future
development please consider donating:
http://www.glassechidna.com.au/donate/

Initialising connection...
Detecting device...
Claiming interface...
Setting up interface...

Initialising protocol...
Protocol initialisation successful.

Beginning session...

Some devices may take up to 2 minutes to respond.
Please be patient!

Session begun.

Downloading device's PIT file...
ERROR: Failed to send request to end PIT file transfer!
ERROR: Failed to download PIT file!
Ending session...
ERROR: Failed to send end session packet!
Releasing device interface...
```
I tried again with --verbose but the phone was in some weird halfway state (blue progress bar onscreen) and it didn't get very far. I rebooted the phone back into download mode and tried the Print-PIT function instead:
```
gargravarr@Hercules:~/Downloads$ sudo heimdall print-pit --verbose
Heimdall v1.4.1

Copyright (c) 2010-2014 Benjamin Dobell, Glass Echidna
http://www.glassechidna.com.au/

This software is provided free of charge. Copying and redistribution is
encouraged.

If you appreciate this software and you would like to support future
development please consider donating:
http://www.glassechidna.com.au/donate/

Initialising connection...
Detecting device...
      Manufacturer: "Sasmsung"
           Product: "MSM8960"

            length: 18
      device class: 2
               S/N: 0
           VID:PID: 04E8:685D
         bcdDevice: 0100
   iMan:iProd:iSer: 1:2:0
          nb confs: 1

interface[0].altsetting[0]: num endpoints = 1
   Class.SubClass.Protocol: 02.02.01
       endpoint[0].address: 82
           max packet size: 0010
[... 156 lines cut ...]
          polling interval: 00
Claiming interface...
Setting up interface...

Initialising protocol...
ERROR: Failed to receive handshake response. Result: -7
ERROR: Protocol initialisation failed!

Releasing device interface...
```
My phone is an SM-G901F if that helps, which doesn't seem to be a vanilla S5. I have the following versions of libusb installed:
```
||/ Name                                           Version                      Architecture                 Description
+++-==============================================-============================-============================-==================================================================================================
ii  libusb-0.1-4:amd64                             2:0.1.12-28                  amd64                        userspace USB programming library
ii  libusb-1.0-0:amd64                             2:1.0.20-1                   amd64                        userspace USB programming library
ii  libusb-1.0-0:i386                              2:1.0.20-1                   i386                         userspace USB programming library
ii  libusb-1.0-0-dev:amd64                         2:1.0.20-1                   amd64                        userspace USB programming library development files
ii  libusb-1.0-doc                                 2:1.0.20-1                   all                          documentation for userspace USB programming
un  libusb-dev                                     <none>                       <none>                       (no description available)
un  libusbmuxd-tools                               <none>                       <none>                       (no description available)
ii  libusbmuxd4:amd64                              1.0.10-2ubuntu0.1            amd64                        USB multiplexor daemon for iPhone and iPod Touch devices - library
ii  libusbredirhost1:amd64                         0.7.1-1                      amd64                        Implementing the usb-host (*) side of a usbredir connection (runtime)
ii  libusbredirparser1:amd64                       0.7.1-1                      amd64                        Parser for the usbredir protocol (runtime)
```

SandPox, 2016-07-26:
Hello, my device is SC-02F which is very similar to Galaxy S5 and have the same issue, both Heimdall on Windows and Ubuntu 16.04 repo/build source are not working and have output exactly like you guys, it have android 5.0 stock ROM which i guess android 5.0 bootloader?

and here is the USB Logger for @Benjamin-Dobell , I'm pretty sure if you solve my problem then you'll solve the problem for S5 too.

https://build.sandpox.org/odin-w8.1.zip

JochenHoch2, 2016-08-08:
Hi @all,

for what it's worth: A self-compiled heimdall from current master (as of 28.7.2016 - sorry, no commit ID...) was had absolutely no problems with my SM-G900F (S5 international version/klte). I did have a bad USB port/cable combination (USB device resetting after every open()), but changing cables did the trick.

What I did:
* Compiled heimdall from master on Kubuntu 16.04 by following Linux/README.
* Updated stock ROM to 6.0.1 on phone
* Entered Download Mode
* Downloaded PIT without rebooting, worked correctly
* Printed PIT without rebooting, worked correctly
* Flashed TWRP 3.0.1, worked correctly
* Booted into Recovery

The SM-G901F is a Galaxy S5 plus and not really the same phone/hardware. 

Hope this helps anyone!

methornton, 2016-09-02:
This is still not working on Samsung SM-G900T3.  Here is my error log. what actually is libusb error -7?
```
<gserver@gserver:/opt/heimdall/Heimdall/build/bin$ heimdall download-pit --output mypit.pit --verbose --stdout-errors --usb-log-level debug --no-reboot
Heimdall v1.4.1

Copyright (c) 2010-2014 Benjamin Dobell, Glass Echidna
http://www.glassechidna.com.au/

This software is provided free of charge. Copying and redistribution is
encouraged.

If you appreciate this software and you would like to support future
development please consider donating:
http://www.glassechidna.com.au/donate/

Initialising connection...
Detecting device...
[timestamp] [threadID] facility level [function call] <message>
--------------------------------------------------------------------------------
[ 0.008476] [00004d62] libusb: debug [libusb_get_device_list] 
[ 0.008535] [00004d62] libusb: debug [discovered_devs_append] need to increase capacity
[ 0.008569] [00004d62] libusb: debug [libusb_get_device_descriptor] 
[ 0.008584] [00004d62] libusb: debug [libusb_get_device_descriptor] 
[ 0.008599] [00004d62] libusb: debug [libusb_get_device_descriptor] 
[ 0.008620] [00004d62] libusb: debug [libusb_get_device_descriptor] 
[ 0.008633] [00004d62] libusb: debug [libusb_get_device_descriptor] 
[ 0.008647] [00004d62] libusb: debug [libusb_get_device_descriptor] 
[ 0.008661] [00004d62] libusb: debug [libusb_get_device_descriptor] 
[ 0.008686] [00004d62] libusb: debug [libusb_open] open 1.11
[ 0.008736] [00004d62] libusb: debug [usbi_add_pollfd] add fd 10 events 4
[ 0.008746] [00004d62] libusb: debug [libusb_get_device_descriptor] 
[ 0.008756] [00004d62] libusb: debug [libusb_alloc_transfer] transfer 0x17035f8
[ 0.008764] [00004d62] libusb: debug [libusb_submit_transfer] transfer 0x17035f8
[ 0.008769] [00004d62] libusb: debug [add_to_flying_list] arm timerfd for timeout in 1000ms (first in line)
[ 0.008799] [00004d62] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
[ 0.008808] [00004d62] libusb: debug [handle_events] poll fds modified, reallocating
[ 0.008818] [00004d62] libusb: debug [handle_events] poll() 3 fds with timeout in 60000ms
[ 0.008915] [00004d62] libusb: debug [handle_events] poll() returned 1
[ 0.008928] [00004d62] libusb: debug [reap_for_handle] urb type=2 status=0 transferred=4
[ 0.008936] [00004d62] libusb: debug [handle_control_completion] handling completion status 0
[ 0.008942] [00004d62] libusb: debug [disarm_timerfd] 
[ 0.008951] [00004d62] libusb: debug [usbi_handle_transfer_completion] transfer 0x17035f8 has callback 0x7fe367206520
[ 0.008959] [00004d62] libusb: debug [sync_transfer_cb] actual_length=4
[ 0.008973] [00004d62] libusb: debug [libusb_free_transfer] transfer 0x17035f8
[ 0.008984] [00004d62] libusb: debug [libusb_alloc_transfer] transfer 0x17035f8
[ 0.008991] [00004d62] libusb: debug [libusb_submit_transfer] transfer 0x17035f8
[ 0.008996] [00004d62] libusb: debug [add_to_flying_list] arm timerfd for timeout in 1000ms (first in line)
[ 0.009010] [00004d62] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
[ 0.009020] [00004d62] libusb: debug [handle_events] poll() 3 fds with timeout in 60000ms
[ 0.009157] [00004d62] libusb: debug [handle_events] poll() returned 1
[ 0.009168] [00004d62] libusb: debug [reap_for_handle] urb type=2 status=0 transferred=18
[ 0.009176] [00004d62] libusb: debug [handle_control_completion] handling completion status 0
[ 0.009181] [00004d62] libusb: debug [disarm_timerfd] 
[ 0.009188] [00004d62] libusb: debug [usbi_handle_transfer_completion] transfer 0x17035f8 has callback 0x7fe367206520
[ 0.009196] [00004d62] libusb: debug [sync_transfer_cb] actual_length=18
[ 0.009204] [00004d62] libusb: debug [libusb_free_transfer] transfer 0x17035f8
      Manufacturer: "Sasmsung"
[ 0.009219] [00004d62] libusb: debug [libusb_alloc_transfer] transfer 0x17035f8
[ 0.009227] [00004d62] libusb: debug [libusb_submit_transfer] transfer 0x17035f8
[ 0.009231] [00004d62] libusb: debug [add_to_flying_list] arm timerfd for timeout in 1000ms (first in line)
[ 0.009245] [00004d62] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
[ 0.009253] [00004d62] libusb: debug [handle_events] poll() 3 fds with timeout in 60000ms
[ 0.009406] [00004d62] libusb: debug [handle_events] poll() returned 1
[ 0.009417] [00004d62] libusb: debug [reap_for_handle] urb type=2 status=0 transferred=4
[ 0.009425] [00004d62] libusb: debug [handle_control_completion] handling completion status 0
[ 0.009429] [00004d62] libusb: debug [disarm_timerfd] 
[ 0.009434] [00004d62] libusb: debug [usbi_handle_transfer_completion] transfer 0x17035f8 has callback 0x7fe367206520
[ 0.009442] [00004d62] libusb: debug [sync_transfer_cb] actual_length=4
[ 0.009451] [00004d62] libusb: debug [libusb_free_transfer] transfer 0x17035f8
[ 0.009459] [00004d62] libusb: debug [libusb_alloc_transfer] transfer 0x17035f8
[... 67 lines cut ...]
[ 1.010434] [00004d62] libusb: debug [handle_events] poll() returned 1
[ 1.010444] [00004d62] libusb: debug [reap_for_handle] urb type=3 status=-2 transferred=0
[ 1.010452] [00004d62] libusb: debug [handle_bulk_completion] handling completion status -2 of bulk urb 1/1
[ 1.010459] [00004d62] libusb: debug [handle_bulk_completion] abnormal reap: urb status -2
[ 1.010466] [00004d62] libusb: debug [handle_bulk_completion] abnormal reap: last URB handled, reporting
[... cut ...]
ERROR: Failed to receive handshake response. Result: -7
ERROR: Protocol initialisation failed!

ERROR: Protocol initialisation failed!

Releasing device interface...
[ 1.010581] [00004d62] libusb: debug [libusb_release_interface] interface 1

[ 1.010612] [00004d62] libusb: debug [libusb_close] 
[ 1.010622] [00004d62] libusb: debug [usbi_remove_pollfd] remove fd 10
[ 1.010637] [00004d62] libusb: debug [libusb_exit] 
[ 1.010644] [00004d62] libusb: debug [libusb_exit] destroying default context

```

nmandery, 2016-09-29:
Hi @all,

I can confirm the comment from @JochenHoch2 . I used heimdall @ b6fe7f8535355eb8025f4872efef6cd7d3993db6 on a SM-G900F and it worked perfectly.

turboyz, 2016-10-12:
today I **Solved** for Samsung S5 after having uninstalled .deb package Heimdall from package

Download the latest source code to your machine.

! before compiling the source code do this 2 little modifications:

1. cd into your heimdall downloaded folder root and remove the following code in CMakeLists.txt (I used only terminal, and I have no need of UI at all)
> if(NOT DISABLE_FRONTEND)
    add_subdirectory(heimdall-frontend)
    add_dependencies(heimdall-frontend heimdall)
endif()

2. search for the file "source/BridgeManager.cpp" and substitute the line as described here: https://github.com/Benjamin-Dobell/Heimdall/pull/225/commits/6b826dc1707840240438646b62011edb8fb87a53

Now you are ready for compiling the source starting installing the minimal set of libraries (no bullshit for the UI interface)
$ sudo apt-get install build-essential cmake zlib1g-dev libusb-1.0-0-dev
$ mkdir -p YourRootHeimdall/build 
$ cd YourRootHeimdall/build
$ cmake -DCMAKE_BUILD_TYPE=Release ..
$ make
$ sudo cp bin/* /usr/local/bin

Start Heimdall by issuing "heimdall"

Enjoy ;)

Toxantron, 2016-11-06:
By merging [ssimko](https://github.com/sshimko/Heimdall) into **master** I was able to flash _TWRP 3.0.1_.

If you are interested you can find it [here](https://github.com/Toxantron/Heimdall).

shiftee, 2016-11-11:
@Toxantron  Yes this works, are the developers aware of this issue?

Toxantron, 2016-11-11:
@shiftee  I think it's just one and it seems his priorities shifted. 

By reading through the commits sometime around 2014 they included an empty packet into the USB as a bugfix. I am starting to think this was actually a bug introduced by samsung engineers accidentally. The realized a little later and fixed it. Because if use `-v` you can see each of this calls fails. So this is not really a fix, it simply represses the error.

Benjamin-Dobell, 2016-11-12:
A little bit, yes. But the main issue is that these patches whilst fixing support for one device model, will break support for others.

We need detect the device that I'd connected and alter the behaviour accordingly. I'm certainly open to pull requests. It's mostly too difficult for me on my own as the cost of phones (flagships in particular) have risen significantly, so I no longer have access to variety of current models.

Toxantron, 2016-11-14:
@Benjamin-Dobell thanks for replying to this. My "fix" doesn't seem to remove behavior, it simply becomes more tolerant to the error. Never the less I agree with you. I think it is rather a problem of different Bootloader/Stock-Firmware versions. Devices running Android 5.x seem to work with my version. 

Maybe I find time to dig a little deeper into this, but just like you I only have access to one device, so this would have to be a community effort.

lope, 2016-11-29:
I just want to add my experience.

I tried download-pit with Heimdall 1.4 incl in Ubuntu 14.04. It had the issue described here.
I compiled Heimdall from source, and then it worked. I was also able to successfully flash APNHLOS and MODEM :)
**HOWEVER** heimdall-frontend still had the problem. While the commandline version compiled in the same make command worked!!! Weird. Perhaps the frontend should have a warning if errors occur suggesting to try the commandline version.
Luckily I first ran print-pit which worked. Then I tried to flash using the frontend, and it couldn't download the pit. So that gave me the *crazy idea* to try flash with the commandline binary even though I assumed the heimdal-frontend binary probably uses the same code.

Full experience here
http://forum.xda-developers.com/showpost.php?p=69843417&postcount=6864

la-costa, 2016-11-30:
I did what @turboyz posted to do and on the Samsung T320 worked perfectly.
I installed the CM 13 and it ran smoothly.

teleshoes, 2017-01-08:
thanks! <3 @turboyz @sshimko @Toxantron 
victory, s5 G900T

mariovor, 2017-01-29:
Just to let you guys know that @turboyz fork worked for I9301I Galaxy S3 Neo.

nknapp, 2017-06-01:
I tried compiling Heimdall and I was able to download the pit-file once. But after that, it never worked, until I did `sudo apt-get remove libmtp-runtime`... Then it worked...

DavidGriffith, 2017-06-14:
@nknapp I followed your lead and then found that I could reinstall libmtp-runtime and get pit files once again without trouble.