Benjamin-Dobell/Heimdall #352: No releases since 2014: please make it explicit if the project was abandoned

giacecco, 2016-07-03:
Hi All, 
I've found out about Heimdall while studying how to flash my Samsung S7 Edge firmware. I am a Linux, user, so Heimdall appeared to be the obvious solution, however the latest Heimdall distributed with Fedora 23 is very old, and from the looks of it the code in this repo is, too. 

If the project was abandoned, or if development stopped at some point in time, could you please make it very clear, so that users with newer phone can understand immediately that Heimdall is not an option? Thanks.

G.

Benjamin-Dobell, 2016-07-04:
There's commits in this repo from only a few months ago.

Don't open pointless issues.

closed 2016-07-04 by Benjamin-Dobell

dandv, 2017-10-13:
I'm also a Linux user, and glad to see the project is still active.

However, I can see where @giacecco is coming from:

* the [downloads page on BitBucket](https://bitbucket.org/benjamin_dobell/heimdall/downloads/?tab=downloads) refers to Ubuntu up to only 13
* [other sites](http://www.webupd8.org/2012/06/heimdall-odin-alternative-that-runs-on.html) link to the GitHub downloads page, which only lists versions from 2012
* the [home page](http://glassechidna.com.au/heimdall/) only lists very old Samsung devices (circa 2010).

I've just learned about Heimdall, and as an S7 owner, I actually can't tell if it will work with my device or not, unless I compile the source, or install a binary made for Ubuntu 13.

My experience in the OSS world tells me that many users will give up at this point, which hurts the project.

How can we improve this situation? Adding an updated list of devices maybe?

Benjamin-Dobell, 2017-10-13:
In the Linux world in particular package maintainers are not the same as project maintainers. I'm not at all stopping anyone from maintaining packages for the various Linux variants, however I don't have time to do so. Contributions are, and have always been welcome.