Benjamin-Dobell/Heimdall #354: Installing Heimdall Suite from Source not working on windows

rzytko, 2016-07-14:
Hello, I am having troubles with installing Heimdall Suite from source, using MSYS2. I strictly followed commands from this site: https://wiki.cyanogenmod.org/w/Install_and_compile_Heimdall
Everything works smooth until I execute the "make" command. The heimdall.exe is made correctly, but the heimdall-frontend.exe is not created. It says

>make[2]: ***  There are no principles to create the object ''heimdall-frontend/heimdall-frontend_autogen.cpp', which is required by 'heimdall-frontend/CMakeFiles/heimdall-frontend.dir/heimdall-frontend_autogen.cpp.obj'. Stop.
make[1]: *** [CMakeFiles/Makefile2:198: heimdall-frontend/CMakeFiles/heimdall-frontend.dir/all] Error 2
make: *** [Makefile:84: all] Error 2

Could anyone tell me what am I doing wrong?

capitaladot, 2016-08-07:
Same situation. Out of my depth, here.

castanley, 2016-08-16:
Same issue here