Benjamin-Dobell/Heimdall #356: Heimdall-front using wrong heimdall binary

AlexGein, 2016-07-20:
After not getting packaged heimdall to work (both cli and gui), I compiled the master from git. The cli version worked perfectly (great work btw :) ) but the heimdall-frontend wouldnt. As an experiment I uninstalled the packaged heimdall and suddenly I get this message:

FRONTEND ERROR: Failed to start Heimdall!

I havent had a look at the code, but I assume its relying on whichever heimdall executable it finds in the path, as opposed to the one in the build/bin directory.

properlyindented, 2016-07-24:
This isn't really unreasonable behaviour. Try instead to add your build/bin to the front of your path.

dandv, 2017-10-14:
@properlyindented I don't think this is reasonable. The app should search its own directory first.

That said, this worked: `PATH=./ ./heimdall-frontend`.

bugith, 2017-10-14:
Yes, it seems to be the same origin.
@properlyindented : although, I prefer my workaround way for 2 reasons:
1st: I have a graphical reminder of the problem (for me in synaptic)
2nd : I wouldn't like to have hidden workaround (like changing the PATH), and further more adding an entry to the begin of the PATH for a single particular program will have inpact on all other installed. And if all other installed programs would think that way, then the PATH would turn in a list of all installed programs, all fighting to get the first place in PATH.

Although I admit that if this PATH change is temporary, i.e. only when the program is launched and for the program itself only, this is acceptable way to do that should be handled at compile time so that non expert and casual users don't have to worry about, thinking (guessing) to create a script that would handle the issue.