Benjamin-Dobell/Heimdall #358: Solution to problems with MSVCP100.dll

TerraD, 2016-08-08:
I tried to use Heimdall on Win7 in order to put files on a Samsung Galaxy 10.1 P7510. I found the following Heimdall install problem:

1) The README.txt gives a wrong link. The wrong link is
- http://www.microsoft.com/download/en/details.aspx?id=5555
The correct link (7-Jul-2016) is 
- https://www.microsoft.com/en-us/download/confirmation.aspx?id=30679
Trying the above link ended in a message telling me that I had a newer version of the Redistributable installed (whiuch was not correct in this moment, I had only a newer version of the 2012 version x64 installed, not the 2012 x86 that is needed).

2) http://glassechidna.com.au/heimdall/ indicates correctly that you need to download 'Microsoft Visual C++ 2012 redistributables' and gives the correct link. It misses to point out, that you absolutely need to install the x86 version, even when installing on an x64 system. The given link did not work for me: Google Chrome always _automatically_ started the download of the x64 version (without any interaction)! Heimdall will not work with the x64 version even when started on an x64 system.

3) Using FireFox I was able to select between the x86 and the x64 version of 'Visual C++ Redistributable for Visual Studio 2012' (I found an Update 4 which I installed). I downloaded the x86 package ant this installed and seems to run. 
In case your FireFox shows the same behaviour as Google Chrome: I have the NoScript plugin. But I think I configured it simply to download everything from microsoft.

Umeaboy, 2017-06-05:
The correct link doesn't work anymore.

HcLightning, 2017-06-14:
You can find the needed package here:
http://www.standaloneofflineinstallers.com/2015/12/Microsoft-Visual-C-Redistributable-2015-2013-2012-2010-2008-2005-32-bit-x86-64-bit-x64-Standalone-Offline-Installer-for-Windows.html