Benjamin-Dobell/Heimdall #359: SM-T285 Failed to confirm end of file transfer sequence! when flashing custom system.img

jedld, 2016-08-10:
Tried flashing a custom system.img (I just repacked the stock system.img) on a Galaxy Tab A 7.0 LTE (SM-T285). 
Reflashing the stock system.img works fine. The tablet is soft bricked after I do this and need to reflash stock. Tried using Odin and it does not work as well so I'm not sure what is happening could be something deeper.

Heimdall v1.4.1

Copyright (c) 2010-2014 Benjamin Dobell, Glass Echidna
http://www.glassechidna.com.au/

This software is provided free of charge. Copying and redistribution is
encouraged.

If you appreciate this software and you would like to support future
development please consider donating:
http://www.glassechidna.com.au/donate/

Initialising connection...
Detecting device...
Claiming interface...
Setting up interface...

Initialising protocol...
Protocol initialisation successful.

Beginning session...

Some devices may take up to 2 minutes to respond.
Please be patient!

Session begun.

Downloading device's PIT file...
PIT file download successful.

Uploading SYSTEM
3%
ERROR: Failed to confirm end of file transfer sequence!
ERROR: SYSTEM upload failed!

Ending session...
ERROR: Failed to receive session end confirmation!
Releasing device interface...

--------------------------------------------------------------------------------------------------------------
Details usb debug logs

[ 8.459884] [000020c0] libusb: debug [disarm_timerfd] 
[ 8.459902] [000020c0] libusb: debug [usbi_handle_transfer_completion] transfer 0x1ba4198 has callback 0x7efcce247520
[ 8.459922] [000020c0] libusb: debug [sync_transfer_cb] actual_length=0
[ 8.459948] [000020c0] libusb: debug [libusb_free_transfer] transfer 0x1ba4198
[ 8.459979] [000020c0] libusb: debug [libusb_alloc_transfer] transfer 0x1ba3d88
[ 8.459999] [000020c0] libusb: debug [libusb_submit_transfer] transfer 0x1ba3d88
[ 8.460013] [000020c0] libusb: debug [add_to_flying_list] arm timerfd for timeout in 120000ms (first in line)
[ 8.460036] [000020c0] libusb: debug [submit_bulk_transfer] need 1 urbs for new transfer with length 8
[ 8.460082] [000020c0] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
[ 8.460144] [000020c0] libusb: debug [handle_events] poll() 3 fds with timeout in 60000ms
[ 8.460171] [000020c0] libusb: debug [handle_events] poll() returned 1
[ 8.460192] [000020c0] libusb: debug [reap_for_handle] urb type=3 status=0 transferred=8
[ 8.460213] [000020c0] libusb: debug [handle_bulk_completion] handling completion status 0 of bulk urb 1/1
[ 8.460233] [000020c0] libusb: debug [handle_bulk_completion] last URB in transfer --> complete!
[ 8.460250] [000020c0] libusb: debug [disarm_timerfd] 
[ 8.460270] [000020c0] libusb: debug [usbi_handle_transfer_completion] transfer 0x1ba3d88 has callback 0x7efcce247520
[ 8.460292] [000020c0] libusb: debug [sync_transfer_cb] actual_length=8
[ 8.460317] [000020c0] libusb: debug [libusb_free_transfer] transfer 0x1ba3d88

ERROR: Failed to confirm end of file transfer sequence!
ERROR: SYSTEM upload failed!

Ending session...
[ 8.460414] [000020c0] libusb: debug [libusb_alloc_transfer] transfer 0x1b9e9a8

SED4906, 2021-08-10:
I'm having the same issue you're describing on an SM-J327U
Any clues by now?

amo13, 2021-08-14:
You could try the binary I compiled with a fix from a pull request: [here](https://heimdall.free-droid.com/)

jedld, 2021-08-14:
What happened for me is that the the img files that is generated by the android build system won't work by default since it requires samsung "magic" headers. I needed to patch some things so that it correctly generates the img files.

You can look at libsparse/output_file.c

https://github.com/lineage-gtexslte/android_system_core/commit/efd1968db730e320e2c2d31845e287efa06b8eeb

amo13, 2021-08-15:
I don't really get it: aren't you supposed to flash TWRP using heimdall? Anything LineageOS-related should only be of any concern after TWRP has been successfully booted.

jedld, 2021-08-15:
I built TWRP from source so it uses the android sources from my Lineage OS
repository. TWRP was not available on the SM-T285 so I had to make one
myself and encountered this issue.