Benjamin-Dobell/Heimdall #360: unable to flash i9000

wwortel, 2016-08-21:
Saw same phenomenon for other phones:
using Heimdall 1.4.1 on Debian8 system. Tried compile from git and also Debian package, both 1.4.1, same result.
Entering command on PC
'heimdall print-pit --verbose --no-reboot':
Initialising protocol...
ERROR: Failed to receive handshake response. Result: -7
ERROR: Protocol initialisation failed!

Now the interesting observation:
took battery out and returned it. Brought phone in download mode via 3-button. Then the command works and pit info is shown, ending with 
'Ending session...
Releasing device interface...
Re-attaching kernel driver...'

However issuing the exact same  command a second time gives the error. Does not go away until a next fresh start of Download mode.
 
dmesg shows:
usb 4-1.5: USB disconnect, device number 49
usb 4-1.5: new high-speed USB device number 50 using ehci-pci
usb 4-1.5: New USB device found, idVendor=04e8, idProduct=6601
usb 4-1.5: New USB device strings: Mfr=1, Product=2, SerialNumber=0
usb 4-1.5: Product: Gadget Serial
usb 4-1.5: Manufacturer: SAMSUNG
cdc_acm 4-1.5:2.0: ttyACM0: USB ACM device

and no new entries in dmesg after having done the command a first time.

Using Heimdall via heimdall-frontend to flash a complete image gives error message:
Initialising protocol...
ERROR: Protocol initialisation failed!

jaskij, 2016-09-28:
Similar issue with 1.4.0, worked around it by using 1.3.2. Using a USB 2 port if that matters.

FreedomFighterSparrow, 2016-09-29:
Had the same issue with 1.4.1 on Linux Mint 18, switching to a USB 2 port seems to have fixed it.