Benjamin-Dobell/Heimdall #362: Heimdall for Mac build is outdated

aviallon, 2016-09-04:
Heimdall for Mac build is outdated and a lot of users are blocked. I built myself a version from the main branch in github. It's stable I used it just a few hours ago to flash my phone.
If you want, you can add them to the download website.
[Heimdall.zip](https://github.com/Benjamin-Dobell/Heimdall/files/453994/Heimdall.zip)

spoutnik12, 2016-09-08:
Hi aviallon,

Thanks for the build. however I still have a problem running it. The  app is not launching. I get this message:

Dyld Error Message:
  Library not loaded: /opt/local/libexec/qt5/lib/QtWidgets.framework/Versions/5/QtWidgets
  Referenced from: /Users/USER/Downloads/*/heimdall-frontend.app/Contents/MacOS/heimdall-frontend
  Reason: image not found

Do you have an idea what's the problem?

Benjamin-Dobell, 2016-09-22:
There's already a more recent Mac build available in these Github issues. It's built by myself and properly signed (including the kext). Please use the search feature.

closed 2016-09-22 by Benjamin-Dobell