Benjamin-Dobell/Heimdall #364: Samsung S3 --  ERROR: Protocol initialisation failed!

thekswenson, 2016-09-18:
Hello, I'm trying to install Cyanogenmod on my Verizon S3 that is running 4.2.2.

I've cloned the default git branch.
"detect" and "print-pit" work as expected.

I have problems when trying to flash the ABOOT:
"""
=> sudo Heimdall/build/bin/heimdall flash --ABOOT aboot.mbn --no-reboot
Heimdall v1.4.1

Copyright (c) 2010-2014 Benjamin Dobell, Glass Echidna
http://www.glassechidna.com.au/

This software is provided free of charge. Copying and redistribution is
encouraged.

If you appreciate this software and you would like to support future
development please consider donating:
http://www.glassechidna.com.au/donate/

Initialising connection...
Detecting device...
Claiming interface...
Setting up interface...

Initialising protocol...
ERROR: Protocol initialisation failed!

Releasing device interface...
"""

Benjamin-Dobell, 2016-09-19:
Did you reboot your device into download mode again after running `print-pit`? You must do so unless you've specified `--no-reboot` when printing pit, and then `--resume` when flashing.

thekswenson, 2016-09-19:
Thanks for the quick response!

I was using --no-reboot with print-pit but didn't know about --resume.
I just tried again and get the following:

"""
Uploading ABOOT
100%
ERROR: Failed to confirm end of file transfer sequence!
ERROR: ABOOT upload failed!
"""
Should I report another bug?

thekswenson, 2016-09-19:
The device says:
"SW REV CHECK FAIL :  Fused 2 > Binary 8"

Benjamin-Dobell, 2016-09-19:
Yes, a new issue please. However, you need to provide more information. There are heaps of Galaxy S3 devices with different model numbers - you need to include model number not the branding name.

I've never heard of that issue occurring on an S3 (regardless of the model), only on newer devices like a Note 3, and from what I've read it's something to do with locked bootloader (nothing to do with Heimdall).

Please do some research first before opening an issue. Github issues are for bug reports, not help requests. This policy isn't in place to be annoying, it's just I don't have access to every device out there. If you post on a forum you're going to get a *lot* more people see your question, and they'll probably be able to help you out.

You'll probably find the information you're after on XDA developers.

closed 2016-09-19 by Benjamin-Dobell

thekswenson, 2016-09-19:
OK.  Thanks

pemill, 2017-02-02:
Hi, i came accross this today when i tried flashing my S3 i9300 with TWRP from Ubuntu 16.04.
After some trial and error i got it working by:

1)Disconnect the USB cable
2)Setup the flash in Heimdall Frontend or comand line
3)Boot the Phone into Download Mode
4)AFTER it's in Download Mode plug in the USB and flash immediately

try this sequence

SolitaryKoda, 2017-05-15:
Been using GitHub for years but never had an account, never need to, up until now!

But the steps above by @pemill saved me from a 4/5 day headache, S3 wasn't connecting to either Heimdall nor JOdin3 to restore factory image. But now following these steps, it works and connects every time now!

Thank you again @pemill.. Saved me!

photoTed, 2017-06-18:
I got a serious headache trying to load LineageOS onto my Galaxy Note Pro 12.2 and it came down to two small details:
- connecting the tablet directly to the computer (not through a hub).
- making sure the tablet did not auto-reboot after installing the twrp recovery image: whenever the device reboots under the Samsung OS the recovery partition is overwritten with the Samsung default (no doubt a brick-recovery strategy not malicious). So don't auto-reboot (on Heimdall command line add **--no-reboot** option) and when you reboot manually be sure to be holding the volume up and home keys to go straight to twrp.

sizrar, 2017-10-23:
@pemill Thank you ! I confirm the sequence is efficient on my GT I1890L :D

jbanegas, 2018-05-17:
@pemill I try the suggested sequence in a Galaxy S3 I9300 but it didn't work. I got this error: `Initialising protocol...
ERROR: Failed to send data!Releasing device interface...`

Could you or someone else can help me please?

Thanks